thesofproject/sof #8562: [BUG][LNL] playback test causes DSP Panic

fredoh9, 2023-12-01:
**Describe the bug**
Probably replated with current LNL issues. This can be reproducible with simple playback test.

dmesg
```
[  264.900203] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx      : 0x4400000f|0x30200018: MOD_LARGE_CONFIG_SET [data size: 24]
[  264.900497] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx reply: 0x64000000|0x30200018: MOD_LARGE_CONFIG_SET
[  264.900530] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx done : 0x4400000f|0x30200018: MOD_LARGE_CONFIG_SET [data size: 24]
[  264.900536] kernel: Message payload: 00000000: 00000018 00000002 ffffff00 ffffffff
[  264.900540] kernel: Message payload: 00000010: ffff00ff ffffffff
[  264.900543] kernel: snd_sof:sof_widget_setup_unlocked: sof-audio-pci-intel-lnl 0000:00:1f.3: widget smart_amp.2.1 setup complete
[  264.900549] kernel: snd_sof:sof_ipc4_widget_setup: sof-audio-pci-intel-lnl 0000:00:1f.3: Create widget dai-copier.SSP.NoCodec-0.playback (pipe 2) - ID 4, instance 1, core 0
[  264.900553] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx      : 0x40010004|0x10056: MOD_INIT_INSTANCE [data size: 344]
[  264.904110] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc rx      : 0x1b0a0000|0x0: GLB_NOTIFICATION|EXCEPTION_CAUGHT
[  264.904115] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  264.904119] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: DSP panic!
[  264.904121] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[  264.904128] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ROM status: 0x5, ROM error: 0x0
[  264.904130] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ROM debug status: 0x0, ROM debug error: 0x0
[  264.904135] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ROM feature bit enabled
[  264.904156] kernel: snd_sof:sof_ipc4_find_debug_slot_offset_by_type: sof-audio-pci-intel-lnl 0000:00:1f.3: Slot type 0x4c455400 is not available in debug window
[  264.904158] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ------------[ DSP dump end ]------------
```

mtrace
```
[    3.649705] <inf> dai_intel_ssp: dai_ssp_pm_runtime_dis_ssp_power: dai_ssp_pm_runtime_dis_ssp_power index 0
[    3.649843] <err> dai_intel_ssp: dai_ssp_poll_for_register_delay: dai_ssp_poll_for_register_delay poll timeout reg 3332 mask 8388608 val 8388608 us 125
[    3.649848] <wrn> dai_intel_ssp: dai_ssp_pm_runtime_dis_ssp_power: dai_ssp_pm_runtime_dis_ssp_power warning: timeout
[    3.649853] <inf> dai_intel_ssp: dai_ssp_pm_runtime_dis_ssp_power: dai_ssp_pm_runtime_dis_ssp_power I2SLCTL
[    3.658881] <inf> ipc: ipc_cmd: rx	: 0x11000007|0x1
[    3.658890] <inf> pipe: pipeline_new: pipeline new pipe_id 0 priority 0
[    3.659463] <inf> ipc: ipc_cmd: rx	: 0x40000004|0x15
[    3.659518] <inf> dma: dma_get: dma_get() ID 0 sref = 1 busy channels 0
[    3.659996] <inf> ipc: ipc_cmd: rx	: 0x40000006|0x10
[    3.661133] <inf> ipc: ipc_cmd: rx	: 0x44000006|0x30000018
[    3.661675] <inf> ipc: ipc_cmd: rx	: 0x40000002|0xa
[    3.662228] <inf> ipc: ipc_cmd: rx	: 0x11010008|0x0
[    3.662233] <inf> pipe: pipeline_new: pipeline new pipe_id 1 priority 0
[    3.662746] <inf> ipc: ipc_cmd: rx	: 0x40000003|0x1000a
[    3.663388] <inf> ipc: ipc_cmd: rx	: 0x40010006|0x10010
[    3.663941] <inf> ipc: ipc_cmd: rx	: 0x44010006|0x30000018
[    3.664296] <inf> ipc: ipc_cmd: rx	: 0x4000000f|0x10026
[    3.665033] <inf> ipc: ipc_cmd: rx	: 0x4400000f|0x30200018
[    3.665428] <inf> ipc: ipc_cmd: rx	: 0x40010004|0x10056
[    3.665488] <inf> dai_intel_ssp: dai_ssp_pm_runtime_en_ssp_power: dai_ssp_pm_runtime_en_ssp_power en_ssp_power index 0
[    3.665495] <inf> dai_intel_ssp: dai_ssp_pm_runtime_en_ssp_power: dai_ssp_pm_runtime_en_ssp_power I2SLCTL
[    3.665501] <err> os: print_fatal_exception:  ** FATAL EXCEPTION
[    3.665506] <err> os: print_fatal_exception:  ** CPU 0 EXCCAUSE 13 (load/store PIF data error)
[    3.665510] <err> os: print_fatal_exception:  **  PC 0xa0038e5f VADDR 0x28108
[    3.665513] <err> os: print_fatal_exception:  **  PS 0x60120
[    3.665516] <err> os: print_fatal_exception:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:1 CALLINC:2)
[    3.665520] <err> os: z_xtensa_dump_stack:  **  A0 0xa005de54  SP 0xa00f0c20  A2 0x400dc5b4  A3 0x28100
[    3.665523] <err> os: z_xtensa_dump_stack:  **  A4 0x400d9440  A5 0x301  A6 0xa00c9cac  A7 0xf4240
[    3.665526] <err> os: z_xtensa_dump_stack:  **  A8 0xa005e044  A9 0xa00f0bb0 A10 0xa00d91d8 A11 0x20c0
[    3.665530] <err> os: z_xtensa_dump_stack:  ** A12 0xfff001ff A13 0x12 A14 0x400f6840 A15 0xa00f0bb0
[    3.665531] <err> os: z_xtensa_dump_stack:  ** LBEG 0xa0035db8 LEND 0xa0035dca LCOUNT 0xa005aa6c
[    3.665535] <err> os: z_xtensa_dump_stack:  ** SAR 0x14

Backtrace:0xa0038e5c:0xa00f0c20 0xa005de51:0xa00f0c90 0xa005de0b:0xa00f0cc0 0xa003687c:0xa00f0ce0 0xa006255c:0xa00f0d00 0xa00472a2:0xa00f0d40 0xa00505b0:0xa00f0d70 0xa0070671:0xa00f0e40 0xa006da44:0xa00f0e70 0xa006d004:0xa00f0ea0 0xa0070543:0xa00f0ed0 0xa003bd18:0xa00f0ef0 0xa00600a9:0xa00f0f50 0xa005fe85:0xa00f0f80 0xa003b221:0xa00f0fb0 0xa0061ae6:0xa00f0fe0 0xa00615e0:0xa00f1000 0xa0064c9d:0xa00f1020 0xa0055b3f:0xa00f1040 0xa00599df:0xa00f1070 

[    3.665665] <err> os: z_fatal_error: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[    3.665671] <err> os: z_fatal_error: Current thread: 0x400f8698 (unknown)
[    3.668726] <err> zephyr: k_sys_fatal_error_handler: Halting system
Terminated
```

**To Reproduce**
TPLG=/lib/firmware/intel/development/sof-lnl-nocodec.tplg MODEL=LNLM_RVP_NOCODEC SOF_TEST_INTERVAL=5 ~/sof-test/test-case/check-playback.sh -d 3 -l 1 -r 1 -F

**Reproduction Rate**
100% with above command

**Environment**

- Linux Branch: [topic/sof-dev](https://github.com/thesofproject/linux/tree/topic/sof-dev)
- Linux Commit: [266ee31b71f8](https://github.com/thesofproject/linux/commit/266ee31b71f88434f317d78a9ce3e3aec31fdbd8)
- SOF Branch: [main](https://github.com/thesofproject/sof/tree/main)
- SOF Commit: [a8ad6ebfc852](https://github.com/thesofproject/sof/commit/a8ad6ebfc852a9248508b1308665f75beaf87b9e)
- Zephyr Commit: [efc32081893d](https://github.com/zephyrproject-rtos/zephyr/commit/efc32081893dd607dfc51938ef93787872008fe2)

**Screenshots or console output**

[dmesg-lnl-playback.txt](https://github.com/thesofproject/sof/files/13531002/dmesg-lnl-playback.txt)
[mtrace-lnl-playback.txt](https://github.com/thesofproject/sof/files/13531003/mtrace-lnl-playback.txt)

jxstelter, 2023-12-05:
The issue seems to be caused by inverted condition for SSP CPA check, that was introduced during some code improvements.
Test code with fix available on my private branch:
https://github.com/jxstelter/sof/tree/lnl-rvp-d3-workarround
SOF: https://github.com/thesofproject/sof/commit/27cdad0aa5efac94de50c73a7fb4b6e191e263b2

I tested reproduction scenario on this build many times without reproduction.
Please verify if it on Your side.

keqiaozhang, 2023-12-06:
Confirmed that https://github.com/zephyrproject-rtos/zephyr/pull/66215 can fix this issue.

lgirdwood, 2023-12-18:
@keqiaozhang @fredoh9 can we close ?

keqiaozhang, 2023-12-19:
> @keqiaozhang @fredoh9 can we close ?

This issue cannot be reproduced in CI. Closing it.

closed 2023-12-19 by keqiaozhang