thesofproject/sof #8606: [FEATURE] remove duplicated CMake rules for audio modules

kv2019i, 2023-12-11:
**Is your feature request related to a problem? Please describe.**
The list of files to build is duplicated for the shared-library build. On time in top-level
src/audio/CMakeLists.txt and second time in subfolders.

marc-hb, 2023-12-11:
Found in
- #8595
- https://github.com/thesofproject/sof/pull/8514/files#r1404344228

Should probably be done before zephyr cleanup:
- #8260

kv2019i, 2024-03-04:
Stable-v2.9 branched, this didn't make the cut, bumping to 2.10.

kv2019i, 2024-06-05:
Stable branched for 2.10, so moving this to 2.11.

kv2019i, 2024-12-13:
Feature cutoff for v2.12, moving this to v2.13.

kv2019i, 2025-04-28:
https://github.com/thesofproject/sof/issues/8260 was completed for v2.13, but this won't be ready. Moving to v2.14.