thesofproject/sof #8626: [BUG] xrun on LNL_RVP_AIOC in multiple pipelien test

aiChaoSONG, 2023-12-14:
**Describe the bug**
In multiple pipeline(playback/capture/or_both) test, we see clear io error in the terminal, and there are errors in dmesg.

```
[ 2412.885835] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx reply: 0x33000007|0x0: GLB_SET_PIPELINE_STATE
[ 2412.885865] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: FW reported error: 7 - Unsupported operation requested
[ 2412.885948] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc error for msg 0x13010004|0x0
[ 2412.885972] kernel: soundwire_intel soundwire_intel.link.0: ASoC: error at soc_dai_trigger on SDW0 Pin2: -22
[ 2412.888769] kernel:  Deepbuffer Jack Out: ASoC: trigger FE cmd: 1 failed: -22
[ 2412.888774] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-lnl 0000:00:1f.3: pcm: trigger stream 31 dir 0 cmd 0
[ 2412.888776] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-lnl 0000:00:1f.3: trigger cmd: 0 state: 3
[ 2412.890990] kernel: snd_sof_intel_hda_common:hda_dai_trigger: soundwire_intel soundwire_intel.link.0: cmd=1 dai SDW0 Pin2 direction 0
[ 2412.891035] kernel: snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc4 set pipeline instance 1 state 4
[ 2412.891039] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx      : 0x13010004|0x0: GLB_SET_PIPELINE_STATE
[ 2412.891652] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx reply: 0x33000007|0x0: GLB_SET_PIPELINE_STATE
[ 2412.891660] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: FW reported error: 7 - Unsupported operation requested
[ 2412.891699] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc error for msg 0x13010004|0x0
[ 2412.891705] kernel: soundwire_intel soundwire_intel.link.0: ASoC: error at soc_dai_trigger on SDW0 Pin2: -22
[ 2412.894727] kernel:  Jack Out: ASoC: trigger FE cmd: 1 failed: -22
```
In mtrace, there are `no bytes to copy` message, and also request dma channel failure:
```
[   26.765328] <inf> copier: copier_prepare: comp:1 0x10004 copier_prepare()
[   26.765338] <err> dai_comp: dai_common_config_prepare: comp:1 0x10004 dai_common_config_prepare(): dma_request_channel() failed
[   26.765341] <err> module_adapter: module_prepare: comp:1 0x10004 module_prepare() error -5: module specific prepare failed, comp_id 65540
[   26.765353] <err> module_adapter: module_adapter_prepare: comp:1 0x10004 module_adapter_prepare() error fffffffb: module prepare failed
[   26.765358] <err> pipe: pipeline_prepare: pipe:1 0x0 pipeline_prepare(): ret = -5, dev->comp.id = 3
[   26.765361] <err> ipc: ipc4_pcm_params: ipc: pipe 1 comp 0 prepare failed -5
[   26.765415] <inf> component: comp_set_state: comp:1 0x10004 comp_set_state(), state already set to 1
[   26.765420] <err> ipc: ipc_cmd: ipc4: FW_GEN_MSG failed with err 7
[   26.769585] <wrn> dai_comp: dai_common_copy: comp:3 0x30004 dai_zephyr_copy(): nothing to copy
[   26.770575] <wrn> dai_comp: dai_common_copy: comp:3 0x30004 dai_zephyr_copy(): nothing to copy
[   26.771248] <inf> ipc: ipc_cmd: rx	: 0x13010004|0x0
[   26.771326] <inf> copier: copier_prepare: comp:1 0x10004 copier_prepare()
[   26.771336] <err> dai_comp: dai_common_config_prepare: comp:1 0x10004 dai_common_config_prepare(): dma_request_channel() failed
[   26.771348] <err> module_adapter: module_prepare: comp:1 0x10004 module_prepare() error -5: module specific prepare failed, comp_id 65540
[   26.771351] <err> module_adapter: module_adapter_prepare: comp:1 0x10004 module_adapter_prepare() error fffffffb: module prepare failed
[   26.771356] <err> pipe: pipeline_prepare: pipe:1 0x0 pipeline_prepare(): ret = -5, dev->comp.id = 3
[   26.771360] <err> ipc: ipc4_pcm_params: ipc: pipe 1 comp 0 prepare failed -5
[   26.771413] <inf> component: comp_set_state: comp:1 0x10004 comp_set_state(), state already set to 1
[   26.771418] <err> ipc: ipc_cmd: ipc4: FW_GEN_MSG failed with err 7
```

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)

**Reproduction Rate**
run multiple-pipline test, for example: `TPLG=/lib/firmware/intel/sof-ace-tplg/sof-lnl-rt711-l0-rt1316-l23-rt714-l1.tplg MODEL=LNLM_SDW_AIOC SOF_TEST_INTERVAL=5 ~/sof-test/test-case/multiple-pipeline.sh -f p -l 50`

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {topic/sof-dev    c052a47323f9}
    * SOF: {main    3e09f233a48d}
2) Name of the topology file
    * Topology: {sof-lnl-rt711-l0-rt1316-l23-rt714-l1.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {LNL SDW rt711@l0 rt1316@l23 rt714@l1}

[dmesg.txt](https://github.com/thesofproject/sof/files/13669440/dmesg.txt)
[mtrace.txt](https://github.com/thesofproject/sof/files/13669441/mtrace.txt)

keqiaozhang, 2023-12-18:
A duplicate of https://github.com/thesofproject/sof/issues/8275.

plbossart, 2024-05-28:
I think we can close this one? @marc-hb @fredoh9 @lgirdwood I don't think we've seen this in a very long time.

marc-hb, 2024-05-28:
Indeed I haven't.

fredoh9, 2024-05-28:
I haven't seen this error for a long time. Closing now

closed 2024-05-28 by fredoh9