thesofproject/sof #8630: [FEATURE] organize audio module source code to per-module directories

kv2019i, 2023-12-14:
**Is your feature request related to a problem? Please describe.**
Currently audio module source files are a bit spread around the codebase. Clean up the codebase to so that all key files and definitions for a module are in a single place.

**Describe the solution you'd like**

1) Per module directory which contains its own headers, Kconfig and CMakefile.

2) Any metadata should be in this folder (.e.g IPC4 toml file sections to describe the cycle budgets)

3) A README.MD that describes module features, capabilities and performance.

kv2019i, 2023-12-14:
Related PRs
- https://github.com/thesofproject/sof/pull/8587
- https://github.com/thesofproject/sof/pull/8557
- https://github.com/thesofproject/sof/pull/8514
- https://github.com/thesofproject/sof/pull/8595

kv2019i, 2023-12-14:
One open issue is how to tackel modules that build on each others functionality directly (like crossover, drc and mutliband-drc). Reusing functionality is something we want to promite as well, so the effort to isolate modules should not make this harder.

marc-hb, 2023-12-15:
`zephyr_library()` and `zephyr_library_include_directories()` is what Zephyr seems to use to solve this problem. 

```
west grep zephyr.*include_dir | wc -l
805

west manifest --path
zephyr/west.yml
```

I couldn't find documentation about this in https://docs.zephyrproject.org/latest/ but there is a lot in the source of https://github.com/zephyrproject-rtos/zephyr/blob/main/cmake/modules/extensions.cmake

cc: @andyross, @cujomalainey , @johnylin76 , @btian1 , @lyakh , @singalsu , @RanderWang

btian1, 2023-12-18:
@marc-hb ， if we use zephyr syntax， how about non-zephyr case？ I mean CONFIG_ZEPHYR_NATIVE_DRIVERS not defined?

marc-hb, 2023-12-18:
> how about non-zephyr case？ I mean CONFIG_ZEPHYR_NATIVE_DRIVERS not defined?

Then we need an indirection like the one used in
- #8548

lgirdwood, 2023-12-19:
> @marc-hb ， if we use zephyr syntax， how about non-zephyr case？ I mean CONFIG_ZEPHYR_NATIVE_DRIVERS not defined?
> 
> There are three ways to cover this:
> 
> 1. keep current ../../
> 2. Marc's proposal.
> 3. move those common to a unify header and store it in math folder.

Any modules that have any public module APIs can have these APIs published in the include/module directory. e.g. I think IIR and FIR will have public APIs for other modules to use/call.

kv2019i, 2023-12-20:
@lgirdwood wrote:
> Any modules that have any public module APIs can have these APIs published in the include/module directory. e.g. I think IIR 
> and FIR will have public APIs for other modules to use/call.

I'm not sure the crosssover-drc-multibanddrc falls into this case. This is rather a closely related family of algorithms where multiband-drc uses the e.g. crossover completely. Exposing a complete algorithm under "math" does not seem good. I still think the "../../" solution fits this case better. Or we keep the headers of "shared modules" in the include directory (include/module perhaps) -- and not apply this reorganization to these shared modules.

lgirdwood, 2023-12-21:
> @lgirdwood wrote:
>
>
> I'm not sure the crosssover-drc-multibanddrc falls into this case. This is rather a closely related family of algorithms where multiband-drc uses the e.g. crossover completely. [...]

I mean ```src/include/module/module/crossover``` where clients would ```#include <module/crossover/x.h>```

btian1, 2024-01-05:
most modules refactor are done for now, main open for this feature is: README.MD, need @singalsu input on this.

lgirdwood, 2024-01-08:
> most modules refactor are done for now, main open for this feature is: README.MD, need @singalsu input on this.

Great, whats' still remaining - I still see some active PRs for some 3P modules.

closed 2024-01-24 by btian1