thesofproject/sof #8641: [BUG] DSP panic when testing suspend/resume with audio on TGLU_RVP_NOCODEC

keqiaozhang, 2023-12-18:
**Describe the bug**
Observed this issue on TGLU_RVP_NOCODEC only when testing suspend/resume case with audio, it seems like a regression and can be reproduced manually. I will do the further check.

**dmesg**
```
[  392.104470] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13000003|0x1: GLB_SET_PIPELINE_STATE [data size: 16]
[  392.105825] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000000|0x1: GLB_SET_PIPELINE_STATE
[  392.105905] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x13000003|0x1: GLB_SET_PIPELINE_STATE [data size: 16]
[  392.105911] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13000004|0x1: GLB_SET_PIPELINE_STATE [data size: 16]
[  392.109341] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx      : 0x1b0a0000|0x0: GLB_NOTIFICATION|EXCEPTION_CAUGHT
[  392.109348] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  392.109351] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: DSP panic!
[  392.109353] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[  392.109376] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x00000005: module: ROM, state: FW_ENTERED, running
[  392.109576] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW is built with XCC toolchain
[  392.109579] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: DSP Firmware Oops
[  392.109580] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[  392.109582] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EXCCAUSE 0x0000000d EXCVADDR 0x00000010 PS       0x00060120 SAR     0x00000019
[  392.109584] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPC1     0xbe04c00e EPC2     0x00000000 EPC3     0x00000000 EPC4    0x00000000
[  392.109586] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[  392.109588] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPS2     0x00000000 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[  392.109589] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000000
[  392.109590] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: stack dump from 0x00000000
[  392.109592] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: AR registers:
[  392.109594] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x0: be03f3f5 be0b0e40 be0bf240 be0bf304
[  392.109596] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x10: 00000000 be0bf2e4 00000000 be04c008
[  392.109597] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x20: be0576e9 00000000 00000000 00000000
[  392.109599] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x30: be0576e9 00000000 00000000 00000000
[  392.109601] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
```

**To Reproduce**
~/sof-test/test-case/check-suspend-resume-with-audio.sh -l 5 -m playback

**Reproduction Rate**
50%+

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {[c052a47323f9](https://github.com/thesofproject/linux/commit/c052a47323f9f544937e90749edabc031a6d114e)}
    * SOF: {[ddf70fa033a8](https://github.com/thesofproject/sof/commit/ddf70fa033a81f88950c5baa48e20ea965291e02)}
2) Name of the topology file
    * Topology: {development/sof-tgl-nocodec.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {TGLU_RVP_NOCODEC-ipc4}

[dmesg.txt](https://github.com/thesofproject/sof/files/13700093/dmesg.txt)

[mtrace.txt](https://github.com/thesofproject/sof/files/13700096/mtrace.txt)

keqiaozhang, 2023-12-18:
This issue a FW regression and caused by https://github.com/thesofproject/sof/pull/8580.

RanderWang, 2023-12-18:
NO issue on MTL ?

keqiaozhang, 2023-12-19:
> NO issue on MTL ?

Yes, TGLU-NOCODEC only, not even on ADL-NOCODEC.

lgirdwood, 2023-12-21:
@ujfalusi nocodec has faster timing and probably guarantees at least 1 LL tick has occurred between a host IPC and host IPC programming. Could we enforce some delays in host between IPC responses and IP programming to ensure current LL tick is done in FW.

RanderWang, 2023-12-22:
should be fixed by https://github.com/thesofproject/sof/pull/8657

RanderWang, 2023-12-22:
@keqiaozhang , please close it, thanks

keqiaozhang, 2023-12-25:
Confirmed that this issue has been fixed by https://github.com/thesofproject/sof/pull/8657. Closing this bug.

closed 2023-12-25 by keqiaozhang