thesofproject/sof #8659: [BUG] Missing CPC data for SDW DMIC capture

kv2019i, 2023-12-20:
**Describe the bug**
Kernel emits the following warnings:

```
[   11.699159] sof-audio-pci-intel-mtl 0000:00:1f.3: COPIER (UUID: 9BA00C83-CA12-4A83-943C-1FA2E82F9DDA): No CPC match in the firmware file's manifest (ibs/obs: 384/192)
[   11.726603] sof-audio-pci-intel-mtl 0000:00:1f.3: EQIIR (UUID: 5150C0E6-27F9-4EC8-8351-C705B642D12F): No CPC match in the firmware file's manifest (ibs/obs: 384/384)
```

**To Reproduce**
Boot system to GUI 

**Reproduction Rate**
100%

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
Annoyance, warning emitted whenever PCM opened

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: n/a
    * SOF: SOF2.8
2) Name of the topology file
    * Topology: sof-mtl-rt713-l0-rt1316-l12-rt1713-l3.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: mtl

kv2019i, 2023-12-20:
@btian1 @bardliao  We missed one case with https://github.com/thesofproject/sof/issues/8464 . FYI to @abonislawski

btian1, 2023-12-21:
yes, there is no case for:
COPIER: 384 --> 192
EQ-IIR:  384 --> 384.

I would propose add these two cases, @kv2019i

btian1, 2023-12-25:
two patches from kernel and sof are both merged, closing:
1. sof https://github.com/thesofproject/sof/pull/8660, add missed CPC in this PR.
2. down grade CPC warning in kernel logging:https://github.com/thesofproject/linux/pull/4753

Closing as fixed.

closed 2023-12-25 by btian1