thesofproject/sof #8661: [BUG][LNL] Speaker playback start to distort in the middle of playing

aiChaoSONG, 2023-12-21:
**Describe the bug**
In verifying kernel patch https://github.com/thesofproject/linux/pull/4748, I found that the speaker playback on LNL_RVP_AIOC with RT711 + 2xRT1316 + RT714 codecs is distorted in the middle of the playback. That is, on playback start up, the sound is good, and after playing for a while (about 10s), the sound starts to become distorted.

This issue only happens on LNL_RVP_AIOC, MTL_RVP_AIOC is good.

Below picture is the wave I recorded with my phone:
[image]

Normal Sound, zoomed in
[image]

Distorted Sound, zoomed in
[image]

**To Reproduce**
aplay -D hw:0,2 -fdat -vvv sine_wave.wav, and play for a while.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {topic/sof-dev}, Since the enabling commit, the issue could be reproduced.  I tested the latest kernel with https://github.com/thesofproject/linux/pull/4748 included, commit id: 0cba055bc21c, and our daily kernel from 2023/10/26, commit id: c0dfcde0414.
    * SOF: {main/stable-v2.8}
2) Name of the topology file
    * Topology: {sof-lnl-rt711-l0-rt1316-l23-rt714-l1.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {LNL_RVP_AIOC}

ujfalusi, 2023-12-21:
@aiChaoSONG, this is puzzling for sure.
Can you provide kernel and firmware log (only from playback start to distortion start+10 seconds).

If you stop and re-start the playback right away when it is distorted (not allowing DSP to suspend), will the distortion be gone again and re-appear?

Is it possible to test similar case with HDA setup?

ujfalusi, 2023-12-21:
@plbossart, @bardliao, FYI

aiChaoSONG, 2023-12-21:
@ujfalusi Just work with Bard, and we come to the conclusion that this issue is caused by link aggregation on LNL. We are using one DMA to drive two SDW links, which HW guys said won't work.