thesofproject/sof #8662: [BUG][LNL] Firmware boot failure during the suspend/resume test with audio

keqiaozhang, 2023-12-21:
**Describe the bug**
This issue happens on LNL-NOCODEC platform when testing suspend/resume with audio, the reproduction rate is not high, but it can be reproduced manually.

**dmesg**
```
[  923.205471] kernel: snd_sof:snd_sof_run_firmware: sof-audio-pci-intel-lnl 0000:00:1f.3: booting DSP firmware
[  923.205474] kernel: snd_sof_intel_hda_common:hda_dsp_cl_boot_firmware: sof-audio-pci-intel-lnl 0000:00:1f.3: IMR restore supported, booting from IMR directly
[  923.206545] kernel: snd_sof_pci_intel_mtl:mtl_dsp_core_power_up: sof-audio-pci-intel-lnl 0000:00:1f.3: FW Poll Status: reg[0x178d04]=0x2000101 successful
[  923.206554] kernel: snd_sof_pci_intel_mtl:mtl_dsp_cl_init: sof-audio-pci-intel-lnl 0000:00:1f.3: Primary core power up successful
[  923.206561] kernel: snd_sof_pci_intel_mtl:mtl_dsp_cl_init: sof-audio-pci-intel-lnl 0000:00:1f.3: FW Poll Status: reg[0x73214]=0x80000000 successful
[  923.206587] kernel: snd_sof_pci_intel_mtl:mtl_enable_interrupts: sof-audio-pci-intel-lnl 0000:00:1f.3: FW Poll Status: reg[0x1800]=0x41 successful
[  923.206595] kernel: snd_sof_pci_intel_mtl:mtl_enable_interrupts: sof-audio-pci-intel-lnl 0000:00:1f.3: FW Poll Status: reg[0x1140]=0x1 successful
[  923.214965] kernel: nvme nvme0: Shutdown timeout set to 10 seconds
[  923.221067] kernel: nvme nvme0: 8/0/0 default/read/poll queues
[  925.210279] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  925.210295] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: Firmware boot failure due to timeout
[  925.210301] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: fw_state: SOF_FW_BOOT_IN_PROGRESS (3)
[  925.210351] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ROM status: 0x0, ROM error: 0x0
[  925.210358] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ROM debug status: 0x0, ROM debug error: 0x0
[  925.210385] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ROM feature bit enabled
[  925.210484] kernel: snd_sof:sof_ipc4_find_debug_slot_offset_by_type: sof-audio-pci-intel-lnl 0000:00:1f.3: Slot type 0x4c455400 is not available in debug window
[  925.210492] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  925.210497] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: error: failed to boot DSP firmware after resume -5
[  925.210504] kernel: snd_sof:sof_set_fw_state: sof-audio-pci-intel-lnl 0000:00:1f.3: fw_state change: 3 -> 4
[  925.210511] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: PM: dpm_run_callback(): pci_pm_resume+0x0/0xe0 returns -5
[  925.210546] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: PM: failed to resume async: error -5
[  925.234494] kernel: OOM killer enabled.
[  925.234496] kernel: Restarting tasks ... 
```

**To Reproduce**
~/sof-test/test-case/check-suspend-resume-with-audio.sh -l 5 -m playback

**Reproduction Rate**
20%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {[0cba055bc21c](https://github.com/thesofproject/linux/commit/0cba055bc21c6356a5dc68fd1b86419cfec8890c)}
    * SOF: {[60212cf3870e](https://github.com/thesofproject/sof/commit/60212cf3870ed403d8ce8c335669b7426827d353)}
2) Name of the topology file
    * Topology: {development/sof-lnl-nocodec.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {LNLM_RVP_NOCODEC}

[dmesg.txt](https://github.com/thesofproject/sof/files/13736607/dmesg.txt)
[mtrace.txt](https://github.com/thesofproject/sof/files/13736608/mtrace.txt)

lrudyX, 2024-04-26:
@fredoh9 Could you check if the issue is still reproducible?

fredoh9, 2024-04-26:
I haven't seen this for a while

closed 2024-04-29 by lgirdwood