thesofproject/sof #8680: rimage: segfault when parsing the wrong key

teburd, 2023-12-29:
I mistakenly put public rather than private key as the key argument and got a confusing segfault in place of a nice error message. @marc-hb mentioned this should probably be ticketed, so here it is!

aryansharma6827, 2024-02-29:
@marc-hb please assign me this issue i would like to work on this.

lgirdwood, 2024-02-29:
@aryansharma6827 assigned - thanks !

marc-hb, 2024-02-29:
Thanks @aryansharma6827 ! Note you don't need to wait to be "officially" assigned to start working. Adding a comment is enough. fter 2 years without activity you can safely assume no one else is working on this :-)

Also note A LOT of rimage has changed in the meantime. As the first thing please compare today's version with the versions in the report. Make sure they still reproduce the same. Either way please comment and share what the new status is.

The main rimage branch is now in the main sof repo, it's not in the separate rimage repo any more.

kv2019i, 2024-06-05:
@aryansharma6827 Any update? We have the feature cutoff now for v2.10, so any fix will need to be first merged to main and then backported to stable-v2.10 in order to make the official 2.10 release.

jsarha, 2024-06-12:
I have trouble reproducing the issue. I checked out 3e442f5f311b SOF commit, to find a version where rimage was still a submodule under SOF, and reset rimage repo to 32052ad and run `PRIVATE_KEY_OPTION='-DRIMAGE_PRIVATE_KEY=/home/oku/work/sof-ancient/sof/keys/otc_public_key_3k.pem' ./scripts/xtensa-build-all.sh tgl` only to get quite reasonable error:
```
Firmware completing manifest v2.5
 meta: completing ADSP manifest
 meta: limit is 0x44ac0
 auth: completing authentication manifest
 cse: ri_css_v2_5_hdr_create completing CSS manifest
 css: set build date to 2024:05:12
 cse: completing CSE V2.5 manifest
 cse: cse checksum b292b361
Firmware file size 0x49000 page count 67
 rimage_read_key: read key '/home/oku/work/sof-ancient/sof/keys/otc_public_key_3k.pem'
error: validating RSA private key.
gmake[3]: *** [src/arch/xtensa/CMakeFiles/run_rimage.dir/build.make:70: src/arch/xtensa/CMakeFiles/run_rimage] Error 234
gmake[2]: *** [CMakeFiles/Makefile2:2215: src/arch/xtensa/CMakeFiles/run_rimage.dir/all] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:2302: src/arch/xtensa/CMakeFiles/bin.dir/rule] Error 2
gmake: *** [Makefile:858: bin] Error 2
```
I am not sure what piece I am missing, but it could be the SOF commit used in reproducing the problem. Maybe the rimage flags were different, but I am unable to guess what they were. I'll continue to study the rimage shortcomings with valgrind, but since I can not reproduce the original problem, there is no way to tell if I have fixed the original issue.

jsarha, 2024-06-12:
Hmm, makes me wonder if the original problem was in libcrypto.so, and if its now fixed and that is why its now impossible to reproduce the problem. @lgirdwood @marc-hb any ideas on how to pursue this issue further, are welcome.

lgirdwood, 2024-06-12:
@jsarha rename the public key using the private key name and rerun.

jsarha, 2024-06-12:
> @jsarha rename the public key using the private key name and rerun.

Makes no difference. Still the same error:
```
Firmware completing manifest v2.5
 meta: completing ADSP manifest
 meta: limit is 0x44ac0
 auth: completing authentication manifest
 cse: ri_css_v2_5_hdr_create completing CSS manifest
 css: set build date to 2024:05:12
 cse: completing CSE V2.5 manifest
 cse: cse checksum b292b361
Firmware file size 0x49000 page count 67
 rimage_read_key: read key '/home/oku/work/sof-ancient/sof/keys/otc_private_key_3k.pem'
error: validating RSA private key.
gmake[3]: *** [src/arch/xtensa/CMakeFiles/run_rimage.dir/build.make:70: src/arch/xtensa/CMakeFiles/run_rimage] Error 234
gmake[2]: *** [CMakeFiles/Makefile2:2215: src/arch/xtensa/CMakeFiles/run_rimage.dir/all] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:2302: src/arch/xtensa/CMakeFiles/bin.dir/rule] Error 2
gmake: *** [Makefile:858: bin] Error 2
```

lgirdwood, 2024-06-12:
> > @jsarha rename the public key using the private key name and rerun.
> 
> @lgirdwood , Makes no difference. Still the same error:
> 
> ```
> Firmware completing manifest v2.5
>  meta: completing ADSP manifest
>  meta: limit is 0x44ac0
>  auth: completing authentication manifest
>  cse: ri_css_v2_5_hdr_create completing CSS manifest
>  css: set build date to 2024:05:12
>  cse: completing CSE V2.5 manifest
>  cse: cse checksum b292b361
> Firmware file size 0x49000 page count 67
>  rimage_read_key: read key '/home/oku/work/sof-ancient/sof/keys/otc_private_key_3k.pem'
> error: validating RSA private key.
> gmake[3]: *** [src/arch/xtensa/CMakeFiles/run_rimage.dir/build.make:70: src/arch/xtensa/CMakeFiles/run_rimage] Error 234
> gmake[2]: *** [CMakeFiles/Makefile2:2215: src/arch/xtensa/CMakeFiles/run_rimage.dir/all] Error 2
> gmake[1]: *** [CMakeFiles/Makefile2:2302: src/arch/xtensa/CMakeFiles/bin.dir/rule] Error 2
> gmake: *** [Makefile:858: bin] Error 2
> ```

ok, crash no longer reproducible and we can close.

closed 2024-06-12 by lgirdwood

marc-hb, 2024-06-14:
> Hmm, makes me wonder if the original problem was in libcrypto.so, 

I think you must be right. I checked out the old commit too and found  the expected error message without any crash. valgrind did not complain either.

Just for the record, detailed reproduction steps:

```
git clone https://github.com/thesofproject/sof
git checkout cdcf283391af2928
git submodule update --init --recursive 
git -C rimage checkout 32052add9968a22d
git cherry-pick -x 55e862c92ec66 # docker-run.sh: switch from "sof" shortcut to full "thesoftproject/sof"

docker pull thesofproject/sof

# first, verify the correct key works
./scripts/docker-run.sh ./scripts/xtensa-build-all.sh tgl

# Try the wrong key
export PRIVATE_KEY_OPTION='-DRIMAGE_PRIVATE_KEY=/home/sof/work/sof.git/keys/otc_public_key_3k.pem' 
./scripts/docker-run.sh ./scripts/xtensa-build-all.sh -v tgl

rimage_read_key: read key '/home/sof/work/sof.git/keys/otc_public_key_3k.pem'
error: validating RSA private key.
gmake[3]: *** [src/arch/xtensa/CMakeFiles/run_rimage.dir/build.make:70: src/arch/xtensa/CMakeFiles/run_rimage] Error 234

# Show the rimage command used so it can re-run with valgrind outside the container
./scripts/docker-run.sh  make -C build_tgl_gcc/ -j1 VERBOSE=1

cd /home/sof/work/sof.git/build_tgl_gcc/src/arch/xtensa &&
   ../../../rimage_ep/build/rimage -o sof-tgl.ri -c /home/sof/work/sof.git/rimage/config/tgl.toml -k /home/sof/work/sof.git/keys/otc_public_key_3k.pem -i 3 -f 2.0 -b 1 -e bootloader-tgl sof-tgl
```

marc-hb, 2024-06-14:
```
./scripts/docker-run.sh /bin/bash

~/work/sof.git$ ldd build_tgl_gcc/rimage_ep/build/rimage

	libcrypto.so.3 => /lib/x86_64-linux-gnu/libcrypto.so.3 (0x000071ba84400000)
```

marc-hb, 2024-06-15:
segfault reproduced when rewinding the docker image to a version which has `/lib/x86_64-linux-gnu/libcrypto.so.1.1`

```diff
--- a/scripts/docker-run.sh
+++ b/scripts/docker-run.sh
@@ -32,4 +32,4 @@ docker run -i -v "${SOF_TOP}":/home/sof/work/sof.git \
 	--env https_proxy="$https_proxy" \
 	--user "$(id -u)" \
 	$SOF_DOCKER_RUN \
-	thesofproject/sof "$@"
+	thesofproject/sof:20230206 "$@"
```

Case closed.