thesofproject/sof #8686: [BUG]IPC error when testing multiple-pipeline on MTL-RVP NOCODEC

RanderWang, 2024-01-03:
**Kernel log:**
[ 2924.250651] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-mtl 0000:00:1f.3: trigger cmd: 0 state: 2
[ 2924.261128] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x33000000|0x1: GLB_SET_PIPELINE_STATE
[ 2924.261147] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x13000003|0x1: GLB_SET_PIPELINE_STATE [data size: 12]
[ 2924.261155] kernel: Message payload: 00000000: 00000002 00000002 00000003
[ 2924.261183] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x13000002|0x1: GLB_SET_PIPELINE_STATE [data size: 12]
[ 2924.261543] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x330000a4|0x1: GLB_SET_PIPELINE_STATE
[ 2924.261547] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: FW reported error: 164 - Unknown
[ 2924.261626] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc error for msg 0x13000002|0x1
[ 2924.261675] kernel: Message payload: 00000000: 00000002 0000000a 00000009
[ 2924.261680] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to set final state 2 for all pipelines

**FW log:**
[ 2924.260746] <inf> ipc: ipc_cmd: rx   : 0x13000002|0x1^M
[ 2924.260775] <inf> pipe: pipeline_trigger: pipe:10 0x0 pipe trigger cmd 0^M
[ 2924.260783] <wrn> pm_device: runtime_suspend: Unbalanced suspend^M
[ 2924.260788] <err> host_comp: host_common_trigger: comp:10 0xd0004 host_trigger(): dma stop failed: -120^M
[ 2924.260791] <err> pipe: pipeline_trigger_run: pipe:10 0x0 pipeline_trigger_run(): ret = -120, host->comp.id = 851972, cmd = 0^M
[ 2924.260803] <err> ipc: ipc4_pipeline_trigger: pipeline 10: trigger cmd 0 failed with: -120^M
[ 2924.260806] <err> ipc: ipc_compound_post_start: failed to process msg 19 status 164^M

RanderWang, 2024-01-03:
solution: https://github.com/zephyrproject-rtos/zephyr/pull/67135

marc-hb, 2024-01-12:
Similar looking issue #8729 with a lot more details.

RanderWang, 2024-01-15:
fixed

marc-hb, 2024-01-17:
multiple-pause-resume is still very unreliable even with your fix @RanderWang , can you please take a look?
- https://github.com/thesofproject/sof/pull/8747#issuecomment-1895265451

kv2019i, 2024-02-01:
@RanderWang @marc-hb How come all these bugs are closed even if the fix is not merged to SOF mainline? I don't see these errors in PR testing -- how can that be if the fix is not merged?

marc-hb, 2024-02-01:
> How come all these bugs are closed even if the fix is not merged to SOF mainline? 

Good question for @RanderWang who closed it as soon as the fix was merged in Zephyr.

In my book bugs are closed after at least one round of daily tests but opinions might differ?

Github's wrong opinion and really bad practice does not help: https://github.com/orgs/community/discussions/66741

kv2019i, 2024-03-07:
Fixed and merged to both SOF main and stable-v2.9.