thesofproject/sof #8698: [BUG] audio glitches on LNL-NOCODEC

keqiaozhang, 2024-01-05:
**Describe the bug**
Observed this issue in CI daily test, From the waveform, there are some glitches during alsabat check, most of them look like below:
[image]

The reproduction rate is about 20%. 

**To Reproduce**
~/sof-test/test-case/check-alsabat.sh -c hw:sofnocodec,0 -p hw:sofnocodec,0 -C 2 -F 599

**Reproduction Rate**
About 20%

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {[ca6f04c7a4f9](https://github.com/thesofproject/linux/commit/ca6f04c7a4f929d4a1520145bc96b55751c10239)}
    * SOF: {[0698fa334ae1](https://github.com/thesofproject/sof/commit/0698fa334ae16b7709f08d093f70f5a9795479cb)}
2) Name of the topology file
    * Topology: {sof-lnl-nocodec.tplg }
3) Name of the platform(s) on which the bug is observed.
    * Platform: {LNL-NOCODEC}

[dmesg.txt](https://github.com/thesofproject/sof/files/13839310/dmesg.txt)

plbossart, 2024-01-05:
Just to be clear, this glitch happens when doing an SSP loopback?

This could be a problem on playback, on capture or both.

Can we check if this happens on playback only by attaching a scope or Logic Pro to the SSP pins?

keqiaozhang, 2024-01-05:
> this glitch happens when doing an SSP loopback?

Yes, it happens when checking an SSP loopback, will try to check the signals on SSP pins.

plbossart, 2024-01-05:
My BKM for testing such issues is to inject a sawtooth signal with a counter value increasing to e.g. 1023 and falling back to 0. I record the samples, export to excel and look for deltas different to 1 or 1023.

keqiaozhang, 2024-01-15:
The new finding shows that there are 48 anomalous sampling points for each glitch.

[image]
[image]

[wav.zip](https://github.com/thesofproject/sof/files/13935230/wav.zip)

plbossart, 2024-01-15:
that's exactly one firmware tick...

btian1, 2024-01-16:
Below are also observed yesterday during my testing, it is around 40 period missing, however it can't be reproduced if I open mtrace to collect logs:

[image]

lgirdwood, 2024-01-17:
@btian1 would be interesting to check the saw tooth in your pipeline, maybe modify the copiers in each direction to see where in the pipeline the data is being missed. i.e. data should always increment by 1 and you could log when it didn't and where. This could help rule out host DMA and perhaps one stream direction.

btian1, 2024-01-22:
@lgirdwood ， I tried with sawtooth wav file, with help from Keqiao, we can reproduce this case with below simple cmd:
aplay -Dplughw:0,0 -f dat sawtooth_440_5min30s.wav  -d 30 & arecord -Dhw:0,0 -f dat -c 2 -d 30 record30_3.wav

Below is some observation:

1. clock freq is correct: 393216000
2. mtl nocodec can't reproduce this issue(> 10 times triage), only lnl nocodec an reproduce(20-30% rate)
3. Confirmed zero samples are coming from mixin mixout process, here is the code location:
    https://github.com/thesofproject/sof/blob/main/src/audio/mixin_mixout/mixin_mixout.c#L428
    https://github.com/thesofproject/sof/blob/main/src/audio/mixin_mixout/mixin_mixout.c#L540
4. both application and sof have underrun or overrun happened:
   app:
  ubuntu@sh-lnlm-rvp-nocodec-01:~/baofeng$ aplay -Dplughw:0,0 -f dat sawtooth_440_5min30s.wav -d 30 & arecord -Dhw:0,0 -f dat -c 2 -d 30 record30_1.wav
[1] 4379
Playing WAVE 'sawtooth_440_5min30s.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
Recording WAVE 'record30_1.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
overrun!!! (at least 59.689 ms long)
underrun!!! (at least 95.357 ms long)
underrun!!! (at least 158.302 ms long)
overrun!!! (at least 137.827 ms long)
underrun!!! (at least 1.218 ms long)
underrun!!! (at least 0.115 ms long)
  
sof side:

[  730.232996] <inf> host_comp: host_get_copy_bytes_normal: comp:3 0x50004 no bytes to copy, available samples: 0, free_samples: 384
[  730.233996] <inf> host_comp: host_get_copy_bytes_normal: comp:3 0x50004 no bytes to copy, available samples: 0, free_samples: 384

As Rander mentioned, LNL are using HDDMA and MTL is using GPDMA, so suspect it maybe a DMA config related issue on LNL.

plbossart, 2024-01-22:
wait, is this an issue with playback and capture concurrently, or do the xruns happen for either direction separately?

These would be completely different issues, the former pointing to SSP configs or programming sequences that interfere, and the latter with a DMA programming issue.

Can we please start with a single playback?

btian1, 2024-01-23:
Thanks for reminder, @plbossart , I do checked single playback again this morning.

Single playback also have a rate to reproduce this issue(lower can playback and capture concurrent), I do observed host underrun during playback:

[67434.714441] <inf> ll_schedule: zephyr_domain_thread_fn: ll core 0 timer avg 1602, max 5679, overruns 0
[67435.738441] <inf> ll_schedule: zephyr_domain_thread_fn: ll core 0 timer avg 1581, max 1611, overruns 0
[67436.762441] <inf> ll_schedule: zephyr_domain_thread_fn: ll core 0 timer avg 1581, max 1617, overruns 0
[67437.786441] <inf> ll_schedule: zephyr_domain_thread_fn: ll core 0 timer avg 1581, max 1614, overruns 0
[67438.810441] <inf> ll_schedule: zephyr_domain_thread_fn: ll core 0 timer avg 1581, max 1615, overruns 0
[67439.834441] <inf> ll_schedule: zephyr_domain_thread_fn: ll core 0 timer avg 1581, max 1613, overruns 0
[67440.858441] <inf> ll_schedule: zephyr_domain_thread_fn: ll core 0 timer avg 1581, max 1610, overruns 0
[67441.882441] <inf> ll_schedule: zephyr_domain_thread_fn: ll core 0 timer avg 1581, max 1610, overruns 0
[67442.519426] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x4 no bytes to copy, available samples: 0, free_samples: 192
[67442.519440] <err> mixer: mixin_process: comp:0 0x2 tbftbf No source info    // just added for indicate mixin silent process.
[67442.906441] <inf> ll_schedule: zephyr_domain_thread_fn: ll core 0 timer avg 1581, max 1855, overruns 0
[67443.930441] <inf> ll_schedule: zephyr_domain_thread_fn: ll core 0 timer avg 1581, max 1621, overruns 0
[67444.954440] <inf> ll_schedule: zephyr_domain_thread_fn: ll core 0 timer avg 1581, max 1613, overruns 0
[67445.978441] <inf> ll_schedule: zephyr_domain_thread_fn: ll core 0 timer avg 1581, max 1614, overruns 0

Based on your comments: " the latter with a DMA programming issue", should we go with zephyr DMA programming?

btian1, 2024-01-23:
no I did not observe XRUN from linux side.

"no bytes to copy" happened at the middle of playback, not beginning, means there is playback pipeline underrun happened, agree on this?

plbossart, 2024-01-23:
pipeline xrun does not mean xrun on the interface unfortunately, we have countless examples of this warning having no impact on the results.

wszypelt, 2024-03-08:
@keqiaozhang does this issue still occur? if so, please send logs with payloads

keqiaozhang, 2024-03-15:
From the TRC logs, this issue no longer exists in CI tests.
[image]

closed 2024-03-15 by keqiaozhang