thesofproject/sof #8704: [BUG] Using Internal Speaker on Laptop with Tiger Lake-H, Audio is delayed when playing from silence

Aikoyori, 2024-01-06:
**Describe the bug**
When I try to play audio using my laptop (MSI GP76 Leopard 11UG with Intel i7-11800H), the audio is muted for around a second before playing, skipping the silent part. Only happens on Internal Speaker and does not happen on Headphones. 

**To Reproduce**
1. Be on Tiger Lake
2. Play some music from silent

**Reproduction Rate**
Every time I started playing any audio, even system beep  (the water drop sound)

**Expected behavior**
The sound plays normally without silence part

**Impact**
It is in fact very annoying when I don't want to wear earbuds.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 6.6.9-arch1 (24a07cdb07452c7e631da9d3913ea5d3bddbc71f)
    * SOF: 2023.12-1 (from Arch Linux Repository)
2) Name of the topology file
    * Topology: snd-intel-hda
3) Name of the platform(s) on which the bug is observed.
    * Platform: Arch Linux

**Screenshots or console output**

https://github.com/thesofproject/sof/assets/12034280/32e46aa7-f89b-4dff-87da-395e23835f19

https://github.com/thesofproject/sof/assets/12034280/6f5be9aa-b074-44c5-aee4-ad6a3d8618b4

Aikoyori, 2024-01-06:
Also happens on my Ultramarine Linux 38 USB stick running `alsa-sof-firmware 2.2.5-1.fc38` and Linux Kernel 6.3.5-200.fc38.x86_64

plbossart, 2024-01-08:
Please follow the recommendations in https://thesofproject.github.io/latest/getting_started/intel_debug/suggestions.html#disable-sof-on-pci-hdaudio-devices-to-test-audio-playback and check if the problem happens with

````
options snd-intel-dspcfg dsp_driver=1
````

Aikoyori, 2024-01-09:
I will try that out later today. Thanks.

Aikoyori, 2024-01-09:
The problem still happens to a lesser degree (the delay is shorter but still exists).

plbossart, 2024-01-09:
It's probably a slow ramp on the amplifier side, not an SOF bug proper but a codec driver configuration.

If you are using pulseaudio or pipewire the easiest 'fix' would be to disable the suspend-on-idle which would keep streaming zeroes into the audio device and prevent it from suspending. Not great for power but it would keep the missing initial audio.

Fixing the driver would require help from the codec vendor, or reverse-engineering the Windows setup https://thesofproject.github.io/latest/getting_started/intel_debug/suggestions.html#reverse-engineer-the-windows-audio-driver, which isn't straightforward and does not come with any guarantees of success.

korewaChino, 2024-06-12:
Cannot reproduce on same chipset,
Tried to reproduce on:
```
        *-multimedia
             description: Multimedia audio controller
             product: Tiger Lake-LP Smart Sound Technology Audio Controller
             vendor: Intel Corporation
             physical id: 1f.3
             bus info: pci@0000:00:1f.3
             logical name: card0
             logical name: /dev/snd/controlC0
             logical name: /dev/snd/hwC0D0
             logical name: /dev/snd/hwC0D2
             logical name: /dev/snd/pcmC0D0c
             logical name: /dev/snd/pcmC0D0p
             logical name: /dev/snd/pcmC0D31p
             logical name: /dev/snd/pcmC0D3p
             logical name: /dev/snd/pcmC0D4p
             logical name: /dev/snd/pcmC0D5p
             logical name: /dev/snd/pcmC0D6c
             logical name: /dev/snd/pcmC0D7c
             version: 20
             width: 64 bits
             clock: 33MHz
             capabilities: pm msi bus_master cap_list
             configuration: driver=sof-audio-pci-intel-tgl latency=32
             resources: iomemory:600-5ff iomemory:600-5ff irq:154 memory:6001138000-600113bfff memory:6001000000-60010fffff
```

Problematic machine:
```
id: multimedia
description: 	Multimedia audio controller
product: 	Tiger Lake-H HD Audio Controller
vendor: 	Intel Corporation
physical id: 	1f.3
bus info: 	pci@0000:00:1f.3
logical name: 	card0
logical name: 	/dev/snd/controlC0
logical name: 	/dev/snd/hwC0D0
logical name: 	/dev/snd/hwC0D2
logical name: 	/dev/snd/pcmC0D0c
logical name: 	/dev/snd/pcmC0D0p
logical name: 	/dev/snd/pcmC0D31p
logical name: 	/dev/snd/pcmC0D3p
logical name: 	/dev/snd/pcmC0D4p
logical name: 	/dev/snd/pcmC0D5p
logical name: 	/dev/snd/pcmC0D6c
logical name: 	/dev/snd/pcmC0D7c
version: 	11
width: 	64 bits
clock: 	33MHz
capabilities: 	pm msi bus_master cap_list
configuration:	
driver	=	sof-audio-pci-intel-tgl
latency	=	32
resources:	
iomemory	:	620-61f
iomemory	:	620-61f
irq	:	200
memory	:	6203148000-620314bfff
memory	:	6203000000-62030fffff
```

The firmware version seems different, @Aikoyori's version is 11 while mine is 20. So probably a firmware issue.
Any idea how you would update this?