thesofproject/sof #8715: [BUG][ARL] Boot from IMR fails after system suspend

kv2019i, 2024-01-09:
**Describe the bug**
DSP boot fails after system resume.

When cold boot is forced (with "sof_debug=0x80"), issue is not seen. 

**To Reproduce**

**Reproduction Rate**
TBD (not seen on all systems)

**Expected behavior**
IMR boot works after system suspend

**Impact**
high

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev
    * SOF: 2024-01-08 daily
2) Name of the topology file
    * Topology: hda-genreic
3) Name of the platform(s) on which the bug is observed.
    * Platform: arl

**Screenshots or console output**

mengdonglin, 2024-01-11:
There is a potential fix https://github.com/thesofproject/linux/pull/4770

kv2019i, 2024-01-11:
This issue is limited to resume from S3.

kv2019i, 2024-01-23:
With https://github.com/thesofproject/linux/pull/4775 merged on SOF Linux driver side, we can close this as invalid/wontfix, right?

ujfalusi, 2024-01-23:
@kv2019i, yes, we are not going to try IMR boot in case when we resume from S3 suspend.

closed 2024-01-24 by kv2019i