thesofproject/sof #8717: [BUG][META] alsa_conformance_test fails due to too high rate

kv2019i, 2024-01-09:
**Describe the bug**
This is a meta bug to track issues related to problems found by https://chromium.googlesource.com/chromiumos/platform/audiotest/+/HEAD/alsa_conformance_test.md with SOF driver.

Issues related to this meta bug:

- https://github.com/thesofproject/sof/issues/8458
- https://github.com/thesofproject/sof/issues/4617
- https://github.com/thesofproject/sof/issues/8512
- https://github.com/thesofproject/sof/issues/6756

** What is the test **

ALSA conformance rate test monitors (by busylooping on host CPU) value of snd_pcm_avail() and records any change of hardware pointer and uses a linear regression to estimate the rate. 

** Why many SOF targets fail the test **

SOF inserts a programmable DSP between the application and the audio codec. The SOF devices must adhere to ALSA PCM API semantics, but within the limits set, the DSP can optimize data transfers based on its needs and the needs of the overall SoC. To alsa_conformance_test , this shows as high degree of burstiness of the snd_pcm_avail() values, when it polls the values many times within one ALSA period. 

** Ways for driver to alert user-space of its particular behaviour **

ALSA has a capability interface for driver to declare particular characteristics of drivers. Following interface could be of potential benefit for test apps like alsa_conformance_test to better understand the device it is testing:

| Driver flag   | alsa-lib  |  Usage in applications  |  Can help with SOF and alsa_conformance_test  | 
|---|---|---|---|
|SNDRV_PCM_INFO_BATCH   |  snd_pcm_hw_params_is_batch()  |  Widely used by apps to detect drivers that update hw_ptr only once per period.  |  Not helpful, SOF drivers can updated hw_ptr continuously   |
|SNDRV_PCM_INFO_DOUBLE  | sndpcm_hw_params_is_double()   |  No known usage in apps. Only a few kernel drivers set. |  Not helpful, SOF is not double-buffering like the RME drivers in kernel   | 
|SNDRV_PCM_INFO_BLOCK_TRANSFER   |  snd_pcm_hw_params_is_block_transfer()  | No known usage in apps, but set by most ALSA driver... but not by any SOF driver.  | Semantics very unclear, device transfers data in block but that's it. Given this is set by so many drivers without any actual user-space usage, starting to use that with more specific semantics does not seem like a good idea. | 

** Alternative solutions **

In the linked bugs, it has been pointed out that a more robust test for sample rate correctness and sample rate stability is to playback/capture a known reference tone and analyze its characteristics (as described in https://github.com/thesofproject/sof/issues/4617#issuecomment-919776192 . The alsa_conformance_test allows to capture failures of this type without need for an external test setup (reference tone generator/capture), so it has its own merits and places of use (and it has been used long in ChromeOS and Android), but the methodology limitations should be noted.

plbossart, 2024-01-10:
"in most cases, setting the merge threshold to be equal to period size, will result in a passing test with low error rate."

did you mean equal to the DSP firmware period size (1ms) ?

kv2019i, 2024-01-12:
@plbossart wrote:
> "in most cases, setting the merge threshold to be equal to period size, will result in a passing test with low error rate."
> 
> did you mean equal to the DSP firmware period size (1ms) ?

I did actually refer to period size on purpose. The below commentary is potentially a bit redudant (or at least beside the point) as my latest tests this week demonstrate that the existing method in alsa-conformance-test to figure out the step increase median, actually works very well with SOF, and e.g. with the TGL and MTL targets, finds the 1ms burst rate reliably.

But with that disclaimer, my rationale to not use LL tick length is that most SOF targets will have extra buffering towards the host. E.g. Intel SOF targets have a DMA buffer of 4 x LL period, e.g. 4ms in typical configuration (and this really doesn't scale for e.g. 10ms LL tick). The exact reasons why this is 4 (and not 2) is subject to some long discussions as this magic number dates back to origins of the SOF project. Notably this is platform specific and other SOF targets may use a lower/higher number. What we can say though is:

In normal operating conditions:
 - SOF DSP will consume/produce data every 1ms, there's no 4ms delay added by design

In exceptional cases:
 - If DDR transactions are delayed (but DSP continues LL ticks), the host side SRAM  buffer can mitigate this (but transient longer bursts are seen)
 - If DSP LL scheduler is overloaded (but host DMA continues), the host-side buffer can mitigate this and avoid data loss.

Both are in practise extremely rare occurences, but the extra slices of buffering protect again audible glitches. But they also add to the maximum burst size.

OTOH, bursts that would exceed the set period size, would cause problems to large amount of ALSA apps, so this woud seem to be a safe limit for the application in terms of max expected burst size.

That brings me to my opening sentence in the comment: in practise the alsa-conformance-test's mechanism to discover the median step size, seems to be working well. I'll make another comment on other sources of errors I've discovered.

marc-hb, 2024-01-16:
> On many SOF platforms, the driver can report accurate hardware pointer value even during DMA bursts. From alsa_conformance_test's point of view, this shows up as high degree of burstiness of the snd_pcm_avail() values, when it polls the values many times within one ALSA period.

I don't understand all the details but something sounds fundamentally wrong.

- Either the test is a low-level, fine-grained, white box test that has knowledge of **all relevant information** about DMAs, scheduling and burstiness,
- or the test is high-level, abstract black box test that does not know implementation details. Then it should look ONLY at the final product (= the audio)

A test cannot be both high level and low level at the same time. Yet a lot of the discussions seem to constantly swing back and forth between high level and low level. This does not make sense; which is it?

> In summary, there is no means to relay the size of DMA bursts from driver to user-space application.

So this can't be a "white box" test for now, not until it is possible to reliably expose the needed implementation details. Until the required information is available, tests should stay high-level and they should stop trying and failing to make burstiness *guesses*. For now testing should stick instead to actual, confirmed Xruns and not what happens before that point. Real-time does not care about averages, it's only about making deadlines.

What am I missing?

>  works for the majority of cases

A test that makes guesses and works "most of the time" does a lot more harm than good.

Don't get me wrong: collecting and graphing  information is still extremely useful, even with incomplete knowledge. But at the end of the day the PASS/FAIL decision cannot rely on guesses.

kv2019i, 2024-01-19:
@marc-hb wrote:
> A test cannot be both high level and low level at the same time. Yet a lot of the discussions seem to constantly swing back and > forth between high level and low level. This does not make sense; which is it?
[...]
> So this can't be a "white box" test for now, not until it is possible to reliably expose the needed implementation details. Until the > required information is available, tests should stay high-level and they should stop trying and failing to make burstiness 
> guesses.

I think you hit the nail on the head, this is the core issue and something I want to address in this metabug. So questions discussed in linked bugs:
1) does ALSA PCM interface limit how bursty hw_ptr can be for driver to still comply with ALSA PCM interface?
2) is the merge-threshold mechanism using information that is inherently implementation specific?

So I looked at (1) again and my take is there are no strict established limits other than the period-size as configured by the application.  I surveyed the hw info flags and the facilities to describe hw_ptr behaviour and they are very limited. Also the almost non-existant use of these flags in applications (and alsa-lib) shows this is not something apps depend on (with the named exception of BATCH which I describe in above, used by all sound servers).

As for (2), in multiple older bugs, we'd had the hypothesis merge-threshold is not handlng the DMA bursts correctly and this would then be invalid assumption for a black-box ALSA PCM interface test. When I've now done fresh debug on current gen platforms, I can't confirm this hypothesis. In fact, in the cases I've looked at, merge-threshold is working correctly and the rate error comes from a different source.

So status as of now is that this is a valid blackbox test of the ALSA PCM interface. We'll need to take another look at bugs like https://github.com/thesofproject/sof/issues/8458 to see where the rate error comes from (or have new proof it comes from DMA bursts).

ford-prefect, 2025-01-09:
Back when we had Linux Audio Miniconferences, there was a recurring theme that would come up, viz. `SNDRV_PCM_INFO_BLOCK_TRANSFER` is not a fine-grained enough indicator for user-space, and that perhaps having a way for the driver to expose the transfer size would be useful -- then userspace could use that for some sort of safety margin.

lgirdwood, 2025-01-13:
@ford-prefect ack - we were all discussing this again last week. @kv2019i @ujfalusi @ranj063 @bardliao @sathya-nujella @udaymb 

Our DMA is non linear vs time or asynchronous when copying audio, this is to conserve power and only copy when memory bus is active already. It has logic to make sure data is always copied before deadline though. The delta to real position is also variable between a min and max i.e. we cant pass up a constant to userspace. I think @kv2019i is working on some proposals.

kv2019i, 2025-01-29:
Just to add a note to @lgirdwood @ford-prefect , in context of alsa_conformance_test (i.e. this bug), I'm standing by my previous comment (see my comment above Jan 2024). IOW, "So status as of now is that this is a valid blackbox test of the ALSA PCM interface. " The hw_ptr burstiness makes debugging conformance test bugs harder, but as far as I can tell surveying our existing base of bugs, it's not the rootcause of failures.

The new bug with Pipewire is something different:
https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/4489
... for this, we can reconsider exposing the maximum burst size info. I can file a separate ticket for this.

kv2019i, 2025-05-02:
And follow-up to previous comment, clearing out v2.13 from this.