thesofproject/sof #8721: [BUG] DSP panic during the suspend/resume test w/ audio on TGL-NOCODEC

keqiaozhang, 2024-01-10:
**Describe the bug**
Observed this issue in CI daily test, it happened during the suspend/resume test. the reproduction rate is high. Maybe it's a regression, I will check it later.

**dmesg**
```
[12543.439854] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-tgl 0000:00:1f.3: trigger cmd: 0 state: 2
[12543.439867] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13000002|0x1: GLB_SET_PIPELINE_STATE [data size: 16]
[12543.443817] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx      : 0x1b0a0000|0x0: GLB_NOTIFICATION|EXCEPTION_CAUGHT
[12543.443825] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[12543.443828] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: DSP panic!
[12543.443830] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[12543.443839] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x00000005: module: ROM, state: FW_ENTERED, running
[12543.443947] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW is built with XCC toolchain
[12543.443951] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: DSP Firmware Oops
[12543.443953] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EXCCAUSE 0x0000003f EXCVADDR 0x00000000 PS       0x00060e20 SAR     0x0000001b
[12543.443957] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPC1     0xbe059a37 EPC2     0x00000000 EPC3     0x00000000 EPC4    0x00000000
[12543.443961] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[12543.443963] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPS2     0x00000000 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[12543.443966] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000000
[12543.443968] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: stack dump from 0x00000000
[12543.443971] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: AR registers:
[12543.443974] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x0: be014fb8 be0b0000 00000004 be0b0010
[12543.443977] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x10: be0afff0 00000004 00000020 00045147
[12543.443980] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x20: be014179 be0aff70 be07bb30 be0b0010
[12543.443983] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x30: be014179 be0aff70 be07bb30 be0b0010
[12543.443986] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[12543.443988] kernel: snd_sof:sof_set_fw_state: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state change: 7 -> 8
[12543.444015] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done : 0x1b0a0000|0x0: GLB_NOTIFICATION|EXCEPTION_CAUGHT
[12543.943887] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13000002|0x1
[12543.943897] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Attempting to prevent DSP from entering D3 state to preserve context
[12543.943901] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[12543.943920] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[12543.943925] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[12543.943947] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Host IPC initiator: 0x93000002|0x1|0x0, target: 0x0|0x0|0x80000000, ctl: 0x3
[12543.943952] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[12543.943955] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[12543.943957] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
[12543.943960] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_CRASHED (8)
[12543.943970] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x00000005: module: ROM, state: FW_ENTERED, running
[12543.944080] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW is built with XCC toolchain
[12543.944084] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: DSP Firmware Oops
[12543.944087] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EXCCAUSE 0x0000003f EXCVADDR 0x00000000 PS       0x00060e20 SAR     0x0000001b
[12543.944091] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPC1     0xbe059a37 EPC2     0x00000000 EPC3     0x00000000 EPC4    0x00000000
[12543.944095] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[12543.944098] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPS2     0x00000000 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[12543.944101] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000000
[12543.944104] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: stack dump from 0x00000000
[12543.944106] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: AR registers:
[12543.944110] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x0: be014fb8 be0b0000 00000004 be0b0010
[12543.944114] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x10: be0afff0 00000004 00000020 00045147
[12543.944117] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x20: be014179 be0aff70 be07bb30 be0b0010
[12543.944121] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x30: be014179 be0aff70 be07bb30 be0b0010
[12543.944124] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
```

**To Reproduce**
 ~/sof-test/test-case/check-suspend-resume-with-audio.sh -l 5 -m playback

**Reproduction Rate**
High, >50%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {[623c00b5e977](https://github.com/thesofproject/linux/commit/623c00b5e977117ffac07952244fe341b383997b)}
    * SOF: {[b07d6e9161a6](https://github.com/thesofproject/sof/commit/b07d6e9161a6a66b8da2b58523e9db4e4779ec06)}
2) Name of the topology file
    * Topology: {sof-tgl-nocodec.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {TGLU-NOCODEC-IPC4}

[dmesg.txt](https://github.com/thesofproject/sof/files/13885091/dmesg.txt)
[mtrace.txt](https://github.com/thesofproject/sof/files/13885092/mtrace.txt)

keqiaozhang, 2024-01-12:
Confirmed that this is a sof regression and caused by:https://github.com/thesofproject/sof/commit/4bafdee66dd92a01def3b4820c33c2445d7bdbf9
```
commit 4bafdee66dd92a01def3b4820c33c2445d7bdbf9
Author: Andy Ross <andyross@google.com>
Date:   Tue Jan 2 16:43:44 2024 -0800

    ipc4: Use saved base_cfg_ext data in ipc_comp_connect()

    The code here never really worked right.  The module base config
    values weren't saved by the core layers, so querying the attribute
    returned nothing.  A few components have been modified with a whitebox
    workaround where they write directly to the basecfg_ext field in the
    module_config, but that's not really an API we should want to support.

    Get the values from the now-correctly-saved-and-parsed fields the
    module init layer has left for us, with considerable code savings and
    much less heap thrashing for the temporary copy.

    Signed-off-by: Andy Ross <andyross@google.com>

 src/ipc/ipc4/helper.c | 58 ++++++++++++++++++++++++++++++++++++++-------------
 1 file changed, 43 insertions(+), 15 deletions(-)
```

keqiaozhang, 2024-01-15:
@abonislawski @andyross FYI

keqiaozhang, 2024-01-16:
kernel log with IPC payload enabled:
```
[  442.168957] nvme nvme0: Shutdown timeout set to 10 seconds
[  442.170864] nvme nvme0: 8/0/0 default/read/poll queues
[  442.223272] snd_sof_intel_hda_common:cl_dsp_init: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x80000]=0x5000001 successful
[  442.223279] snd_sof_intel_hda_common:hda_cl_copy_fw: sof-audio-pci-intel-tgl 0000:00:1f.3: Code loader DMA starting
[  442.232276] snd_sof_intel_hda_common:hda_cl_copy_fw: sof-audio-pci-intel-tgl 0000:00:1f.3: Code loader DMA done, waiting for FW_ENTERED status
[  442.246264] snd_sof_intel_hda_common:hda_cl_copy_fw: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x80000]=0x5 successful
[  442.246269] snd_sof_intel_hda_common:hda_cl_copy_fw: sof-audio-pci-intel-tgl 0000:00:1f.3: Code loader FW_ENTERED status
[  442.246751] snd_sof_intel_hda_common:hda_dsp_stream_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x140000 successful
[  442.246757] snd_sof_intel_hda_common:hda_cl_copy_fw: sof-audio-pci-intel-tgl 0000:00:1f.3: Code loader DMA stopped
[  442.246761] snd_sof_intel_hda_common:hda_dsp_cl_boot_firmware: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware download successful, booting...
[  442.263106] r8152 4-1:1.0 enx9cebe8ae4a46: carrier on
[  442.279583] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx      : 0x1b080000|0x0: GLB_NOTIFICATION|FW_READY
[  442.279594] snd_sof:sof_set_fw_state: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state change: 3 -> 6
[  442.279605] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc rx done : 0x1b080000|0x0: GLB_NOTIFICATION|FW_READY
[  442.279628] snd_sof:snd_sof_run_firmware: sof-audio-pci-intel-tgl 0000:00:1f.3: firmware boot complete
[  442.279631] snd_sof:sof_set_fw_state: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state change: 6 -> 7
[  442.279644] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[  442.279651] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIW: 00000002 at  00000048
[  442.279662] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000004 at  00000044
[  442.279668] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIW: 00000000 at  00000044
[  442.284022] OOM killer enabled.
[  442.284025] Restarting tasks ... done.
[  442.287428] random: crng reseeded on system resumption
[  442.287554] snd_sof:sof_pcm_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 41 dir 0 cmd 1
[  442.287566] snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-tgl 0000:00:1f.3: trigger cmd: 1 state: 4
[  442.287579] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13000003|0x1: GLB_SET_PIPELINE_STATE [data size: 16]
[  442.287927] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000009|0x1: GLB_SET_PIPELINE_STATE
[  442.287938] sof-audio-pci-intel-tgl 0000:00:1f.3: FW reported error: 9 - Specified resource not found
[  442.287992] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc error for msg 0x13000003|0x1
[  442.288012] Message payload: 00000000: 00000003 00000003 00000001 00000000
[  442.288021] sof-audio-pci-intel-tgl 0000:00:1f.3: failed to pause all pipelines
[  442.288031] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[  442.288043]  Port0 Aux Playback: ASoC: trigger FE cmd: 1 failed: -22
[  442.288054] snd_sof:sof_pcm_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 41 dir 0 cmd 0
[  442.288063] snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-tgl 0000:00:1f.3: trigger cmd: 0 state: 3
[  442.290465] PM: suspend exit
[  442.391806] atkbd serio0: Failed to deactivate keyboard on isa0060/serio0
[  442.463139] ata1: SATA link down (SStatus 4 SControl 300)
[  442.767760] atkbd serio0: Failed to enable keyboard on isa0060/serio0
[  442.895535] snd_sof:sof_pcm_hw_free: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: free stream 41 dir 0
[  442.895565] snd_sof_intel_hda_common:hda_dsp_stream_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x180]=0x40000 successful
[  442.895574] snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-tgl 0000:00:1f.3: trigger cmd: 0 state: 2
[  442.895589] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13000002|0x1: GLB_SET_PIPELINE_STATE [data size: 16]
[  442.895996] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000009|0x1: GLB_SET_PIPELINE_STATE
[  442.896004] sof-audio-pci-intel-tgl 0000:00:1f.3: FW reported error: 9 - Specified resource not found
[  442.896042] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc error for msg 0x13000002|0x1
[  442.896054] Message payload: 00000000: 00000003 00000003 00000001 00000000
[  442.896059] sof-audio-pci-intel-tgl 0000:00:1f.3: failed to set final state 2 for all pipelines
[  442.896104] snd_sof:sof_ipc4_route_free: sof-audio-pci-intel-tgl 0000:00:1f.3: unbind modules host-copier.41.playback:0 -> gain.21.1:0
[  442.896116] snd_sof:sof_widget_free_unlocked: sof-audio-pci-intel-tgl 0000:00:1f.3: widget host-copier.41.playback freed
[  442.896122] snd_sof:sof_ipc4_route_free: sof-audio-pci-intel-tgl 0000:00:1f.3: unbind modules gain.21.1:0 -> mixin.21.1:0
[... cut ...]
[  445.412707] sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
[  445.412716] Message payload: 00000000: 00000001 00000000
[  445.412721] sof-audio-pci-intel-tgl 0000:00:1f.3: failed to disable target core: 0 for pipeline pipeline.2
[  445.412729] sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to free connected widgets
[  445.412744] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_free on 0000:00:1f.3: -110
[  445.412828] snd_sof:sof_pcm_close: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: close stream 41 dir 0

```

ujfalusi, 2024-01-16:
@keqiaozhang, do you have the full dmesg, the fragment is missing the DSP panic.

keqiaozhang, 2024-01-16:
> @keqiaozhang, do you have the full dmesg, the fragment is missing the DSP panic.

@ujfalusi full dmesg attached.
[dmesg.txt](https://github.com/thesofproject/sof/files/13948678/dmesg.txt)

lyakh, 2024-01-16:
Apparently hitting one of `__ASSERT()` statements in `sys_heap_free()`, which might either indicate a heap corruption or trying to free a wrong address

marc-hb, 2024-01-17:
This looks similar: https://sof-ci.01.org/sofpr/PR8743/build1905/devicetest/index.html?model=TGLU_RVP_NOCODEC-ipc4&testcase=check-suspend-resume-with-playback

kv2019i, 2024-01-24:
Potential fix https://github.com/thesofproject/sof/pull/8785/commits/70f3d541cb2994c189454c10102bb66cbd353682

marc-hb, 2024-01-25:
> Potential fix https://github.com/thesofproject/sof/commit/70f3d541cb2994c189454c10102bb66cbd353682

This was a private commit.

Just merged commit c3da4e3d623a9 reviewed in #8760.

marc-hb, 2024-01-25:
> but now an error with SET_DX on non-primary core is left and seen PR test results)

Which test results? Remember: links to test results are LOST after a force-pushed.

keqiaozhang, 2024-01-29:
Confirmed that this issue cannot be reproduced in CI.

closed 2024-01-29 by keqiaozhang