thesofproject/sof #8742: [BUG] performance test is failing

btian1, 2024-01-16:
**Describe the bug**
A clear and concise description of what the bug is.
CI build performance test is failing

**To Reproduce**
use local mtrace and dmesg file can reproduce as well

**Reproduction Rate**
100%.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
CI performance build test should pass.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: since ww03.1
    * SOF: since ww03.1
2) Name of the topology file
    * Topology: default
3) Name of the platform(s) on which the bug is observed.
    * Platform: mtl

[image]

btian1, 2024-01-16:
passed: https://sof-ci.ostc.intel.com/#/result/planresultdetail/36559
failed: https://sof-ci.ostc.intel.com/#/result/planresultdetail/36620

passed mtrace logging:
[image]

failed mtrace logging:
[image]

Current performance testing are requesting skip trace until find trace timestamp between [0, 1] as the start
of valid mtrace performance logging, due to current sof-test test case may collect some traces belonging to previous
test case due to mtrace is configured in deferred mode.

So once Rander aligned kernel and DSP timestamp, DSP timestamp only restart(from 0) on DSP power off/on, this means
only the first test case(or test case require DSP restart) have timestamp with [0,1].

with current option, until mtrace logging end, there may no timestamp between [0,1], so raise this exception and cause performance test failed.

Current solution is remove this option and python script will directly try to find logs include "comp_copy"  as valid module performance logging, then collect and calculate its performance data.

btian1, 2024-01-16:
PR submitted for review and discuss:
https://github.com/thesofproject/sof-test/pull/1150

marc-hb, 2024-01-16:
Please avoid screenshots: https://docs.zephyrproject.org/latest/develop/getting_started/index.html#use-copy-paste

marc-hb, 2024-01-16:
Thanks for collecting these logs and describing the change, this is great progress.

> So once Rander aligned kernel and DSP timestamp, DSP timestamp only restart(from 0) on DSP power off/on
> this means only the first test case(or test case require DSP restart) have timestamp with [0,1].

The initial timestamp sync message (`0x44000000|0x31400008`) is sent only every time the DSP boots. This sync message can be seen in all the logs you shared. So we can see that the DSP boots in all the logs you shared. Yet the timestamp does not always restarts from zero. So no: "DSP boot" != "timestamp from 0"; it's more complicated.

I investigated this a bit and I think I found the key fact and key difference that you have been constantly missing so far: booting _from IMR_.

> Current solution is remove this option and python script will directly try to find logs include "comp_copy" as valid module performance logging, then collect and calculate its performance data.

Very interesting, so you're saying that the `--skip-trace` option was ALREADY useless, even BEFORE Rander's timestamp change? This would be another, very important fact. @aiChaoSong was `--skip-trace` always useless from the beginning? If not, what other change made `--skip-trace` already useless before Rander's timestamp change?