thesofproject/sof #8766: [BUG] Failure to compile tools/logger/convert.c

dbaluta, 2024-01-22:
**Describe the bug**

Compiling the sof-logger for aarch64 we get the following warnings which gets translated into an error and compilation stops:

```
-- Build files have been written to: /opt/builds/OBNand/build/tmp/work/armv8a-poky-linux/sof-tools/2.8.0/git/tools/sof-tools
| [ 20%] Building C object logger/CMakeFiles/sof-logger.dir/convert.c.o
| /opt/builds/OBNand/build/tmp/work/armv8a-poky-linux/sof-tools/2.8.0/git/tools/logger/convert.c: In function 'convert':
| /opt/builds/OBNand/build/tmp/work/armv8a-poky-linux/sof-tools/2.8.0/git/tools/logger/convert.c:357:34: error: '%*s' directive output between 4294967264 and 4294967284 bytes exceeds 'INT_MAX' [-Werror=format-overflow=]
|   357 |                 fprintf(out_fd, "%*s(us)%*s  ", -ts_width, " TIMESTAMP", ts_width, "DELTA");
|       |                                  ^~~                       ~~~~~~~~~~~~
| In file included from /opt/builds/OBNand/build/tmp/work/armv8a-poky-linux/sof-tools/2.8.0/recipe-sysroot/usr/include/stdio.h:964,
|                  from /opt/builds/OBNand/build/tmp/work/armv8a-poky-linux/sof-tools/2.8.0/git/tools/logger/convert.h:13,
|                  from /opt/builds/OBNand/build/tmp/work/armv8a-poky-linux/sof-tools/2.8.0/git/tools/logger/convert.c:21:
| In function 'fprintf',
|     inlined from 'print_table_header' at /opt/builds/OBNand/build/tmp/work/armv8a-poky-linux/sof-tools/2.8.0/git/tools/logger/convert.c:357:3,
|     inlined from 'logger_read' at /opt/builds/OBNand/build/tmp/work/armv8a-poky-linux/sof-tools/2.8.0/git/tools/logger/convert.c:846:3,
|     inlined from 'convert' at /opt/builds/OBNand/build/tmp/work/armv8a-poky-linux/sof-tools/2.8.0/git/tools/logger/convert.c:1151:8:
| /opt/builds/OBNand/build/tmp/work/armv8a-poky-linux/sof-tools/2.8.0/recipe-sysroot/usr/include/bits/stdio2.h:79:10: note: '__fprintf_chk' output between 4294967283 and 4294967323 bytes
|    79 |   return __fprintf_chk (__stream, __USE_FORTIFY_LEVEL - 1, __fmt,
|       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
|    80 |                         __va_arg_pack ());
|       |                         ~~~~~~~~~~~~~~~~~
```
Compiler version:  aarch64-poky-linux-gcc (GCC) 13.2.0

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF: 16b53b423 (main) audio: host: reload DMA buffer on non-zero partial_size

It looks like the issues was introduced by:

```
commit 4c905f5234faf2cb32b0663dfd0b1c494b0a2a93
Author: Adrian Warecki <adrian.warecki@intel.com>
Date:   Mon Sep 18 15:31:28 2023 +0200

    logger: convert: Simplified printing of a tim

```

marc-hb, 2024-01-22:
> It looks like the issues was introduced by 4c905f5234faf2cb32b0663dfd0b1c494b0a2a93

Yes but before #8216 the code was worse :-)

Can you try something like this below? (untested)

```diff
--- a/tools/logger/convert.c
+++ b/tools/logger/convert.c
@@ -318,7 +318,7 @@ static double to_usecs(uint64_t time)
 }
 
 /** Justified timestamp width for printf format string */
-static unsigned int timestamp_width(unsigned int precision)
+static uint8_t timestamp_width(uint8_t precision)
 {
        /* 64bits yields less than 20 digits precision. As reported by
         * gcc 9.3, this avoids a very long precision causing snprintf()
@@ -352,7 +352,7 @@ static inline void print_table_header(void)
        }
 
        if (global_config->time_precision >= 0) {
-               const unsigned int ts_width = timestamp_width(global_config->time_precision);
+               const uint8_t ts_width = timestamp_width(global_config->time_precision);
 
                fprintf(out_fd, "%*s(us)%*s  ", -ts_width, " TIMESTAMP", ts_width, "DELTA");
        }
diff --git a/tools/logger/convert.h b/tools/logger/convert.h
index dcf8d0db06b7..8117e023b2f9 100644
--- a/tools/logger/convert.h
+++ b/tools/logger/convert.h
@@ -41,7 +41,7 @@ struct convert_config {
        int dump_ldc;
        int hide_location;
        int relative_timestamps;
-       int time_precision;
+       uint8_t time_precision;
        struct snd_sof_uids_header *uids_dict;
        struct snd_sof_logs_header *logs_header;
 };

```
cc: @softwarecki

dbaluta, 2024-01-23:
@marc-hb this works. Feel free to ACK this PR: https://github.com/thesofproject/sof/pull/8778

marc-hb, 2024-01-25:
Submitted to stable-v2.2 here: #8798

closed 2024-01-25 by marc-hb