thesofproject/sof #8779: [BUG] stream_start_offset set incorrectly leading to av sync errors

kv2019i, 2024-01-23:
**Describe the bug**
Branch from Linux SOF driver bug https://github.com/thesofproject/linux/issues/4781

The FW is not reporting correct start offset, which leads to incorrect delay reporting by the SOF driver to application. This can lead to video playback failures with applications using ALSA delay reporting.

**To Reproduce**
Play a video with mplayer and native ALSA backend:

- e.g. video from https://www.photojoseph.com/tips/2021/3/29/updated-march-2021-downloadable-audiovideo-sync-tests-2398-24-25-2997-50-5994
- play to system speaker: "mplayer -ao alsa:device=hw=0.2 Audio_Video_Sync_23,98_HEVC_1080p-by_PhotoJoseph.mov"

**Reproduction Rate**
50%

**Expected behavior**
Video should play reliably with accurate AV sync with applications using ALSA snd_pcm_delay().

**Impact**
High

**Environment**
kernel: https://github.com/thesofproject/linux/commit/66ee247636e52ea6ebb136edec9abed6891d7d6c (sof-dev)
SOF: 6c188298b958 (SOF main)
platform: MTL
topoplogy: sof-mtl-rt713-l0-rt1316-l12.tplg

kv2019i, 2024-01-23:
Potentially fixed by https://github.com/thesofproject/sof/pull/8756

RanderWang, 2024-01-24:
> Potentially fixed by #8756 UPDATE: I still get super large delay values reported when I tested this today with the above PR cherry-picked. So cannot confirm this fix yet.

@kv2019i  please help to review this PR: https://github.com/thesofproject/linux/pull/4791, thanks

kv2019i, 2024-01-31:
The FW part is merged, so let's close this and continue in https://github.com/thesofproject/linux/issues/4781

closed 2024-01-31 by kv2019i