thesofproject/sof #8782: [BUG] CONFIG_ZEPHYR_HAL_NXP_MODULE occured during tgl/mtl/lnl build

btian1, 2024-01-24:
**Describe the bug**
A clear and concise description of what the bug is.
please check build log, you will find:
[image]
or check .config file with build-mtl/zephyr/.config
CONFIG_ZEPHYR_HAL_NXP_MODULE=y

**To Reproduce**
check above

**Reproduction Rate**
100%.

**Expected behavior**
above definition should not happen in intel platform build.

**Impact**
not well evaluated.

lgirdwood, 2024-01-24:
adding @dbaluta - worth checking also if Intel Kconfigs are not being set for NXP builds. @btian1 any clue here ?

btian1, 2024-01-25:
below link generated file Kconfig.modules during pre-build:
https://github.com/zephyrproject-rtos/zephyr/blob/main/cmake/modules/zephyr_module.cmake#L60
and in generated file: 
build-mtl/Kconfig/Kconfig.modules

there is definition for:

menu "hal_nxp (/home/btian1/sof_mtl/modules/hal/nxp)"
osource "$(ZEPHYR_HAL_NXP_KCONFIG)"
config ZEPHYR_HAL_NXP_MODULE
        bool
        default y
endmenu

I just don't know how Kconfig.modules was generated with above link, @marc-hb , @kv2019i , could you guide here?

kv2019i, 2024-01-25:
Aa, I think this comes automatically for all modules we list in sof/west.yml. So we have hal_nxp there, so this config is added automatically in the build
https://docs.zephyrproject.org/latest/develop/modules.html

So I think this is harmless. Nothing gets build due to this module being there, it's jus a kconfig that tells the module is present in the work area. We have these two (and not all are used by all SOF builds):

../build-mtl/zephyr/.config:
....
CONFIG_ZEPHYR_SOF_MODULE=y
CONFIG_ZEPHYR_HAL_NXP_MODULE=y
CONFIG_ZEPHYR_XTENSA_MODULE=y
CONFIG_ZEPHYR_MBEDTLS_MODULE=y
CONFIG_ZEPHYR_MIPI_SYS_T_MODULE=y
CONFIG_ZEPHYR_TINYCRYPT_MODULE=y

iuliana-prodan, 2024-01-25:
As @kv2019i mentioned above the nxp_hal is imported, by default, for all platforms from SOF's west.yml - https://github.com/thesofproject/sof/blob/main/west.yml#L57

We can change this issue in a feature - import nxp_hal only for NXP platforms. 
But, TBH I didn't find a solution for this (or at least a few months ago when I checked this).

Maybe we can make it optional?
https://github.com/zephyrproject-rtos/zephyr/blob/main/submanifests/optional.yaml#L36
https://docs.zephyrproject.org/latest/contribute/external.html#integration-as-optional-modules
@kv2019i @marc-hb what do you think?
I'll check to see if is possible

marc-hb, 2024-01-25:
Making west modules optional would complicate CI and other things. I don't think we need to get into SOF package management yet (like https://github.com/zephyrproject-rtos/zephyr/pull/61505 just did), I think we have enough more important problems to solve.

Please just stop dumping gigabyte-big, entire SDKs in that repo :-)
https://github.com/zephyrproject-rtos/hal_nxp/pull/270#issuecomment-1872623423

PS: thank you @btian1 for paying attention to the build logs. Most people seem to miss warnings there :-) However I think this one can be closed.

marc-hb, 2024-01-25:
> above definition should not happen in intel platform build.

Try  `menuconfig` for an "Intel build" of the Linux kernel and see how many non-Intel things you will find in Kconfig ;-)

btian1, 2024-01-26:
it is not only have a config during build: CONFIG_ZEPHYR_HAL_NXP_MODULE

it have include path involved, this is also acceptable for tgl/mtl/lnl build? check build.ninja:
I am afraid it will eventually cause conflicts when build:

**ZEPHYR_HAL_NXP_MODULE_DIR=/home/btian1/sof_mtl/modules/hal/nxp ZEPHYR_HAL_NXP_KCONFIG=/home/btian1/sof_mtl/zephyr/modules/hal_nxp/Kconfig_**** 

../build-mtl/build.ninja:241:  COMMAND = cd /home/btian1/sof_mtl/build-mtl/zephyr/kconfig && /usr/bin/cmake -E env ZEPHYR_BASE=/home/btian1/sof_mtl/zephyr PYTHON_EXECUTABLE=/usr/bin/python3.8 srctree=/home/btian1/sof_mtl/zephyr KERNELVERSION=0x3056300 APPVERSION= CONFIG_=CONFIG_ KCONFIG_CONFIG=/home/btian1/sof_mtl/build-mtl/zephyr/.config ARCH=xtensa ARCH_DIR=/home/btian1/sof_mtl/zephyr/arch BOARD_DIR=/home/btian1/sof_mtl/zephyr/boards/xtensa/intel_adsp_ace15_mtpm BOARD_REVISION= KCONFIG_BINARY_DIR=/home/btian1/sof_mtl/build-mtl/Kconfig APPLICATION_SOURCE_DIR=/home/btian1/sof_mtl/sof/app ZEPHYR_TOOLCHAIN_VARIANT=xt-clang TOOLCHAIN_KCONFIG_DIR=/home/btian1/sof_mtl/zephyr/cmake/toolchain/xt-clang TOOLCHAIN_HAS_NEWLIB=n TOOLCHAIN_HAS_PICOLIBC=n EDT_PICKLE=/home/btian1/sof_mtl/build-mtl/zephyr/edt.pickle ZEPHYR_SOF_MODULE_DIR=/home/btian1/sof_mtl/sof **_### **ZEPHYR_HAL_NXP_MODULE_DIR=/home/btian1/sof_mtl/modules/hal/nxp ZEPHYR_HAL_NXP_KCONFIG=/home/btian1/sof_mtl/zephyr/modules/hal_nxp/Kconfig_**** ZEPHYR_XTENSA_MODULE_DIR=/home/btian1/sof_mtl/modules/hal/xtensa ZEPHYR_MBEDTLS_MODULE_DIR=/home/btian1/sof_mtl/modules/crypto/mbedtls ZEPHYR_MBEDTLS_KCONFIG=/home/btian1/sof_mtl/zephyr/modules/mbedtls/Kconfig ZEPHYR_MIPI_SYS_T_MODULE_DIR=/home/btian1/sof_mtl/modules/debug/mipi-sys-t ZEPHYR_TINYCRYPT_MODULE_DIR=/home/btian1/sof_mtl/modules/crypto/tinycrypt SHIELD_AS_LIST= DTS_POST_CPP=/home/btian1/sof_mtl/build-mtl/zephyr/zephyr.dts.pre DTS_ROOT_BINDINGS=/home/btian1/sof_mtl/zephyr/dts/bindings /usr/bin/python3.8 /home/btian1/sof_mtl/zephyr/scripts/kconfig/guiconfig.py /home/btian1/sof_mtl/zephyr/Kconfig

marc-hb, 2024-01-27:
> it have include path involved

Which include path did you see?

btian1, 2024-01-29:
ZEPHYR_HAL_NXP_MODULE_DIR=/home/btian1/sof_mtl/modules/hal/nxp ZEPHYR_HAL_NXP_KCONFIG=/home/btian1/sof_mtl/zephyr/modules/hal_nxp/Kconfig_**

marc-hb, 2024-01-29:
These don't look like include paths.

An "include path" is telling the compiler where to look for .h files.

marc-hb, 2024-01-31:
Please re-open if you find a real issue (= outside Kconfig)

closed 2024-01-31 by marc-hb