thesofproject/sof #8801: [BUG] FW reported error: 7 - Unsupported operation requested when testing Bluetooth offload tplg

keqiaozhang, 2024-01-26:
**Describe the bug**
This issue only happens with Bluetooth tplg when running playback/capture at second time and it's a regression and caused by:
https://github.com/thesofproject/sof/commit/08cf3d2232e4190898670149f8a5acb82e6a1caf
```
commit 08cf3d2232e4190898670149f8a5acb82e6a1caf
Author: Rander Wang <rander.wang@intel.com>
Date:   Thu Jan 18 12:56:29 2024 +0800

    Revert "ipc4: relax the IPC timeout checks and be nicer to other threads"

```
cc: @RanderWang @kv2019i 

**demsg**
```
[ 1011.176592] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x13000003|0x1: GLB_SET_PIPELINE_STATE [data size: 12]
[ 1011.177008] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x33000000|0x1: GLB_SET_PIPELINE_STATE
[ 1011.177027] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x13000003|0x1: GLB_SET_PIPELINE_STATE [data size: 12]
[ 1011.177031] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x13000004|0x1: GLB_SET_PIPELINE_STATE [data size: 12]
[ 1011.178993] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x33000007|0x1: GLB_SET_PIPELINE_STATE
[ 1011.178999] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: FW reported error: 7 - Unsupported operation requested
[ 1011.179021] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc error for msg 0x13000004|0x1
[ 1011.179032] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to set final state 4 for all pipelines
[ 1011.179036] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[ 1011.179042] kernel:  Port2: ASoC: trigger FE cmd: 1 failed: -22
```

**mtrace**
```
[ 1011.177123] <inf> copier: copier_prepare: comp:1 0x10004 copier_prepare()
[ 1011.177133] <err> dma_dw_common: dw_dma_config: dw_dma_config: dma dma@7c000 channel 0 must be inactive to reconfigure, currently 2
[ 1011.177140] <inf> component: comp_set_state: comp:1 0x10004 comp_set_state(), state already set to 1
[ 1011.177146] <err> module_adapter: module_prepare: comp:1 0x10004 module_prepare() error -16: module specific prepare failed, comp_id 65540
[ 1011.177151] <err> module_adapter: module_adapter_prepare: comp:1 0x10004 module_adapter_prepare() error fffffff0: module prepare failed
[ 1011.177158] <err> pipe: pipeline_prepare: pipe:1 0x0 pipeline_prepare(): ret = -16, dev->comp.id = 65540
[ 1011.177163] <err> ipc: ipc4_pcm_params: ipc: pipe 1 comp 0 prepare failed -16
[ 1011.178515] <err> dma_dw_common: dw_dma_stop: dw_dma_stop: dma 0 channel drain time out
[ 1011.178563] <inf> component: comp_set_state: comp:1 0x10004 comp_set_state(), state already set to 1
[ 1011.178570] <err> ipc: ipc_cmd: ipc4: FW_GEN_MSG failed with err 7
[ 1011.680990] <inf> ipc: ipc_cmd: rx	: 0x13000002|0x1
[ 1011.681611] <inf> ipc: ipc_cmd: rx	: 0x46000004|0x10004
[ 1011.682121] <inf> ipc: ipc_cmd: rx	: 0x12000000|0x0
[ 1011.682136] <inf> dma: dma_put: dma_put(), dma = 0x400f6060, sref = 0
[ 1011.682768] <inf> ipc: ipc_cmd: rx	: 0x12010000|0x0
[ 1011.682798] <inf> dma: dma_put: dma_put(), dma = 0x400f5f70, sref = 0
[ 1011.682811] <inf> dai_intel_ssp: dai_ssp_pm_runtime_dis_ssp_power: dai_ssp_pm_runtime_dis_ssp_power index 2
[ 1011.682818] <inf> dai_intel_ssp: dai_ssp_pm_runtime_dis_ssp_power: dai_ssp_pm_runtime_dis_ssp_power I2SLCT
```

**To Reproduce**
~/sof-test/test-case/check-playback.sh -d 1 -l 100 -r 1

**Reproduction Rate**
100%

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF: {08cf3d2232e4190898670149f8a5acb82e6a1caf}
2) Name of the topology file
    * Topology: {sof-nocodec-bt-mtl-lbm.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {MTL-NOCODEC with BT tplg}

[dmesg.txt](https://github.com/thesofproject/sof/files/14059768/dmesg.txt)

[mtrace.txt](https://github.com/thesofproject/sof/files/14059770/mtrace.txt)

marc-hb, 2024-01-27:
This regression makes ALMOST ALL sof-tests fail in this configuration.

The very first, internal daily test run that failed is 37403 

Start Time: 2024-01-25 21:10:21 UTC End Time: 2024-01-26 00:46:42 UTC
Linux Branch: [topic/sof-dev](https://github.com/thesofproject/linux/tree/topic/sof-dev)
Linux Commit: [d23a5e9b58f4](https://github.com/thesofproject/linux/commit/d23a5e9b58f4fc80da2a30c6a982cb0e093ffea9)
KConfig Branch: [master](https://github.com/thesofproject/kconfig/tree/master)
KConfig Commit: [135c26e6f4bf](https://github.com/thesofproject/kconfig/commit/135c26e6f4bf733e39fbc6c94f2efe6fd7418f8a)

SOF Branch: [main](https://github.com/thesofproject/sof/tree/main)
SOF Commit: [08cf3d2232e4](https://github.com/thesofproject/sof/commit/08cf3d2232e4190898670149f8a5acb82e6a1caf)
Zephyr Commit: [d7af6f371034](https://github.com/zephyrproject-rtos/zephyr/commit/d7af6f371034f31b9440b27c694b0be3c87491d3)

Last successful was 37340, Jan 24th

Start Time: 2024-01-24 21:11:09 UTC End Time: 2024-01-25 00:38:23 UTC
Linux Branch: [topic/sof-dev](https://github.com/thesofproject/linux/tree/topic/sof-dev)
Linux Commit: [5344645e226a](https://github.com/thesofproject/linux/commit/5344645e226acaa9abe6ca1ac6517e1c4fbf1aba)
KConfig Branch: [master](https://github.com/thesofproject/kconfig/tree/master)
KConfig Commit: [135c26e6f4bf](https://github.com/thesofproject/kconfig/commit/135c26e6f4bf733e39fbc6c94f2efe6fd7418f8a)

SOF Branch: [main](https://github.com/thesofproject/sof/tree/main)
SOF Commit: [c27eda7edb44](https://github.com/thesofproject/sof/commit/c27eda7edb448a62a4c5ed74f9987bb64f320b5d)
Zephyr Commit: [d7af6f371034](https://github.com/zephyrproject-rtos/zephyr/commit/d7af6f371034f31b9440b27c694b0be3c87491d3)

RanderWang, 2024-02-20:
@keqiaozhang   I can't reproduce bug on main branch now,  can you help to validate it ? thanks!
This issue was caused by DMA stop issue which should be fixed in zephyr size.

keqiaozhang, 2024-02-20:
Confirmed that this issue cannot be reproduced in CI.

closed 2024-02-20 by keqiaozhang

marc-hb, 2024-02-20:
> This issue was caused by DMA stop issue which should be fixed on zephyr size.

So what was the fix exactly?