thesofproject/sof #8809: [BUG]audio glitches on LNL-SDW

keqiaozhang, 2024-01-29:
**Describe the bug**
The JD issue has been fixed on LNL-SDW platforms, so now we are able to check the audio quality on LNL-SDW. 
Now I observed the audio glitches on both playback and capture, but the reproduction rate is low, not easy to reproduce manually.
Here are the failures we captured in CI daily test.

On playback:
[image]
There're a lot of xrun warning during the test.
```
2024-01-28 22:11:39 UTC [REMOTE_INFO] Starting /usr/local/bin/mtrace-reader.py >& /home/ubuntu/sof-test/logs/check-alsabat/2024-01-28-22:11:38-10012/mtrace.txt &
2024-01-28 22:11:39 UTC [REMOTE_INFO] check the PCMs before alsabat test
2024-01-28 22:11:41 UTC [REMOTE_COMMAND] alsabat -Phw:sofsoundwire,0 --standalone -n 240000 -r 48000 -c 2 -f S16_LE -F 821 -k 2.1
2024-01-28 22:11:42 UTC [REMOTE_COMMAND] alsabat -Chw:CODEC,0 -c 2 -r 48000 -f S16_LE -F 821 -k 2.1
Underrun: Broken pipe(-32)
Underrun: Broken pipe(-32)
Underrun: Broken pipe(-32)
Underrun: Broken pipe(-32)
Underrun: Broken pipe(-32)
Underrun: Broken pipe(-32)
Underrun: Broken pipe(-32)
Underrun: Broken pipe(-32)
Underrun: Broken pipe(-32)
Underrun: Broken pipe(-32)
Underrun: Broken pipe(-32)
Underrun: Broken pipe(-32)
Underrun: Broken pipe(-32)
Underrun: Broken pipe(-32)
```

On capture:
[image]

**To Reproduce**

Playback test:~/sof-test/test-case/check-alsabat.sh -p hw:sofsoundwire,0 -c hw:CODEC,0 -C 2 -F 821

Capture test:~/sof-test/test-case/check-alsabat.sh -c hw:sofsoundwire,1 -p hw:CODEC,0 -C 2 -F 821

**Reproduction Rate**
Low, less than 10%.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {[d23a5e9b58f4](https://github.com/thesofproject/linux/commit/d23a5e9b58f4fc80da2a30c6a982cb0e093ffea9)}
    * SOF: {[3930270e3d9b](https://github.com/thesofproject/sof/commit/3930270e3d9b5fc57c11dd793a4be14747094ba2)}
2) Name of the topology file
    * Topology: {sof-lnl-rt711-l0-rt1316-l23-rt714-l1.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {LNL-SDW w/ Gen4.1 AIOC base board}

[wav-file.gz](https://github.com/thesofproject/sof/files/14080424/wav-file.gz)

plbossart, 2024-01-29:
probably the same as https://github.com/thesofproject/linux/issues/4754, it's not worth investigating audio quality issues until fw-based aggregation is merged in firmware and kernel

lgirdwood, 2024-01-31:
> probably the same as [thesofproject/linux#4754](https://github.com/thesofproject/linux/issues/4754), it's not worth investigating audio quality issues until fw-based aggregation is merged in firmware and kernel

Ok, lets wait until aggregation is fully ready. I've put in v2.10 for the moment, can be moved to v2.9 if needed.

plbossart, 2024-05-28:
I don't think we can reach any conclusions, we still have FAILs in alsa-bat, and I don't get if this is still related to the jack setup or now.

bardliao, 2024-05-29:
@plbossart @lgirdwood I checked some recent daily test results, check-alsabat-headset-playback seems pass consistently and some of the check-alsabat-headset-capture are failed. But they are all because of the signal is too low not like this issue's description.
And the issue is tracking at https://github.com/thesofproject/sof/issues/9165.

But the reproduction rate of this issue is low. So, I don't know if it is fixed or just not reproduced recently.

lgirdwood, 2024-06-10:
@fredoh9 @marc-hb @ssavati anyone seen this on daily test recently ? If not we can close. This is different from the bank switch and missing IRQ.

ujfalusi, 2024-06-10:
I cannot find anything like this, but we have at least two corrupted alsabat:
https://sof-ci.ostc.intel.com/#/result/planresultdetail/41949?model=LNLM_SDW_AIOC&testcase=check-alsabat-headset-playback-599
https://sof-ci.ostc.intel.com/#/result/planresultdetail/41810?model=LNLM_SDW_AIOC&testcase=check-alsabat-headset-playback-599

marc-hb, 2024-06-10:
I've seen this "Underrun: Broken pipe" recently. Also, alsabat (which is our only "real" audio test) is still failing left and right on LNL.

marc-hb, 2024-06-13:
> I've seen this "Underrun: Broken pipe" recently.

One just now:
- https://github.com/thesofproject/sof/issues/9164#issuecomment-2164285904

plbossart, 2024-06-13:
> One just now:
> 
> * [LNL HDA alsabat capture failed, "Peak freq too low" #9164 (comment)]

Different issue. We're tracking SoundWire issues here, the HDA alsabat stuff is a different problem relying on a different topology and a different physical interface/timing.

marc-hb, 2024-06-18:
Sorry my bad, went too fast. To make amend, here's the SDW, "signal too weak" failure of the day:

https://sof-ci.01.org/softestpr/PR931/build518/devicetest/index.html?model=LNLM_SDW_AIOC&testcase=check-alsabat-headset-capture-599

marc-hb, 2024-06-18:
... and still wrong, there is yet another issue about that: #9165. Really sorry, I'm losing track

kv2019i, 2024-06-28:
Reproduction rate low in past week, continue with P2 and assign to v2.11.

marc-hb, 2024-07-09:
Recent "Underrun: Broken pipe(-32)" in https://sof-ci.01.org/softestpr/PR1217/build594/devicetest/index.html?model=LNLM_SDW_AIOC&testcase=check-alsabat-headset-playback-599

kv2019i, 2024-09-09:
Reviewing a week worth of daily test plans with this configurations (starting  2024-09-02), this test is passing all the time. It is likely https://github.com/zephyrproject-rtos/zephyr/pull/77805 fix was responsible for the occasional 1ms glitches seen in this test and those are now gone.

kv2019i, 2024-09-10:
Murphy's law, the next daily plan shows a fail again, 2024-09-09 run has a fail. Console has a overrun and the captured bat.wav look like this:

[image]

... rest of file is clean. Reopening.

lrudyX, 2024-09-13:
@kv2019i please create new ticket for this problem

marc-hb, 2024-09-13:
@lrudyX can you summarize how it is a different issue now?

kv2019i, 2024-09-13:
@lrudyX I think this is exactly the original problem (quoting the original bug report from January):

```
2024-01-28 22:11:39 UTC [REMOTE_INFO] Starting /usr/local/bin/mtrace-reader.py >& /home/ubuntu/sof-test/logs/check-alsabat/2024-01-28-22:11:38-10012/mtrace.txt &
2024-01-28 22:11:39 UTC [REMOTE_INFO] check the PCMs before alsabat test
2024-01-28 22:11:41 UTC [REMOTE_COMMAND] alsabat -Phw:sofsoundwire,0 --standalone -n 240000 -r 48000 -c 2 -f S16_LE -F 821 -k 2.1
2024-01-28 22:11:42 UTC [REMOTE_COMMAND] alsabat -Chw:CODEC,0 -c 2 -r 48000 -f S16_LE -F 821 -k 2.1
Underrun: Broken pipe(-32)
Underrun: Broken pipe(-32)
Underrun: Broken pipe(-32)
Underrun: Broken pipe(-32)
```

We still see the underruns on certain DUT configurations.

kv2019i, 2024-12-13:
Not seen in PR/daily tests for a week, closing.

closed 2024-12-13 by kv2019i