thesofproject/sof #8816: [BUG] DSP panic on LNL-SDW

keqiaozhang, 2024-01-30:
**Describe the bug**
This is a sof regression, it happens when checking playback/capture. It caused by https://github.com/thesofproject/sof/commit/a1deb74da048e3d258445231ffb9ff07dd976ab5.
```
commit a1deb74da048e3d258445231ffb9ff07dd976ab5
Author: Ievgen Ganakov <ievgen.ganakov@intel.com>
Date:   Fri Sep 8 12:19:56 2023 +0200

    copier: dai: use HD-A dai to configure soundwire on LNL

    Since LNL soundwire uses HD-A DMA to transfer data. Add LNL specific
    configuration to select HD-A DMA in case of SoundWire audio interface.
    Refactor copier dai code for sndw/alh node id type.

    Add support for sndw link aggregation mode for LNL platform based on
    DMA config being sent during Copier Init Instance IPC.

    Signed-off-by: Ievgen Ganakov <ievgen.ganakov@intel.com>

 src/audio/copier/copier_dai.c | 145 +++++++++++++++++++++++++++++-------------
 src/include/ipc4/alh.h        |   8 +++
 src/ipc/ipc4/dai.c            |  24 +++++++
 3 files changed, 134 insertions(+), 43 deletions(-)

```

**dmesg**
```
[14084.303523] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx      : 0x40010004|0x1005d: MOD_INIT_INSTANCE [data size: 372]
[14084.306402] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc rx      : 0x1b0a0000|0x0: GLB_NOTIFICATION|EXCEPTION_CAUGHT
[14084.306427] sof-audio-pci-intel-lnl 0000:00:1f.3: ------------[ DSP dump start ]------------
[14084.312444] sof-audio-pci-intel-lnl 0000:00:1f.3: DSP panic!
[14084.314832] sof-audio-pci-intel-lnl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[14084.318097] sof-audio-pci-intel-lnl 0000:00:1f.3: 0x50000005: module: ROM_EXT, state: FW_ENTERED, running
[14084.322133] sof-audio-pci-intel-lnl 0000:00:1f.3: Firmware state: 0x5, status/error code: 0x0
[14084.326463] snd_sof:sof_ipc4_find_debug_slot_offset_by_type: sof-audio-pci-intel-lnl 0000:00:1f.3: Slot type 0x4c455400 is not available in debug window
[14084.326466] sof-audio-pci-intel-lnl 0000:00:1f.3: ------------[ DSP dump end ]------------
[14084.330585] snd_sof:sof_set_fw_state: sof-audio-pci-intel-lnl 0000:00:1f.3: fw_state change: 7 -> 8
[14084.330602] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc rx done : 0x1b0a0000|0x0: GLB_NOTIFICATION|EXCEPTION_CAUGHT
[14084.805013] sof-audio-pci-intel-lnl 0000:00:1f.3: ipc timed out for 0x40010004|0x1005d
[14084.810195] sof-audio-pci-intel-lnl 0000:00:1f.3: Attempting to prevent DSP from entering D3 state to preserve context
[14084.810197] sof-audio-pci-intel-lnl 0000:00:1f.3: ------------[ IPC dump start ]------------
[14084.812235] sof-audio-pci-intel-lnl 0000:00:1f.3: Host IPC initiator: 0xc0010004|0x1005d|0x0, target: 0x1b0a0000|0x0|0x0, ctl: 0x3
[14084.815121] sof-audio-pci-intel-lnl 0000:00:1f.3: ------------[ IPC dump end ]------------
[14084.817128] sof-audio-pci-intel-lnl 0000:00:1f.3: IPC timeout
[14084.818583] Message payload: 00000000: 00000000 00000180 00000180 00000001
[14084.818585] Message payload: 00000010: 0000bb80 00000020 ffffff10 00000001
[14084.818586] Message payload: 00000020: 00000000 00002002 0000bb80 00000020
[14084.818588] Message payload: 00000030: ffffff10 00000001 00000000 00002002
[14084.818589] Message payload: 00000040: 00000000 00001000 00000300 00000049
[14084.818591] Message payload: 00000050: 00000000 00000001 00001000 00000003
[14084.818592] Message payload: 00000060: 00001000 00000003 00000000 00000000
[14084.818593] Message payload: 00000070: 00000000 00000000 00000000 00000000
[14084.818595] Message payload: 00000080: 00000000 00000000 00000000 00000000
[14084.818596] Message payload: 00000090: 00000000 00000000 00000000 00000000
[14084.818597] Message payload: 000000a0: 00000000 00000000 00000000 00000000
[14084.818599] Message payload: 000000b0: 00000000 00000000 00000000 00000000
[14084.818600] Message payload: 000000c0: 00000000 00000000 00000000 00000000
[14084.818602] Message payload: 000000d0: 00000000 00000000 00001000 00000094
[14084.818603] Message payload: 000000e0: 00000101 00000000 00000001 00000000
[14084.818604] Message payload: 000000f0: 00000000 00000000 00000000 00000000
[14084.818606] Message payload: 00000100: 00000000 00000000 00000000 00000000
[14084.818607] Message payload: 00000110: 00000000 00000000 00000000 00000000
[14084.818609] Message payload: 00000120: 00000000 00000000 00000000 00000000
[14084.818610] Message payload: 00000130: 00000000 00000000 00000000 00000000
[14084.818611] Message payload: 00000140: 00000000 00000000 00000000 00000000
[14084.818613] Message payload: 00000150: 00000000 00000000 00000000 00000000
[14084.818614] Message payload: 00000160: 00000000 00000000 00000000 00000000
[14084.818615] Message payload: 00000170: 00000000
[14084.818617] sof-audio-pci-intel-lnl 0000:00:1f.3: failed to create module alh-copier.SDW2-Playback.0
[14084.820940] snd_sof:sof_widget_free_unlocked: sof-audio-pci-intel-lnl 0000:00:1f.3: widget gain.21.1 freed
[14084.820945] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx      : 0x12010000|0x0: GLB_DELETE_PIPELINE
[14084.820978] sof-audio-pci-intel-lnl 0000:00:1f.3: ipc4_tx_msg_unlocked: ipc message send for 0x12010000|0x0 failed: -19
[14084.823680] sof-audio-pci-intel-lnl 0000:00:1f.3: failed to free pipeline widget pipeline.21
[14084.825780] snd_sof:sof_widget_free_unlocked: sof-audio-pci-intel-lnl 0000:00:1f.3: widget mixin.20.1 freed
[14084.825782] snd_sof:sof_widget_free_unlocked: sof-audio-pci-intel-lnl 0000:00:1f.3: widget gain.20.1 freed
[14084.825784] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx      : 0x12000000|0x0: GLB_DELETE_PIPELINE
[14084.825786] sof-audio-pci-intel-lnl 0000:00:1f.3: ipc4_tx_msg_unlocked: ipc message send for 0x12000000|0x0 failed: -19
[14084.828450] sof-audio-pci-intel-lnl 0000:00:1f.3: failed to free pipeline widget pipeline.20
[14084.830521] sof-audio-pci-intel-lnl 0000:00:1f.3: Failed to set up connected widgets
[14084.832431] snd_sof:is_virtual_widget: sof-audio-pci-intel-lnl 0000:00:1f.3: sof_unprepare_widgets_in_path: virtual.sdw-amp is a virtual widget
[14084.832434] snd_sof:is_virtual_widget: sof-audio-pci-intel-lnl 0000:00:1f.3: sof_walk_widgets_in_order: virtual.sdw-amp is a virtual widget
[14084.832435] sof-audio-pci-intel-lnl 0000:00:1f.3: error: failed widget list set up for pcm 2 dir 0
[14084.834659] sof-audio-pci-intel-lnl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -110
[14084.837270]  Speaker: ASoC: error at __soc_pcm_hw_params on Speaker: -110
[14084.838922] sdw_deprepare_stream: subdevice #0-Playback: inconsistent state state 1
[14084.840840]  Speaker: ASoC: error at dpcm_fe_dai_hw_params on Speaker: -110
[14084.842520] snd_sof:sof_pcm_hw_free: sof-audio-pci-intel-lnl 0000:00:1f.3: pcm: free stream 2 dir 0
[14084.842592] sdw_deprepare_stream: subdevice #0-Playback: inconsistent state state 6
[14084.844918] snd_sof:sof_pcm_close: sof-audio-pci-intel-lnl 0000:00:1f.3: pcm: close stream 2 dir 0
[14088.315615] soundwire_bus:sdw_bus_wait_for_clk_prep_deprep: soundwire sdw-master-0-3: clock stop prepare done slave:15
[14088.470978] soundwire_bus:sdw_bus_wait_for_clk_prep_deprep: soundwire sdw-master-0-2: clock stop prepare done slave:15
```

**To Reproduce**
aplay   -Dhw:0,0 -r 48000 -c 2 -f S16_LE -d 3 /dev/zero -v -q

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {sof-dev}
    * SOF: {[a1deb74](https://github.com/thesofproject/sof/commit/a1deb74da048e3d258445231ffb9ff07dd976ab5)}
2) Name of the topology file
    * Topology: {sof-lnl-rt711-l0-rt1316-l23-rt714-l1.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {LNL-SDW-AIOC}

plbossart, 2024-01-30:
probably needs https://github.com/thesofproject/linux/pull/4790 at least for playback (capture still open with ch_mask discussion)

bardliao, 2024-01-30:
It is likely https://github.com/thesofproject/linux/pull/4790/commits/d394dc9d08fe1e3886c562df813a483d61807c5b. Let me submit a separate PR for this

bardliao, 2024-01-30:
> It is likely [thesofproject/linux@d394dc9](https://github.com/thesofproject/linux/commit/d394dc9d08fe1e3886c562df813a483d61807c5b). Let me submit a separate PR for this

Hmm, I was wrong. [thesofproject/linux@d394dc9](https://github.com/thesofproject/linux/commit/d394dc9d08fe1e3886c562df813a483d61807c5b) is not sufficient. You will need https://github.com/thesofproject/linux/pull/4790.

mengdonglin, 2024-01-31:
Confirmed this issue cannot be reproduced with kernel PR to support link aggregation https://github.com/thesofproject/linux/pull/4790.

wszypelt, 2024-02-09:
@keqiaozhang does regression still occur? Can we closed this issue?

keqiaozhang, 2024-02-20:
This issue cannot be reproduced in CI. Closing it.

closed 2024-02-20 by keqiaozhang