thesofproject/sof #8908: [BUG] DSP panic with Zephyr upstream main

kv2019i, 2024-03-06:
**Describe the bug**
This is a follow-up to https://github.com/thesofproject/sof/issues/8818 and shadows Zephyr issue https://github.com/zephyrproject-rtos/zephyr/issues/69807

Let's keep this open until we can remove the revert from SOF side.

**To Reproduce**
Build SOF with latest Zephyr main.

**Reproduction Rate**
100%

**Expected behavior**
Tst case passes

**Impact**
DSP panic, audio lost

**Environment**
See https://github.com/thesofproject/sof/issues/8818 and https://sof-ci.01.org/sofpr/PR8901/build3107/devicetest/index.html?model=MTLP_RVP_NOCODEC&testcase=check-playback-10sec

**Screenshots or console output**
https://sof-ci.01.org/sofpr/PR8901/build3107/devicetest/index.html?model=MTLP_RVP_NOCODEC&testcase=check-playback-10sec

marc-hb, 2024-03-08:
Observed with CAVS25 too:
https://sof-ci.01.org/sofpr/PR8781/build3249/devicetest/index.html?model=TGLU_RVP_NOCODEC-ipc4&testcase=check-playback-3times

Funny enough, in the same test run ACE did not crash:
https://sof-ci.01.org/sofpr/PR8781/build3248/devicetest/index.html

So it is not deterministic indeed.

kv2019i, 2024-03-11:
Two fix candidates under work:
- https://github.com/zephyrproject-rtos/zephyr/pull/69937
- https://github.com/zephyrproject-rtos/zephyr/pull/68493

kv2019i, 2024-03-14:
Closed via https://github.com/thesofproject/sof/pull/8938

closed 2024-03-14 by kv2019i

marc-hb, 2024-03-15:
The next panic is #8942