thesofproject/sof #8920: Re-introducing state change in dai_comp_trigger_internal() breaks some tests from Intel CI

LaurentiuM1234, 2024-03-07:
Attempting to bring back the state transition logic in `dai-zephyr`'s `dai_comp_trigger_internal()` that was initially removed by 6ffb0526910e0d3410f3a9dd5b68457d359cc8f4 seems to break some tests from the Intel CI. The logic is needed for NXP platforms and is re-introduced by https://github.com/thesofproject/sof/pull/8859 via 59d05a318982142e862374dccd481b449ed33e88.

As a workaround, the state transition logic has been made IPC3-specific but it might be worth looking into this and trying to make the code IPC version-agnostic if possible.

Links to the aforementioned failing CI builds:

-  https://sof-ci.01.org/sofpr/PR8859/build3069/devicetest/index.html
- https://sof-ci.01.org/sofpr/PR8859/build3070/devicetest/index.html
- https://sof-ci.01.org/sof-pr-viewer/#/build/PR8859/build13666634 (internal build)

lgirdwood, 2024-03-08:
@LaurentiuM1234 probably the best thing todo is to break down the changes into small bi-sectable patches and upstream them incrementally. This would allow more specific targeting when the issue is introduced.

lgirdwood, 2024-05-28:
@LaurentiuM1234 are we good to close now that #8859 is merged or more still todo ?

LaurentiuM1234, 2024-05-28:
> @LaurentiuM1234 are we good to close now that #8859 is merged or more still todo ?

This issue was actually created by the introduction of https://github.com/thesofproject/sof/pull/8859 (1bf6b8c04a686d96d003b1b134a71e0e5860d918). I think the question still stands: why does that code bit break Intel test(s) if made IPC version-agnostic?

kv2019i, 2024-06-05:
Stable branched for 2.10, nobody assigned to work on this, so moving this to 2.11.

kv2019i, 2024-09-05:
No owner -> v2.12

kv2019i, 2025-01-03:
No owner still -> v2.13