thesofproject/sof #8931: [BUG][MTL] - with mtl-008 and main branch FW - video is not smooth with google assistant enabled

udaymb, 2024-03-11:
**Describe the bug**
With mtl-008 and main branch FW - video is not smooth with google assistant enabled

**To Reproduce**

- Boot the chromebook and log in to credentials.
- In Settings, search for “Google assistant”.
- In Google assistant, set “Hey Google/Okay Google” to “On (Recommended)”.
- Retrain the voice to perform voice search.
- Play the video from youtube or Local player.

**Reproduction Rate**
How often does the issue happen ? i.e. 1/10 (once in ten attempts), 1/1000 or all the time.
Does the reproduction rate vary with any other configuration or user action, if so please describe and show the new reproduction rate.
Always

**Expected behavior**
A clear and concise description of what you expected to happen.
Video should run smooth with google assistant enabled

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
Effects 

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA} - https://chromium.googlesource.com/chromiumos/third_party/kernel/+log/refs/heads/release-R123-15786.B-chromeos-6.1 (https://chromium.googlesource.com/chromiumos/third_party/kernel/+/d4f476208c715d8dee7fc7f28134d150f507af6f)
    * SOF: {SHA} - https://github.com/thesofproject/sof/tree/releases/mtl/v8.0 (https://github.com/thesofproject/sof/tree/mtl-008-drop-stable) - top 5 commits
aa56343a9612 (tag: releases/mtl/v8.0, origin/mtl-008-drop-stable,
2886ee383d05 multicore: don't enable timer interrupts on secondary cores
8f12c9d8cdcb Revert "idc: zephyr: fix race between IDC p4wq worker and new msg submission"
49af52610d96 app: perf_overlay: disable CONFIG_ADSP_IDLE_CLOCK_GATING
d533db7a10a1 west: update zephyr to 2fa282b6cc8b
2) Name of the topology file
    * Topology: {FILE} - Built from above branch with alsatplg version 1.2.11
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM} - MTL 

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.
[dmesg-mtl-008-video-issue.log](https://github.com/thesofproject/sof/files/14561056/dmesg-mtl-008-video-issue.log)
[mtrace-mtl-008-video-issue.log](https://github.com/thesofproject/sof/files/14561059/mtrace-mtl-008-video-issue.log)

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

NOTE THAT
Reverting below patch has resolved the issue
https://github.com/thesofproject/sof/commit/08cf3d2232e4

udaymb, 2024-03-13:
@kv2019i - @ujfalusi and @abonislawski suggested to use https://github.com/thesofproject/linux/pull/4851
As the kernel used is - https://chromium.googlesource.com/chromiumos/third_party/kernel/+log/refs/heads/release-R123-15786.B-chromeos-6.1 (https://chromium.googlesource.com/chromiumos/third_party/kernel/+/d4f476208c715d8dee7fc7f28134d150f507af6f) patches cannot be directly applied from the PR - manual changes are done as attached - 
[mtl-6.1-kernel-pr-4851-backport.zip](https://github.com/thesofproject/sof/files/14582819/mtl-6.1-kernel-pr-4851-backport.zip)

With this issue is NOT observed
[dmesg-pr-4851-backport-6.1-24ww113.log](https://github.com/thesofproject/sof/files/14582833/dmesg-pr-4851-backport-6.1-24ww113.log)
[mtrace-pr-4851-backport-6.1-24ww113.log](https://github.com/thesofproject/sof/files/14582834/mtrace-pr-4851-backport-6.1-24ww113.log)

abonislawski, 2024-03-22:
Fixed in https://github.com/thesofproject/linux/pull/4851
+ v8.0.1 opt-in release for older kernel versions, otherwise v8.0 still recommended
@udaymb please close if all is good

closed 2024-04-09 by abonislawski