thesofproject/sof #8960: [BUG] SoundWire topology: need to use 24 bits

plbossart, 2024-03-18:
Reported initially by @bardliao  

All SoundWire topologies currently use S32_LE, which means all 32-bits of data will be pushed on the wire.

The problem is that the default frame shape for SoundWire is 50x4. The column0 is reserved for commands, so in theory we can have 3 streams of 24-bits each on column 1, 2, 3 respectively. 

But since the topologies use S32_LE, we can only support 2 streams: conventional arithmetic tells us there's no way to fit 64-bits of data in 50 bits.

That's a problem for integrated devices, e.g. if you want to have amp+headset at the same time on the same link. We need to switch to 24 valid bits to avoid artificial bandwidth limitations.

We certainly need to avoid the use of 32 bits for topology2 solutions. It needs to be determined if topology1 need this change or not.

lgirdwood, 2024-04-16:
Lets not touch topology1 here - @jsarha @singalsu @ranj063 lets include a fix for this in our Q2 topology updates..

lgirdwood, 2024-05-28:
@plbossart @bardliao @ranj063 @ujfalusi what's the status for this today ?

kv2019i, 2024-06-03:
As per Pierre's feedback, moving to v2.11.

kv2019i, 2024-09-05:
@bardliao @plbossart Can you add a status update on this and propose new milestone (raise priority and get in 2.11 or push to 2.12)?

bardliao, 2024-09-05:
@kv2019i @plbossart I think we can close this issue as https://github.com/thesofproject/sof/pull/9282 has been merged.

plbossart, 2024-09-09:
yeah, this is done already, closing

closed 2024-09-09 by plbossart