thesofproject/sof #8961: [BUG] LNLM_RVP_HDA has a DSP panic error from playback test

fredoh9, 2024-03-18:
**Describe the bug**
LNLM_RVP_HDA has DSP panic with playback test. TPLG=sof-hda-generic-ace1-4ch.tplg is used.
After reverting this PR, https://github.com/thesofproject/sof/pull/8934, it works fine again.

```
[  252.716691] <inf> eq_iir: eq_iir_init: comp:1 0x10 eq_iir_init()
[  252.717266] <inf> ipc: ipc_cmd: rx	: 0x44000010|0x30000058
[  252.717286] <inf> eq_iir: eq_iir_set_config: comp:1 0x10 eq_iir_set_config()
[  252.717818] <inf> ipc: ipc_cmd: rx	: 0x40000011|0x1000a
[  252.717845] <inf> eq_fir: eq_fir_init: comp:1 0x11 eq_fir_init()
[  252.718401] <inf> ipc: ipc_cmd: rx	: 0x44000011|0x30000038
[  252.718413] <inf> eq_fir: eq_fir_set_config: comp:1 0x11 eq_fir_set_config()
[  252.719043] <inf> ipc: ipc_cmd: rx	: 0x40000013|0x1000a
[  252.719280] <err> ipc: ipc4_get_drv: get_drv(): the provided UUID (b36ee4da 47f9006f cbfe6da0 ceb6d8e2) can't be found!
[  252.719286] <err> os: print_fatal_exception:  ** FATAL EXCEPTION
[  252.719293] <err> os: print_fatal_exception:  ** CPU 0 EXCCAUSE 13 (load/store PIF data error)
[  252.719296] <err> os: print_fatal_exception:  **  PC 0xa006f6c0 VADDR 0x4
[  252.719300] <err> os: print_fatal_exception:  **  PS 0x60e20
[  252.719303] <err> os: print_fatal_exception:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:14 CALLINC:2)
[  252.719306] <err> os: xtensa_dump_stack:  **  A0 0xa003c5ef  SP 0xa00f4dc0  A2 (nil)  A3 0x13
[  252.719310] <err> os: xtensa_dump_stack:  **  A4 (nil)  A5 0x47f9006f  A6 0xcbfe6da0  A7 0xa00f4dc0
[  252.719313] <err> os: xtensa_dump_stack:  **  A8 0xa006f6ba  A9 0xa00f4da0 A10 0x400fa000 A11 0x3840
[  252.719318] <err> os: xtensa_dump_stack:  ** A12 0xfff001ff A13 0x1c A14 0x400fa800 A15 (nil)
[  252.719321] <err> os: xtensa_dump_stack:  ** LBEG 0xa0035e15 LEND 0xa0035e24 LCOUNT 0xa005ad2f
[  252.719325] <err> os: xtensa_dump_stack:  ** SAR 0xc
[  252.719328] <err> os: xtensa_dump_stack:  **  THREADPTR (nil)
```

**To Reproduce**
TPLG=/lib/firmware/intel/sof-ipc4-tplg/sof-hda-generic-ace1-4ch.tplg MODEL=LNLM_RVP_HDA SOF_TEST_INTERVAL=5 ~/sof-test/test-case/check-playback.sh -d 3 -l 1 -r 1 -F

**Reproduction Rate**
100%

fredoh9, 2024-03-18:
The reverting PR is
https://github.com/thesofproject/sof/pull/8962

singalsu, 2024-03-19:
@fredoh9 The load of component fails because DRC is not enabled in Kconfig for LNL. Let's fix the Kconfig instead.

singalsu, 2024-03-19:
This should fix it: https://github.com/thesofproject/sof/pull/8963

fredoh9, 2024-03-20:
confirmed this is fixed by https://github.com/thesofproject/sof/pull/8963

closed 2024-03-20 by fredoh9