thesofproject/sof #8966: [BUG]  error 7 unsupported request to SET_PIPELINE_STATE IPC, could not alloc size = 1536

kv2019i, 2024-03-19:
**Describe the bug**
IPC error while running "~/sof-test/test-case/check-xrun-injection.sh -c 10 -m playback" as part of larger test plan (daily sof-test plan).

**To Reproduce**
Run test plan including check-xrun-injection.sh.

**Reproduction Rate**
?

**Expected behavior**
No IPC errors.

**Impact**
Loss of audio.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: [184f97930d2d](https://github.com/thesofproject/linux/commit/184f97930d2d96f098407a44f7720b1b2b3b3979)
    * SOF: https://github.com/thesofproject/sof/commit/dff733232673cdbedf8553a1e7a199b76d9e856d 
2) Name of the topology file
    * Topology: sof-mtl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: Intel mtl

**Screenshots or console output**
Seen on Intel daily test plan #38973
```
 [ 3945.314285] <inf> ipc: ipc_cmd: rx	: 0x13000004|0x1
[ 3945.314380] <err> buffer: buffer_alloc: buffer_alloc(): could not alloc size = 1536 bytes of type = 32
[ 3945.314386] <err> dai_comp: dai_set_dma_buffer: comp:1 0x10004 dai_set_dma_buffer(): failed to alloc dma buffer
[ 3945.314391] <err> dai_comp: dai_common_params: comp:1 0x10004 dai_zephyr_params(): alloc dma buffer failed.
[ 3945.314403] <err> module_adapter: module_prepare: comp:1 0x10004 module_prepare() error -12: module specific prepare failed, comp_id 65540
[ 3945.314410] <err> module_adapter: module_adapter_prepare: comp:1 0x10004 module_adapter_prepare() error fffffff4: module prepare failed
[ 3945.314416] <err> pipe: pipeline_prepare: pipe:1 0x0 pipeline_prepare(): ret = -12, dev->comp.id = 3
[ 3945.314421] <err> ipc: ipc4_pcm_params: ipc: pipe 1 comp 0 prepare failed -12
[ 3945.314460] <inf> component: comp_set_state: comp:1 0x10004 comp_set_state(), state already set to 1
[ 3945.314468] <err> ipc: ipc_cmd: ipc4: FW_GEN_MSG failed with err 7 
```

fredoh9, 2024-03-22:
LNL NOCODEC has same/similar errors.

check-signal-stop-start
TPLG=/lib/firmware/intel/development/sof-lnl-nocodec.tplg MODEL=LNLM_RVP_NOCODEC SOF_TEST_INTERVAL=5 ~/sof-test/test-case/check-signal-stop-start.sh -m capture -c 50

```
[ 3832.775173] <err> buffer: buffer_alloc: buffer_alloc(): could not alloc size = 1536 bytes of type = 32
[ 3832.775176] <err> dai_comp: dai_set_dma_buffer: comp:0 0x4 dai_set_dma_buffer(): failed to alloc dma buffer
[ 3832.775188] <err> dai_comp: dai_common_params: comp:0 0x4 dai_zephyr_params(): alloc dma buffer failed.
[ 3832.775193] <err> module_adapter: module_prepare: comp:0 0x4 module_prepare() error -12: module specific prepare failed, comp_id 4
[ 3832.775198] <err> module_adapter: module_adapter_prepare: comp:0 0x4 module_adapter_prepare() error fffffff4: module prepare failed
[ 3832.775201] <err> pipe: pipeline_prepare: pipe:0 0x0 pipeline_prepare(): ret = -12, dev->comp.id = 4
[ 3832.775206] <err> ipc: ipc4_pcm_params: ipc: pipe 0 comp 0 prepare failed -12
[ 3832.775216] <inf> component: comp_set_state: comp:0 0x4 comp_set_state(), state already set to 1
[ 3832.775220] <err> ipc: ipc_cmd: ipc4: FW_GEN_MSG failed with err 7
[ 3832.775301] <inf> host_comp: host_get_copy_bytes_normal: comp:1 0x10004 no bytes to copy, available samples: 0, free_samples: 384
```

and
multiple-pipeline-playback-50 test
TPLG=/lib/firmware/intel/development/sof-lnl-nocodec.tplg MODEL=LNLM_RVP_NOCODEC SOF_TEST_INTERVAL=5 ~/sof-test/test-case/c.sh -f p -l 50

```
[ 3856.342458] <inf> ipc: buffer_new: buffer new size 0x300 id 0.0 flags 0x0
[ 3856.342463] <err> buffer: buffer_alloc: buffer_alloc(): could not alloc structure
[ 3856.342468] <err> ipc: ipc_comp_connect: failed to allocate buffer to bind 4 to 6
[ 3856.342471] <err> ipc: ipc_cmd: ipc4: MODULE_MSG failed with err 3
```

Intel internal daily test link:
planresultdetail/39105

plbossart, 2024-04-05:
This is the last bug preventing us from adding LNL to the CI pool

see Intel daily test #/result/planresultdetail/39452?model=LNLM_RVP_NOCODEC&testcase=check-signal-stop-start-capture-50

Not sure why it's marked as P2 @kv2019i

kv2019i, 2024-04-08:
Might be related (and even a fix): https://github.com/thesofproject/sof/pull/9006

plbossart, 2024-04-08:
Still happening today in daily tests, see #/result/planresultdetail/39481?model=MTLP_RVP_NOCODEC&testcase=check-xrun-injection-playback-10

marc-hb, 2024-04-09:
Again today in esult/planresultdetail/39547?model=LNLM_RVP_NOCODEC&testcase=volume-basic-test-50

It behaves like a memory leak?

marc-hb, 2024-04-12:
Not seen in today's daily tests.

closed 2024-04-12 by marc-hb