thesofproject/sof #8987: [FEATURE]

Dae-Hang-Sook, 2024-03-27:
**Is your feature request related to a problem? Please describe.**
The inability to install into windows 10 no matter what is done or tried.

**Describe the solution you'd like**
A way to natively install sof into windows 10 for devces that use it. (glkda7219max in my case)

**Describe alternatives you've considered**
Paying for the ability to do this. (Coolstar)

**Additional context**
I'd like to be able to install sof firmware just as easily in windows as in linux and also for free just like in linux.

plbossart, 2024-03-27:
This isn't really a firmware feature, rather a problem on the Windows driver side.

marc-hb, 2024-03-30:
SOF is simply not supported with Windows right now and it will never be with very old products like GLK.

closed 2024-03-30 by marc-hb

Dae-Hang-Sook, 2024-04-04:
> This isn't really a firmware feature, rather a problem on the Windows driver side.

Will anything ever be done or just not at all?

plbossart, 2024-04-04:
It's fair to say that no one has any plans to support this feature on GLK and Windows10. That ship has sailed many moons ago.

Dae-Hang-Sook, 2024-04-04:
> It's fair to say that no one has any plans to support this feature on GLK and Windows10. That ship has sailed many moons ago.

My bad, I meant just SOF in general this time, not just GLK devices.