thesofproject/sof #8997: [BUG] Topology for ALC274 (Digital Microphone Now Have Two Left/Right Inputs Instead of Mono)

gregory112, 2024-04-01:
Describe the bug
----------------
Razer Blade 15 (2022) with Alder Lake PCH-P High Definition Audio Controller Digital Microphone (Realtek ALC274) now has two inputs (stereo, left and right), so applications like Discord can no longer pick it up as microphone properly.

I wold like to know if the default topology installed by the package manager (`options snd-sof-pci tplg_filename=sof-hda-generic-2ch-pdm1.tplg`) is the right one for me? I think microphone is not supposed to have two inputs except for stereo mic even for digital microphone so I think there is a misconfiguration somewhere? Do I need to change the topology file to correctly match my system?

Environment
-----------
1) Version number
sof-firmware 2023.12.1-1

2) Linux distribution and version
6.8.1-arch1-1 SMP PREEMPT_DYNAMIC Sat, 16 Mar 2024 17:15:35 +0000 x86_64 GNU/Linux

3) Hardware vendor and model
Razer Blade 15 2022, Realtek ALC274

4) Output of the following command: `journalctl -k | grep -C 10 sof.*firmware`
   (or the equivalent if your system does not have `journalctl`.
```
kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: enabling device (0000 -> 0002)
kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 5
kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 2
kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware paths/files for ipc type 0:
kernel: sof-audio-pci-intel-tgl 0000:00:1f.3:  Firmware file:     intel/sof/sof-adl.ri
kernel: sof-audio-pci-intel-tgl 0000:00:1f.3:  Topology file:     intel/sof-tplg/sof-hda-generic-2ch-pdm1.tplg
kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:22:1 Kernel ABI 3:23:0
kernel: skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Parent card not yet available, widget card binding deferred
kernel: snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC274: line_outs=1 (0x17/0x0/0x0/0x0/0x0) type:speaker
kernel: input: HDA NVidia HDMI/DP,pcm=8 as /devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card1/input23
kernel: input: Razer Razer Blade as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1532:028A.0004/input/input26
kernel: input: HDA NVidia HDMI/DP,pcm=9 as /devices/pci0000:00/0000:00:01.0/0000:01:00.1/sound/card1/input24
kernel: input: Razer Razer Blade as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:028A.0005/input/input27
kernel: skl_hda_dsp_generic skl_hda_dsp_generic: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 3
kernel: input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card2/input28
kernel: input: Razer Razer Blade as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:028A.0006/input/input33
kernel: input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card2/input29
kernel: input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card2/input30
kernel: input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card2/input31
kernel: input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card2/input32
```

plbossart, 2024-04-01:
Sorry, not following the description of the issue.

The topology exposes ONE PCM device to applications. Application can record with two channels.

What is the issue?

Please attach the log of 'alsa-info' and describe what does not work in more details, thanks.

plbossart, 2024-04-01:
I forgot to mention that the DMICs are not attached physically to the ALC274 but directly to the Intel PCH.

gregory112, 2024-04-16:
Hello, sorry, the simples explanation of the issue is that *I thought* the DMICs should have one mono input instead of two left and right inputs. As you said, application can record with two channels, so I guess this is normal.

When I checked Discord again, what is causing the bug that it does not record any sound is my bluetooth speaker, so I guess we can close this issue now. Sorry for the inconvenience.

closed 2024-04-16 by lgirdwood