thesofproject/sof #9011: [FEATURE] tplg build should catch too long control names

kv2019i, 2024-04-09:
**Is your feature request related to a problem? Please describe.**
Currently SOF topologies can be created with too long control names and these will fail upn Linux kernel loading the topologies.

**Describe the solution you'd like**
It should be possible to constraint length of "string" data and capture invalid entries at topology build time.

**Describe alternatives you've considered**

**Additional context**
Filed as result of https://github.com/thesofproject/sof/pull/9008

lgirdwood, 2024-04-16:
@ranj063 @jsarha @singalsu lets include this fix in our topology Q2 work. Thanks

jsarha, 2024-05-28:
I can take this. Thou I guess this is not a top prioroty.

kv2019i, 2024-09-05:
Not a priority item, pushing to next.

kv2019i, 2024-12-13:
Feature cutoff for v2.12, moving this to v2.13.

kv2019i, 2025-04-23:
@jsarha any update? two weeks for v2.13 feature cutoff.

jsarha, 2025-04-25:
@kv2019i  there has constantly been higher priority jobs assigned to me, so this has not gotten any cycles.

kv2019i, 2025-04-28:
Ack @jsarha . Let me move to TBD milestone and we'll reschedule this later.

kv2019i, 2025-11-12:
Thank you for the contribution @Biancaa-R ! I don't actually know where exactly this check for errors should be made, but I think the tplg_parser is not the right place as that is not run during the build process for topologies. I'm suspecting the correct place is in the conf2tplg compiler in ( https://github.com/alsa-project/alsa-utils/commits/master/topology ). Did you @jsarha took a look at all where this could be done? 

For background, we want to avoid errors like was fixed here https://github.com/thesofproject/sof/pull/9008 
Sorry @Biancaa-R , this may not be the easiest issue to start in SOF.

jsarha, 2025-11-14:
@kv2019i , The correct place would of course be the alsa-utils topology compiler, but I have taken on the task since there has all the time been higher priority tasks on my queue, and respinning a topology compiler work will always take some time. An adhock solution of writing a python checker for our topologies on top of already existing parser, would be faster, but not as elegant solution.