thesofproject/sof #9026: [LNL,MTL] dai_ssp_set_reg_config: SSPx configuration: Ignoring SSCR1:TSRE SSCR1:RSRE from blob

marc-hb, 2024-04-10:
Seems to be printed multiple times in most (passing!) tests on LNL, for instance daily run #/result/planresultdetail/39547?model=LNLM_RVP_NOCODEC&testcase=check-playback-100sec

```
[  392.350033] <inf> pipe: pipeline_new: pipeline new pipe_id 1 priority 0
[  392.350726] <inf> ipc: ipc_cmd: rx	: 0x40000003|0x201000a
[  392.351050] <inf> ipc: ipc_cmd: rx	: 0x40010006|0x2010010
[  392.351481] <inf> ipc: ipc_cmd: rx	: 0x44010006|0x30000018
[  392.352003] <inf> ipc: ipc_cmd: rx	: 0x40010004|0x2010056
[  392.352075] <inf> dai_intel_ssp: dai_ssp_pm_runtime_en_ssp_power: SSP2
[  392.352098] <inf> dma: dma_get: dma_get() ID 0 sref = 1 busy channels 0
[  392.352105] <inf> ipc: dai_config: comp:1 0x10004 dai_config() dai type = 1 index = 2 dd 0x4010fec0
[  392.352115] <inf> dai_intel_ssp: dai_ssp_parse_aux_data: link clock_source 1
[  392.352118] <err> dai_intel_ssp: dai_ssp_set_reg_config: SSP2 configuration:
[  392.352126] <inf> dai_intel_ssp: dai_ssp_set_reg_config:  Ignoring SSCR1:TSRE SSCR1:RSRE from blob
[  392.352131] <inf> dai_intel_ssp: dai_ssp_set_reg_config:  sscr0 = 0xc1d0077f, sscr1 = 0xd0400004, ssto = 0x00000000, sspsp = 0x2200000
[  392.352136] <inf> dai_intel_ssp: dai_ssp_set_reg_config:  sscr2 = 0x00004002, sspsp2 = 0x00000000, sscr3 = 0x07070000
[  392.352140] <inf> dai_intel_ssp: dai_ssp_set_reg_config:  ssioc = 0x00000020, ssrsa = 0x00000003, sstsa = 0x00000003
[  392.352605] <inf> ipc: ipc_cmd: rx	: 0x45000004|0x6
[  392.352641] <inf> ipc: buffer_new: buffer new size 0x80 id 0.0 flags 0x0
[  392.352653] <inf> pipe: pipeline_connect: comp:0 0x4 connect buffer 0 as sink

```

Same message with SSP0, SSP1 and SSP2

If this is not an error then the message should be downgraded.

Linux Branch: [topic/sof-dev](https://github.com/thesofproject/linux/tree/topic/sof-dev)
Linux Commit: [45fa8789fdea](https://github.com/thesofproject/linux/commit/45fa8789fdeaa4de3df02a9ddcc43c3c7a870bdd)
KConfig Branch: [master](https://github.com/thesofproject/kconfig/tree/master)
KConfig Commit: [d885ec26b1a9](https://github.com/thesofproject/kconfig/commit/d885ec26b1a97d31a037401f7625fe5524ee64a0)

SOF Branch: [main](https://github.com/thesofproject/sof/tree/main)
SOF Commit: [0542687ad921](https://github.com/thesofproject/sof/commit/0542687ad92185cc84208dd78522ba6f0fdf230e)
Zephyr Commit: [1f55be8b42df](https://github.com/zephyrproject-rtos/zephyr/commit/1f55be8b42dfd54308038d1e422d8d4e0e7f39ab)

EDIT: also visible for MTL at https://sof-ci.01.org/sofpr/PR8956/build3506/devicetest/index.html?model=MTLP_RVP_NOCODEC&testcase=check-playback-10sec

cc: @dnikodem (from https://github.com/zephyrproject-rtos/zephyr/pull/70660/)

marc-hb, 2024-04-10:
Based on https://github.com/thesofproject/sof-test/pull/1075#issuecomment-2048371469 results this seems to be a NOCODEC thing.

NOCODEC reproduction rate: 100%
"codec" reproduction rate: 0%

marc-hb, 2024-04-10:
Raising the priority of this because this seems to be last thing stopping us for catching (all) other firmware errors.

plbossart, 2024-04-10:
> Based on [thesofproject/sof-test#1075 (comment)](https://github.com/thesofproject/sof-test/pull/1075#issuecomment-2048371469) results this seems to be a NOCODEC thing.
> 
> NOCODEC reproduction rate: 100% "codec" reproduction rate: 0%

well d'uh, that's were we test the SSP. HDaudio and SDW devices don't rely on the SSP.

marc-hb, 2024-04-10:
> It's not an error really, it's a feature but that's really not important to log.

So, just downgrade to the message to WARN level?

plbossart, 2024-04-10:
Warn is even too high IMHO, it's a known behavior. Nothing different to expect.

marc-hb, 2024-04-11:
The LOG_ERR was actually added by https://github.com/zephyrproject-rtos/zephyr/commit/6423bc3bc885 and the level looks clearly unintentional there. I'm submitting a fix.

ujfalusi, 2024-04-11:
The ` dai_intel_ssp: dai_ssp_set_reg_config: SSP2 configuration:` is in wrong level, it should be info, sending a fix right away.

ujfalusi, 2024-04-11:
@marc-hb, I see you already done that, thanks!

lgirdwood, 2024-04-16:
@marc-hb can we close ?

marc-hb, 2024-04-16:
I don't think we've updated Zephyr that far yet.

marc-hb, 2024-04-23:
Zephyr updated, we have the fix now. INF level in https://sof-ci.01.org/softestpr/PR1075/build327/devicetest/index.html

closed 2024-04-23 by marc-hb