thesofproject/sof #9061: [FEATURE] UCM support for DMI named based differentiation

lgirdwood, 2024-04-18:
**Is your feature request related to a problem? Please describe.**
UCM can currently load coefficients and configure HW based on the sound card name. It needs to additionally differentiate configuration based on device DMI name.

UCM to read DMI device name and program some kcontrols differently based on DMI name.

@kv2019i @ujfalusi @ranj063 @singalsu

plbossart, 2024-04-18:
some of the sound card names (can't recall if this is short, long name) are already based on DMI.

UCM will select the conf if it finds a UCM file matching that card name.

There's no need to read DMI information within UCM...

kv2019i, 2024-04-19:
@plbossart wrote:
> There's no need to read DMI information within UCM...

For UCM file maintenance, if the delta is small, having a conditional definition based on DMI, might be a pragmatic approach to variate. Like we handle SOF hda platforms with and without dmic now, these are handled with conditional definitions. For some things, DMI-based conditional bits might be practise. This seems to be use to some degree in upstream UCM files already.

plbossart, 2024-04-19:
I was referring to this:
https://github.com/thesofproject/linux/blob/4dc32018572e43814ef03a421f4aca0ad067b1ad/sound/soc/soc-core.c#L1902

We've used this before to select board-specific UCM conf files: it's even documented!

"
 * This will help the user space to load the device-specific Use Case Manager
 * (UCM) configurations for the card.
"

ujfalusi, 2024-04-19:
@plbossart, I see. `snd_soc_set_dmi_name()` is in practice unused in kernel, but if we start to use then it will break UCM for SOF cards.
Unless we do a DMI based decision to use the `snd_soc_set_dmi_name()`, which is basically the same thing as if we would use the DMI match in UCM (there are examples already to do so).

If we only use UCM then we will take care of dependency on kernel..

plbossart, 2024-04-19:
What do you mean it's unused?

By default if the 'long_name' is not set, it will be populated with a DMI string.

see e.g. with SoundWire with this diff

````diff
diff --git a/sound/soc/intel/boards/sof_sdw.c b/sound/soc/intel/boards/sof_sdw.c
index fed4132e9b72..942ee94cb80c 100644
--- a/sound/soc/intel/boards/sof_sdw.c
+++ b/sound/soc/intel/boards/sof_sdw.c
@@ -1890,7 +1890,7 @@ static int sof_sdw_card_late_probe(struct snd_soc_card *card)
 }
 
 /* SoC card */
-static const char sdw_card_long_name[] = "Intel Soundwire SOF";
+//static const char sdw_card_long_name[] = "Intel Soundwire SOF";
 
 static struct snd_soc_card card_sof_sdw = {
        .name = "soundwire",
@@ -2011,7 +2011,7 @@ static int mc_probe(struct platform_device *pdev)
                        return -ENOMEM;
        }
 
-       card->long_name = sdw_card_long_name;
+       //card->long_name = sdw_card_long_name;
 
        /* Register the card */
        ret = devm_snd_soc_register_card(card->dev, card);
````

before:
````
amixer -Dhw:0 info
Card hw:0 'sofsoundwire'/'Intel Soundwire SOF'
````
after
````
amixer -Dhw:0 info
Card hw:0 'sofsoundwire'/'DellInc.-PrecisionXXXX--XXXXX'
````
(Xs redacted since this is an experimental device).

plbossart, 2024-04-19:
Other examples from the daily test pool. The last string is the long_name based on DMI information

````
Card hw:0 'sofglkda7219max'/'Google-Bobba-rev2
Card hw:0 'sofhdadsp'/'AAEON-UP_WHL01-V1.0'
Card hw:0 'sofhdadsp'/'IntelCorporation-MeteorLakeClientPlatform-0.1-MTL_PDDR5SODIMMSBSRVP'
````

ujfalusi, 2024-04-22:
> What do you mean it's unused?
> 
> By default if the 'long_name' is not set, it will be populated with a DMI string.

I stand corrected, I have not looked at the code in `snd_soc_set_dmi_name()`
sof-sdw did made the DMI based long name configuration ignored by setting the long_name to a static string.

lgirdwood, 2024-04-22:
long names sounds good if always unique - have we tested long name with UCM yet ?

kv2019i, 2024-05-27:
Implementation already in review at https://github.com/alsa-project/alsa-ucm-conf/pull/419 (assigning to Seppo as he has been working on this).

kv2019i, 2024-06-03:
https://github.com/alsa-project/alsa-ucm-conf/pull/419 merged upstream, closing this.

closed 2024-06-03 by kv2019i