thesofproject/sof #9065: [FEATURE] Build SOF for Ubuntu 20.04 / GLIBC < 2.34

hansfn, 2024-04-19:
Preamble: Since this question is neither a bug or feature request, I tried to use the Discord link in the Readme. However, Discord tells me: "No Text Channels - You find yourself in a strange place. You don't have access to any text channels, or there are none in this server." 

**Is your feature request related to a problem? Please describe.**
I'm trying to get audio to work on a panel-PC with Intel Elkhart Lake. The internet seems to agree that I need SOF firmware.

I tried using the [sof-bin binaries](https://github.com/thesofproject/sof-bin) directly (and indirectly using emerge on Gentoo), but the files are unusable for me - I think - since the are compiled on Ubuntu 22.04 requiring GLIB 2.34. Yes, I get "version `GLIBC_2.34' not found" when using the `sof-probes` binary. Unfortunately our SW stack can't use GLIBC 2.34 - we are stuck at GLIBC 2.33. 

I have [modified the Docker setup](https://github.com/thesofproject/sof/files/15042129/sof-docker-ubuntu-20-04.txt) so it uses Ubuntu 20.04 and everything works, but I'm missing several platforms:

````
./scripts/docker-run.sh ./scripts/xtensa-build-all.sh tgl
````

returns

````
Error: Unknown platform specified: tgl
Known platforms are: imx8m imx8ulp rn rmb vangogh mt8186 mt8195 mt8188 acp_6_3
````

**Describe the solution you'd like**
A working Docker setup for building with Zephyr? I guess it is needed to get the platforms I need. Or is there some other way that I can avoid installing the complete toolchain and so on.

plbossart, 2024-04-19:
You don't need sof-probes and other debug executables. Just use the firmware/topology binaries.

hansfn, 2024-04-19:
Thanks for the clarification. But I guess still need firmware/topology binaries for the correct platform (tgl?).

plbossart, 2024-04-19:
use the install scripts and you'll get those binaries copied over in /lib/firmware/intel

hansfn, 2024-04-19:
This isn't about copying/installing the files. The files aren't built for my platform. The instructions (for Bay Trail as an example) in the [docs](https://thesofproject.github.io/latest/getting_started/build-guide/build-with-docker.html#build-with-scripts) doesn't work.

plbossart, 2024-04-19:
ElkhartLake is only supported with the 2.2 release and the binaries are available

https://github.com/thesofproject/sof-bin/blob/main/v2.2.x/sof-v2.2/sof-ehl.ri

hansfn, 2024-04-19:
Thx a lot, I'll try those binaries, but still examples in the docs should work. In particular compiling for a platform.

closed 2024-04-22 by hansfn