thesofproject/sof #9067: [BUG] regression with Intel bt-offload tplgs

kv2019i, 2024-04-22:
**Describe the bug**
Commit c77a4feb2f1f15a3a29edf49b091b4b4a507be48 breaks BT offload tplgs like sof-nocodec-bt-mtl-lbm.tplg .

**To Reproduce**
TPLG=/lib/firmware/intel/development/sof-nocodec-bt-mtl-lbm.tplg MODEL=MTLP_RVP_NOCODEC-bluetooth-offload SOF_TEST_INTERVAL=5 ~/sof-test/test-case/check-playback.sh -d 3 -l 1 -r 1 -F 

**Reproduction Rate**
100%

**Expected behavior**
Playback works.

**Impact**
Playback fails.

**Environment**
1) SOF Commit: 
[c77a4feb2f1f](https://github.com/thesofproject/sof/commit/c77a4feb2f1f15a3a29edf49b091b4b4a507be48)
2) Linux Commit: 
[4ede2a0ed114](https://github.com/thesofproject/linux/commit/4ede2a0ed114a633d8eb9868ad6ad91cf0b08f79)

2) Name of the topology file
    * Topology: sof-nocodec-bt-lbm.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: Intel MTL

**Screenshots or console output**

```
2024-04-17 11:13:38 UTC [REMOTE_COMMAND] sof-tplgreader.py /usr/lib/firmware/intel/development/sof-nocodec-bt-mtl-lbm.tplg -f 'type:playback & id:any & ~pcm:HDMI' -b ' pcm:HDA Digital' -s 0 -e
2024-04-17 11:13:38 UTC [REMOTE_INFO] ===== Testing: (Round: 1/1) (PCM: Port2 [hw:0,2]) (Loop: 1/1) =====
2024-04-17 11:13:38 UTC [REMOTE_COMMAND] aplay   -Dhw:0,2 -r 48000 -c 1 -f S16_LE -d 3 /dev/zero -v -q
aplay: set_params:1416: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 16
CHANNELS: 1
RATE: 48000
PERIOD_TIME: 125000
PERIOD_SIZE: 6000
PERIOD_BYTES: 12000
PERIODS: 4
BUFFER_TIME: 500000
BUFFER_SIZE: 24000
BUFFER_BYTES: 48000
TICK_TIME: 0 
```

kv2019i, 2024-04-22:
FYI @marc-hb @ranj063 @ujfalusi @plbossart

kv2019i, 2024-04-22:
And FYI @fredoh9 
Not sure if there's an issue with tplg-reader as well. 

aplay   -Dhw:0,2 -r 48000 -c 1 -f S16_LE -d 3 /dev/zero -v -q

Is not expected to work, so the test case runs the wrong commeand, but also this fails

aplay   -Dhw:0,2 -r 16000 -c 1 -f S16_LE -d 3 /dev/zero -v -q

ujfalusi, 2024-04-22:
@kv2019i, what is in dmesg? What are not finding?

ujfalusi, 2024-04-22:
I wonder if the same issue present with other setups:
```
# git grep rate_m tools/topology/topology2/

tools/topology/topology2/cavs-benchmark-hda.conf:                       rate_min 8000
tools/topology/topology2/cavs-benchmark-hda.conf:                       rate_max 192000
tools/topology/topology2/cavs-benchmark-hda.conf:                       rate_min 8000
tools/topology/topology2/cavs-benchmark-hda.conf:                       rate_max 192000
tools/topology/topology2/cavs-mixin-mixout-hda.conf:                    rate_min $HDA_ANALOG_PLAYBACK_RATE
tools/topology/topology2/cavs-mixin-mixout-hda.conf:                    rate_max $HDA_ANALOG_PLAYBACK_RATE
tools/topology/topology2/cavs-mixin-mixout-hda.conf:                    rate_min $HDA_ANALOG_CAPTURE_RATE
tools/topology/topology2/cavs-mixin-mixout-hda.conf:                    rate_max $HDA_ANALOG_CAPTURE_RATE
tools/topology/topology2/cavs-nocodec.conf:                                     rate_min 8000
tools/topology/topology2/cavs-nocodec.conf:                                     rate_max 192000
tools/topology/topology2/cavs-nocodec.conf:                                     rate_min 8000
tools/topology/topology2/cavs-nocodec.conf:                                     rate_max 192000
tools/topology/topology2/cavs-sdw-src-gain-mixin.conf:                  rate_min 16000
tools/topology/topology2/cavs-sdw-src-gain-mixin.conf:                  rate_max 48000
tools/topology/topology2/cavs-src-mixin-mixout-hda.conf:                        rate_min 8000
tools/topology/topology2/cavs-src-mixin-mixout-hda.conf:                        rate_max 192000
tools/topology/topology2/include/common/pcm_caps.conf:#         rate_min                48000
tools/topology/topology2/include/common/pcm_caps.conf:#         rate_max                48000
tools/topology/topology2/include/common/pcm_caps.conf:  DefineAttribute."rate_min" {}
tools/topology/topology2/include/common/pcm_caps.conf:  DefineAttribute."rate_max" {}
tools/topology/topology2/platform/intel/bt-generic.conf:                        rate_min 8000
tools/topology/topology2/platform/intel/bt-generic.conf:                        rate_max 48000
tools/topology/topology2/platform/intel/bt-generic.conf:                        rate_min 8000
tools/topology/topology2/platform/intel/bt-generic.conf:                        rate_max 48000
tools/topology/topology2/platform/intel/dmic-generic.conf:                      rate_min $DMIC0_RATE
tools/topology/topology2/platform/intel/dmic-generic.conf:                      rate_max $DMIC0_RATE
tools/topology/topology2/platform/intel/dmic-wov.conf:                  rate_min        16000
tools/topology/topology2/platform/intel/dmic-wov.conf:                  rate_max        16000
tools/topology/topology2/platform/intel/dmic-wov.conf:                  rate_min        16000
tools/topology/topology2/platform/intel/dmic-wov.conf:                  rate_max        16000
tools/topology/topology2/platform/intel/dmic-wov.conf:                  rate_min     16000
tools/topology/topology2/platform/intel/dmic-wov.conf:                  rate_max     16000
tools/topology/topology2/platform/intel/dmic1-passthrough.conf:                        rate_min $DMIC1_RATE
tools/topology/topology2/platform/intel/dmic1-passthrough.conf:                        rate_max $DMIC1_RATE
tools/topology/topology2/platform/intel/hdmi-in-capture.conf:                   rate_min 48000
tools/topology/topology2/platform/intel/hdmi-in-capture.conf:                   rate_max 48000
tools/topology/topology2/platform/intel/hdmi-in-capture.conf:                   rate_min 48000
tools/topology/topology2/platform/intel/hdmi-in-capture.conf:                   rate_max 48000
```

marc-hb, 2024-04-23:
As expected, all MTLP_RVP_NOCODEC-bluetooth-offload tests still failing in today's daily 40182

marc-hb, 2024-04-26:
Daily tests 40387 are still all failing on MTLP_RVP_NOCODEC-bluetooth-offload even with today's commit 278ecc5b74ad "fix rate constraints".

kv2019i, 2024-04-29:
@marc-hb wrote:
> Daily tests 40387 are still all failing on MTLP_RVP_NOCODEC-bluetooth-offload even with today's commit [278ecc5](https://github.com/thesofproject/sof/commit/278ecc5b74adae50547e57cf147b0bb6147cd3d8) "fix rate constraints".

This is expected. The SOF tplg is now correct and FW behaves as expected.

The test fails as sof-test tries a combination that is not supported:
```
2024-04-26 14:12:31 UTC [REMOTE_INFO] ===== Testing: (Round: 1/1) (PCM: Port2 [hw:0,2]) (Loop: 1/1) =====
2024-04-26 14:12:31 UTC [REMOTE_COMMAND] aplay   -Dhw:0,2 -r 48000 -c 1 -f S16_LE -d 3 /dev/zero -v -q
aplay: set_params:1416: Unable to install hw params:
```

The PCM does not support opening 48kHz in mono/1ch mode, so this command is expected to fail. The problem is now how can sof-test no what to test. Current code assumes all combinations of supported channels and support sampling rates are supported, but this is not the case.

Earlier this worked with some luck as sof-test picked a combination to test that happened to work.

marc-hb, 2024-04-29:
> In any case, this is a sof-test problem now.

I was about to transfer this issue to sof-test to avoid creating a new one (the context is spread in too many Github places already) but I'm not so sure now. Someone named @kv2019i wrote this:

https://github.com/thesofproject/sof/pull/9068#discussion_r1575898007
> ack. his is just hitting the limits of what we can express in capabilities. the topology description of PCM cannot express the PCM does not support all combinations of supported channels and rates.

If the required information isn't available in the first place then sof-test cannot divine it and this just not usable and testable, is it? Let's leave sof-test aside for a moment and focus on real use cases (which sof-test should be as close to as possible). How does "real" software know which combinations are valid versus not? Some UCM magic maybe?

kv2019i, 2024-04-30:
@marc-hb There are a possibilities. The alsa-lib C interface allows applications to discover such complex constraints. I.e. application can leave channel count as "open" or use the "_near" variants to ask a specific value but be open to adjust to hardware capabilities. sof-test does not use this interface, but rather it implements its own discover of capability and aplay/arecord are instructed to use specific settings via the "hw:x" layer and explicit setting of all key audio parameters. This makes sense as we want to specifically test the audio drivers and want to control the audio parameters used in the tests.

In the specific case of BT, the set of supported audio formats comes from the set of Bluetooth profiles we support. That is currently HFP and A2DP and there is a enumerated set of formats the Bluetooth chip and audio DSP support. So it's hardcoded in the user-space code for BT offload that these specific audio formats are used. 

So a pragmatic solution would perhaps be to detect (in sof-test) we are testing Bluetooth offload, and ignore known bad audio format combinations.