thesofproject/sof #9069: [BUG] Loooking to compile adl-n firmware and topology

rhenry-hai, 2024-04-22:
Hello, I am new to Sound Open Firmware. Indeed, I need to implement the Alderlake-n with the da7219 IC but I'm not getting any results so far.

I've already had a look at the documentation to build the firmware and I'm stuck building the adl-n platform as described in the documentation. In fact, even on an earlier version I can't find the build script "xtensa-build-all.sh" with support for the Alderlake-n platform.

I've already tried many firmwares available in the "SOF-bin" directory for Alderlake-n, but I haven't been able to find one that works for my case.

Why the latest version of SOF do not accept all hardware platform available on documentation ?

Best Regards Romain H.

plbossart, 2024-04-22:
the binary you need is 
"sof-adl-n.ri",

It's clearly available from https://github.com/thesofproject/sof-bin/blob/main/v2.2.x/sof-v2.2/intel-signed/sof-adl-n.ri

rhenry-hai, 2024-04-23:
Hi plbossart,

Thanks for your feedback.

I try to test sof-adl-n.ri from the v2.2. As I look to the "dmesg" i follow error to put "sof-adl-nocodec.tplg" to load a proper topology and it's working.

In my case, I need to use da7219 for DAC/ADC. I currenlty looking to build custom topology for this purpose.
 Is it possible to load custom topology with generic "sof-adl-n.ri" ?
 
 Best regards Romain.

plbossart, 2024-04-23:
Yes, for a given firmware we can adjust the topology. The topology is only for the Intel DSP, i.e. it's not dependent on what type of codec is connected, only on which I2S links are used.

So your best bet is to start from an existing da7219-based topology, e.g. sof-apl-da7219.m4, and update it depending on which SSP you need to use and which settings are needed for that link (bclck, fsync, valid bits, etc).

Note that all the topology1 files are no longer in main but are still visible in the 2.2 stable branch under tools/topology/topology1/.

Changing and updating topologies is a complicated and error-prone process, make sure you think twice about that you actually need.

Then the other thing to handle is how this platform will be detected by the Linux host, for this you'll have to add a descriptor in sound/soc/intel/common/soc-acpi-intel-adl-match.c 

You will of course need to have some sort of identifier in ACPI/DSDT for the detection to be effective. If your platform does not have this available, you'll need to add it with the initrd override, see https://github.com/thesofproject/acpi-scripts/blob/master/LeafHill/DLGS7219.asl - and for this you'll need to know which I2C controller your device is connected to.

If it looks complicated it's because it is. If you want more support, you'll need to share what this platform of yours actually is, how the connections are handled, etc.