thesofproject/sof #9081: [FEATURE] Ability to use Zephyr shell with SOF

kv2019i, 2024-04-25:
**Is your feature request related to a problem? Please describe.**
Add overlay (or build option) to implement Zephyr shell support to SOF, and a backend that allows to attach to the shell from host CPU (over e.g. a memory window backend). This offers a new debugging and development option for SOF development, opening up possibility to observe system state and execute test sequences on the DSP.

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

kv2019i, 2024-04-25:
Working on this now,

lgirdwood, 2024-04-25:
@andyross @cujomalainey fyi

cujomalainey, 2024-04-26:
Interesting, would it just appear as a tty to the kernel?

lgirdwood, 2024-04-26:
> Interesting, would it just appear as a tty to the kernel?

That's the intention. The kernel tty would just be a passthrough from the DSP. Similar to how GDB would work.

andyross, 2024-05-01:
Ended up here after commenting on this Zephyr PR: (https://github.com/zephyrproject-rtos/zephyr/pull/72158)

This sounds cool.  Ironically I've never been a shell person on the Zephyr side (almost always if you're chasing a bug it's easier to just rebuild/reflash the whole app rather than trying to "use it interactively").

But one big feature it would add here is the ability to do whitebox testing on live audio firmware, something that the kernel driver normally tries its hardest to prevent.   Having a backchannel from userspace would be really nice, even if you have to use the slightly crufty shell API to do it.

lgirdwood, 2024-05-02:
> But one big feature it would add here is the ability to do whitebox testing on live audio firmware, something that the kernel driver normally tries its hardest to prevent. Having a backchannel from userspace would be really nice, even if you have to use the slightly crufty shell API to do it.

Ack, we were thinking things like injecting errors in pipelines, dumping state during tuning- really a method to help increase technical velocity in many ways.

kv2019i, 2024-05-14:
First Zephyr PR submitted for review https://github.com/zephyrproject-rtos/zephyr/pull/72738 . SOF PR to add a config overlay and (possible) Linux SOF client to create the pty, will come later.

cujomalainey, 2024-05-14:
@johnylin76 with relation to #5521

kv2019i, 2024-05-17:
Current set of PRs in progress:
- https://github.com/thesofproject/sof/pull/9133
- https://github.com/zephyrproject-rtos/zephyr/pull/72738
- dependency for above: https://github.com/zephyrproject-rtos/zephyr/pull/72799

kv2019i, 2024-05-24:
Dependencies merged, now waiting for the https://github.com/zephyrproject-rtos/zephyr/pull/72738 to be merged in Zephyr.

Uploaded a sof-docs PR to https://github.com/thesofproject/sof-docs/pull/496 to document shell usage with some examples.

kv2019i, 2024-05-28:
While waiting for Zephyr side PR to be merged, here's one more draft on SOF side that adds first custom command:

https://github.com/thesofproject/sof/pull/9174

kv2019i, 2024-05-31:
Shell backend now merged to Zephyr. Next step is to bring in the changes to SOF main via https://github.com/thesofproject/sof/pull/9179

kv2019i, 2024-06-05:
Main feature completed and merged to v2.10. The SOF side config overlay (convenience to build with shell) and additional SOF custom commands, are still in progress, but these goes go out of the scope for this original feature request.

Follow-up PRs:
- SOF: https://github.com/thesofproject/sof/pull/9133
- SOF: https://github.com/thesofproject/sof/pull/9174
- SOF docs: https://github.com/thesofproject/sof-docs/pull/496

closed 2024-06-05 by kv2019i