thesofproject/sof #9085: [BUG] No sound on Acer Swift SFG14-72-735T (Intel Ultra 7 155H)

liquidev, 2024-04-26:
There seems to be missing firmware for the DSP chip used in this particular laptop. Journalctl mentions a missing topology file:
```
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: SoundWire enabled on CannonLake+ platform, using SOF driver
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: enabling device (0000 -> 0002)
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: use msi interrupt mode
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: No SoundWire machine driver found
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: hda codecs found, mask 4
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: DMICs detected in NHLT tables: 2
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: SOF firmware and/or topology file not found.
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Supported default profiles
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: - ipc type 1 (Requested):
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3:  Firmware file: intel/sof-ipc4/mtl/sof-mtl.ri
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3:  Topology file: intel/sof-ace-tplg/sof-hda-generic-idisp-2ch.tplg
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Check if you have 'sof-firmware' package installed.
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Optionally it can be manually downloaded from:
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3:    https://github.com/thesofproject/sof-bin/
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: error: sof_probe_work failed err: -2
```

I've looked for this topology file but it doesn't seem to exist anywhere. There's only `sof-tplg/sof-hda-generic-idisp-2ch.tplg`, but not `sof-ace-tplg/sof-hda-generic-idisp-2ch.tplg`.

The following audio chip is reported in `lspci`:

```
0000:00:1f.3 Multimedia audio controller [0401]: Intel Corporation Meteor Lake-P HD Audio Controller [8086:7e28] (rev 20)
        Subsystem: Acer Incorporated [ALI] Device [1025:171f]
```

**Environment**
- Arch Linux `6.8.7-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Wed, 17 Apr 2024 15:20:00 +0000 x86_64 GNU/Linux`

lgirdwood, 2024-04-26:
@liquidev can you try installing v2.9 https://github.com/thesofproject/sof-bin/releases/tag/v2024.03

liquidev, 2024-04-26:
@lgirdwood I did try installing the latest release there before submitting this issue but to no avail. (Tried uninstalling the Arch-provided package and then using install.sh from the tarball, the effect was the same.)

I believe the Arch-provided package may simply be the same as the `v2024.03` tag in sof-bin. Unless you're saying I should try cloning the repo and installing from there instead of using the tarball.

liquidev, 2024-04-26:
Just tried installing v2.9 from the `git clone`d repo, no luck either. Same systemd journal logs as before.

lgirdwood, 2024-04-26:
> I've looked for this topology file but it doesn't seem to exist anywhere. There's only `sof-tplg/sof-hda-generic-idisp-2ch.tplg`, but not `sof-ace-tplg/sof-hda-generic-idisp-2ch.tplg`.

Can you try a soft link here so the kernel will find a similar file.

@ujfalusi @plbossart fyi - could the arch kernel be missing a kernel patch ?

ujfalusi, 2024-04-26:
It is a SDW device:
```
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: SoundWire enabled on CannonLake+ platform, using SOF driver
..
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: No SoundWire machine driver found
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: hda codecs found, mask 4
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
Apr 26 11:28:17 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: DMICs detected in NHLT tables: 2
...
Since no SDW machine driver was found the stack falls back to HDMI only topology, but indeed we only have the basic `sof-hda-generic-idisp.tplg` which does not support DMIC (and this machine has SDW+DMIC+HDMI).

The root cause is that we don't know this setup, so we fail to get the correct machine driver and the fallback to HDMI+DMIC does not work due to missing tplg.

@kv2019i, FYI

ujfalusi, 2024-04-26:
```
$ pacman -Qi sof-firmware | grep Version
Version         : 2024.03-1

$ pacman -Ql sof-firmware| grep idisp 
sof-firmware /usr/lib/firmware/intel/sof-ipc4-tplg/sof-hda-generic-idisp.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-hda-generic-idisp-2ch.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-hda-generic-idisp-4ch.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-hda-generic-idisp.tplg
```

liquidev, 2024-04-26:
Is there anything I can do to fix this (or help fix this?)

@lgirdwood I tried making the symlink but it's failing with a different error now:
```
Apr 26 15:41:41 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
Apr 26 15:41:41 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: SoundWire enabled on CannonLake+ platform, using SOF driver
Apr 26 15:41:41 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: enabling device (0000 -> 0002)
Apr 26 15:41:41 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
Apr 26 15:41:41 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
Apr 26 15:41:41 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: use msi interrupt mode
Apr 26 15:41:41 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: No SoundWire machine driver found
Apr 26 15:41:41 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: hda codecs found, mask 4
Apr 26 15:41:41 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
Apr 26 15:41:41 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: DMICs detected in NHLT tables: 2
Apr 26 15:41:41 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Firmware paths/files for ipc type 1:
Apr 26 15:41:41 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3:  Firmware file:     intel/sof-ipc4/mtl/sof-mtl.ri
Apr 26 15:41:41 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3:  Firmware lib path: intel/sof-ipc4-lib/mtl
Apr 26 15:41:41 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3:  Topology file:     intel/sof-ace-tplg/sof-hda-generic-idisp-2ch.tplg
Apr 26 15:41:41 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.9.0.1
Apr 26 15:41:41 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Booted firmware version: 2.9.0.1
Apr 26 15:41:41 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: sof_ipc4_parse_manifest: Invalid topology ABI size: 3
Apr 26 15:41:41 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: error: tplg component load failed -22
Apr 26 15:41:41 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed to load DSP topology -22
Apr 26 15:41:41 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
```

ujfalusi, 2024-04-26:
@liquidev, can this help you: https://github.com/thesofproject/linux/issues/4923

plbossart, 2024-04-26:
https://thesofproject.github.io/latest/getting_started/setup_linux/install_locally.html

plbossart, 2024-04-26:
Can you attach the results of this updated script (bash alsa-info.sh.txt), i'll help us check if this is really a SoundWire topology.

[alsa-info.sh.txt](https://github.com/thesofproject/sof/files/15132674/alsa-info.sh.txt)

liquidev, 2024-04-26:
Sure, here it is:

[alsa-info.txt](https://github.com/thesofproject/sof/files/15132793/alsa-info.txt)

I'll get to setting up a local kernel for testing in the meantime.

liquidev, 2024-04-26:
I'm afraid I can't set up a custom kernel by myself, it's beyond my technical ability at the moment. After a couple hours I managed to compile it successfully but then Arch's `mkinitcpio` fails to produce me a usable initramfs, which causes my root partition to not be loaded properly.

I just want my audio to work :confused:

marc-hb, 2024-04-26:
> I just want my audio to work 😕

From https://www.acer.com/us-en/laptops/swift/swift-go/pdp/NX.KR0AA.001

 Operating System: Windows 11

> I'm afraid I can't set up a custom kernel by myself, it's beyond my technical ability at the moment.

The only way not to have to go through all this pain is to buy a device that officially supports Linux. This is especially true with audio where there is a lot of hardware "creativity".

I hope you can find someone who can help you with this sort of kernel hacking.

ujfalusi, 2024-04-27:
@liquidev, you can try to follow this official Arch documentation: https://wiki.archlinux.org/title/Kernel/Traditional_compilation#Installation
So, something along these lines:
```
make modules
sudo make modules_install
make bzImage
sudo cp -v arch/x86/boot/bzImage /boot/vmlinuz-linuxSOF 
sudo cp /etc/mkinitcpio.d/linux.preset /etc/mkinitcpio.d/linuxSOF.preset
sudo vim /etc/mkinitcpio.d/linuxSOF.preset
# Append SOF to the linux name as described in 4.3.1
sudo mkinitcpio -p linuxSOF
# if you use GRUB, then:
sudo grub-mkconfig -o /boot/grub/grub.cfg
```
grub-mkconfig can break booting (been there several times myself, but 'easy' to recover) on a system where the grub install was done with older version of GRUB, if in doubt you might want to run grub-install prior mkconfig, see
https://wiki.archlinux.org/title/GRUB#Installation
https://wiki.gentoo.org/wiki/Handbook:AMD64/Installation/Bootloader#Install

Reboot, then select Advanced/Additional options or something and select linuxSOF to boot (keep fingers crossed).

liquidev, 2024-04-27:
@marc-hb Sorry if my comment came off as urging for this to get solved, that was not my intention. I too would like Linux to support more hardware configurations :)

I'm having another stab at it today and will probably make slow progress over the week with some support from a friend that's more savvy with using custom kernels on Arch. No hard promises, but fingers crossed I can get it to work and report something back here.

@ujfalusi I mostly followed the instructions described on that Arch wiki page, but when running `mkinitcpio -p linux-sof` (that's the name I assigned to the custom kernel) I'm getting an ominous warning:

```
==> Building image from preset: /etc/mkinitcpio.d/linux-sof.preset: 'default'
==> Using default configuration file: '/etc/mkinitcpio.conf'
  -> -k /boot/vmlinuz-linux-sof -g /boot/initramfs-linux-sof.img
==> Starting build: '6.9.0-rc3-sof-ga2a39cb962fa'
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [autodetect]
  -> Running build hook: [microcode]
  -> Running build hook: [modconf]
  -> Running build hook: [kms]
  -> Running build hook: [keyboard]
==> WARNING: Possibly missing firmware for module: 'xhci_pci'
  -> Running build hook: [keymap]
  -> Running build hook: [consolefont]
==> WARNING: consolefont: no font found in configuration
  -> Running build hook: [block]
  -> Running build hook: [filesystems]
  -> Running build hook: [fsck]
==> Generating module dependencies
==> Creating zstd-compressed initcpio image: '/boot/initramfs-linux-sof.img'
==> WARNING: errors were encountered during the build. The image may not be complete.
==> Building image from preset: /etc/mkinitcpio.d/linux-sof.preset: 'fallback'
==> Using default configuration file: '/etc/mkinitcpio.conf'
  -> -k /boot/vmlinuz-linux-sof -g /boot/initramfs-linux-sof-fallback.img -S autodetect
==> Starting build: '6.9.0-rc3-sof-ga2a39cb962fa'
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [microcode]
  -> Running build hook: [modconf]
  -> Running build hook: [kms]
  -> Running build hook: [keyboard]
==> WARNING: Possibly missing firmware for module: 'xhci_pci'
  -> Running build hook: [keymap]
  -> Running build hook: [consolefont]
==> WARNING: consolefont: no font found in configuration
  -> Running build hook: [block]
  -> Running build hook: [filesystems]
  -> Running build hook: [fsck]
==> Generating module dependencies
==> Creating zstd-compressed initcpio image: '/boot/initramfs-linux-sof-fallback.img'
==> WARNING: errors were encountered during the build. The image may not be complete.
```

and then the image fails to boot.

The way I set up my kernel was:

1. Cloned `thesofproject/kconfig` and `thesofproject/linux`
2. `cd linux`
3. `zcat /proc/config.gz > .config`
4. `../kconfig/kconfig-distro-sof-update.sh`, which prompted me for whether I'd like to install some drivers - I said no to all of them, because they seemed irrelevant to my hardware:
    ```
    BT8XX GPIO abuser (GPIO_BT8XX) [N/m/y/?] (NEW) n
    Analog Devices ADM1021 and compatibles (SENSORS_ADM1021) [N/m/y/?] (NEW) n
    Maxim MAX6642 sensor chip (SENSORS_MAX6642) [N/m/y/?] (NEW) n
    Studio Evolution SE6X (SND_SE6X) [N/m/?] (NEW) n
    ```
    It then proceeded to generate the config successfully (zero exit code.)
5. `make -j22`
6. `doas make modules_install`
7. `make bzImage`
8. `doas cp -v arch/x86/boot/bzImage /boot/vmlinuz-linux-sof`
9. `doas cp /etc/mkinitcpio.d/linux.preset /etc/mkinitcpio.d/linux-sof.preset`
10. Changed the kernel name in the `mkinitcpio` config just copied
11. `doas mkinitcpio -p linux-sof` - this step produced the logs above with that ominous "errors were encountered during the build" warning
12. Rebooted and chose the `linux-sof` kernel in rEF, which failed to boot due to not being able to find `/dev/nvme0n1p2` (my root partition) - it waited 20 seconds for the device to become available, but it never did.

Which leads me to believe the kernel may be missing some modules to mount my disk/partitions. I would debug further whether it's the drive or partitions, but the laptop keyboard is not working in the emergency shell. I might try an external keyboard later.

marc-hb, 2024-04-27:
> Sorry if my comment came off as urging for this to get solved, that was not my intention. 

It didn't :-)

> I too would like Linux to support more hardware configurations :)

The only way for this to happen at scale is for customers to _buy_ more Linux. Volunteering can only go so far. AND it saves all these hassles from a more "selfish" perspective.

> with some support from a friend that's more savvy with using custom kernels on Arch. 

Good idea.

>  I said no to all of them, because they seemed irrelevant to my hardware

In doubt just say "yes". It won't take much more time and unused Linux drivers cannot cause any harm. Optimize only after you have something working.

> Which leads me to believe the kernel may be missing some modules to mount my disk/partitions. 

Yes, looks like something missing in initramfs. Try `lsinitrd`, `lsinitramfs` or `lsinitcpio` and compare with a working image.

Have you considered the user-friendlier https://wiki.archlinux.org/title/Dracut ?

Also:

> https://wiki.archlinux.org/title/Kernel/Traditional_compilation#Installation
> This can be, depending on your background, more complicated than using the [Kernel/Arch build system](https://wiki.archlinux.org/title/Kernel/Arch_build_system). Consider the [Arch build system](https://wiki.archlinux.org/title/Arch_build_system) tools are developed and maintained to make repeatable compilation tasks efficient and safe.

ujfalusi, 2024-04-27:
@liquidev, the issue is at step 4
```
../kconfig/kconfig-distro-sof-update.sh
...
#
# configuration written to .config
#
using config: '.config'
*
* Restart config...
```
So kconfig is starting from a blank config, lacking vital drivers.
This might work better (not boot tested/built the kernel):
```
rm .config*
zcat /proc/config.gz > .config
make olddefconfig
scripts/kconfig/merge_config.sh .config ../kconfig/sof-defconfig
scripts/kconfig/merge_config.sh .config ../kconfig/soundwire-defconfig
make -j22
...
```
Check the sizes of the initrd images (`ls -al /boot/`), try to boot the linux-sof fallback if that is bigger.

liquidev, 2024-04-28:
> The only way for this to happen at scale is for customers to buy more Linux. Volunteering can only go so far. AND it saves all these hassles from a more "selfish" perspective.

That's what would happen in an ideal world, but from my own selfish perspective I really wanted to have a laptop with a nice, HiDPI OLED display in a fairly thin and light 14" form factor. I'm not aware of any Linux-friendly manufacturers offer that yet, not to mention living in Poland further limits my choices.

But let's cut that thread here to not go too offtopic, insted let's focus on getting audio to work on this darn thing :^)

---

I managed to get the kernel running. Turns out it's just that rEFInd's default heuristics for assigning initrds to kernels ended up confused and loaded the vanilla `linux` initrd into the `linux-sof` kernel. But now that the kernel booted, and with the topology from https://github.com/thesofproject/linux/issues/4923#issuecomment-2076802894 loaded into my `/lib/firmware/intel/sof-ace-tplg`, PipeWire actually sees an output - so that's a success \o/

In the journal, I get this, which includes a couple warnings, and then a whole slew of errors:
```
Apr 28 22:09:22 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
Apr 28 22:09:22 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: SoundWire enabled on CannonLake+ platform, using SOF driver
Apr 28 22:09:22 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: enabling device (0000 -> 0002)
Apr 28 22:09:22 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
Apr 28 22:09:22 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
Apr 28 22:09:22 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: use msi interrupt mode
Apr 28 22:09:22 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: DMICs detected in NHLT tables: 2
Apr 28 22:09:22 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: hda codecs found, mask 4
Apr 28 22:09:22 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Firmware paths/files for ipc type 1:
Apr 28 22:09:22 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3:  Firmware file:     intel/sof-ipc4/mtl/sof-mtl.ri
Apr 28 22:09:22 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3:  Firmware lib path: intel/sof-ipc4-lib/mtl
Apr 28 22:09:22 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3:  Topology file:     intel/sof-ace-tplg/sof-mtl-rt712-l0-2ch.tplg
Apr 28 22:09:22 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.9.0.1
Apr 28 22:09:22 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Booted firmware version: 2.9.0.1
Apr 28 22:09:22 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Topology: ABI 3:29:0 Kernel ABI 3:23:0
Apr 28 22:09:22 construct kernel: sof_sdw sof_sdw: ASoC: Parent card not yet available, widget card binding deferred
Apr 28 22:09:22 construct kernel: sof_sdw sof_sdw: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 3
Apr 28 22:09:22 construct kernel: input: sof-soundwire Headset Jack as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input18
Apr 28 22:09:22 construct kernel: input: sof-soundwire HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input19
Apr 28 22:09:22 construct kernel: input: sof-soundwire HDMI/DP,pcm=6 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input20
Apr 28 22:09:22 construct kernel: input: sof-soundwire HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input21
Apr 28 22:09:32 construct alsactl[706]: alsa-lib utils.c:364:(uc_mgr_config_load_into) could not open configuration file /usr/share/alsa/ucm2/sof-soundwire/rt712.conf
Apr 28 22:09:32 construct alsactl[706]: alsa-lib parser.c:78:(uc_mgr_config_load_file) error: failed to open file /usr/share/alsa/ucm2/sof-soundwire/rt712.conf: -2
Apr 28 22:09:32 construct alsactl[706]: Found hardware: "sof-soundwire" "Intel Meteor Lake HDMI" "HDA:8086281d,80860101,00100000  cfg-amp:1 mic:dmic cfg-mics:2 iec61937-pcm:7,6,5 hs:rt712-sdca spk:rt712" "" ""
Apr 28 22:10:25 construct pipewire[1282]: pw.node: (alsa_output.pci-0000_00_1f.3-platform-sof_sdw.pro-output-2-56) suspended -> error (Start error: Invalid argument)
Apr 28 22:10:25 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
Apr 28 22:10:25 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: DSP panic!
Apr 28 22:10:25 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
Apr 28 22:10:25 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: 0x50000005: module: ROM_EXT, state: FW_ENTERED, running
Apr 28 22:10:25 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Firmware state: 0x5, status/error code: 0x0
Apr 28 22:10:25 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Unknown toolchain is used
Apr 28 22:10:25 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: error: DSP Firmware Oops
Apr 28 22:10:25 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: error: Exception Cause: AllocaCause, MOVSP instruction, if caller’s registers are not in the register file
Apr 28 22:10:25 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: EXCCAUSE 0x00000005 EXCVADDR 0x00000000 PS       0x00060520 SAR     0x0000000c
[... cut ...]
Apr 28 22:10:26 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc4_tx_msg_unlocked: ipc message send for 0x12000000|0x0 failed: -19
Apr 28 22:10:26 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to free pipeline widget pipeline.0
Apr 28 22:10:26 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Failed to free connected widgets
```
The last few errors are repeated over and over again.

Now for the practical results - the good part is, the headphone jack works! \o/

The bad part is, the internal speakers don't. I haven't tested HDMI yet, will test it out in a minute.

liquidev, 2024-04-28:
Alright, I plugged the laptop into the TV in my living room, and HDMI audio is not working. In pavucontrol there is no distinct output port for HDMI:

[image]

Video is working fine, though I guess that's to be expected.

liquidev, 2024-04-28:
One weird thing that happens with the headphone jack is a two very loud audible clicks coming from my amplifier whenever there's a new audio source in PipeWire (or any other change in audio configuration, for that matter.) This doesn't happen with my other laptop so I'm thinking it might be a soundcard configuration issue.

Either way, here's a video demonstrating the issue:

https://github.com/thesofproject/sof/assets/16415678/52ecafce-c946-49d2-8989-624bddf0e005

Another odd issue is that when the amplifier is connected to the headphone jack but _off_ (in standby), the GNOME OSD for audio volume shows the headphone output getting muted as if you hit the Volume Down key combo repeatedly in a steady interval:

[Screencast from 2024-04-29 00-07-18.webm](https://github.com/thesofproject/sof/assets/16415678/ce85f94a-3129-4326-9be4-71dced2efa2c)

which actually confused the heck out of me when I first saw it. I presume it may be another soundcard-related issue. (On the video, I'm the counteracting the volume changes manually by holding down the appropriate Fn-key combination.)

bardliao, 2024-04-29:
@liquidev Can you run "sudo alsaucm reload" command and see if there is any error output?

ujfalusi, 2024-04-29:
The firmware panic is a ChainDMA start
```
ci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
Apr 28 22:10:26 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0xe030004|0xc00
Apr 28 22:10:26 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
Apr 28 22:10:26 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0x8e030004|0xc00|0x0, target: 0x1b0a0000|0x0|0x0, ctl: 0x3
Apr 28 22:10:26 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
Apr 28 22:10:26 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
```

To have more information the sof-dyndbg.conf  as described here https://github.com/thesofproject/linux/issues/4923#issuecomment-2052436089 and a full dmesg would probably give more insights.
Locally I cannot reproduce it.

ujfalusi, 2024-04-29:
@bardliao, UCM is looking for rt712.conf? There is not such a file in UCM, not even in git version...
```
Apr 28 22:09:32 construct alsactl[706]: alsa-lib utils.c:364:(uc_mgr_config_load_into) could not open configuration file /usr/share/alsa/ucm2/sof-soundwire/rt712.conf
Apr 28 22:09:32 construct alsactl[706]: alsa-lib parser.c:78:(uc_mgr_config_load_file) error: failed to open file /usr/share/alsa/ucm2/sof-soundwire/rt712.conf: -2
Apr 28 22:09:32 construct alsactl[706]: Found hardware: "sof-soundwire" "Intel Meteor Lake HDMI" "HDA:8086281d,80860101,00100000  cfg-amp:1 mic:dmic cfg-mics:2 iec61937-pcm:7,6,5 hs:rt712-sdca spk:rt712" "" ""
```

liquidev, 2024-04-29:
@bardliao Same thing as in the journal: 
```
ALSA lib utils.c:364:(uc_mgr_config_load_into) could not open configuration file /usr/share/alsa/ucm2/sof-soundwire/rt712.conf
ALSA lib parser.c:78:(uc_mgr_config_load_file) error: failed to open file /usr/share/alsa/ucm2/sof-soundwire/rt712.conf: -2
ALSA lib main.c:1554:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
alsaucm: error failed to open sound card hw:0: No such file or directory
```

@ujfalusi dmesg after booting with the configuration you requested: [dmesg.txt](https://github.com/thesofproject/sof/files/15146273/dmesg.txt) (just ran `dmesg > /tmp/dmesg.txt` as root)

Note this is with headphones plugged in; I can also provide the logs when no headphones are connected as well as when HDMI is connected if that makes a difference.

bardliao, 2024-04-29:
> @bardliao, UCM is looking for rt712.conf? There is not such a file in UCM, not even in git version...
>
> ```
> Apr 28 22:09:32 construct alsactl[706]: alsa-lib utils.c:364:(uc_mgr_config_load_into) [...]

Right, rt722.conf is missing in UCM. @shumingfan FYI

plbossart, 2024-04-29:
Just to me clear, this is a completely different layout compared to the Acer Swift Go 14 handled in https://github.com/thesofproject/sof/pull/9041 and https://github.com/thesofproject/linux/issues/4923

````

!!ACPI SoundWire Device Status Information
!!---------------

Realtek 0x000030025d071201

plbossart, 2024-04-29:
And there's something completely corrupted in the setup reported in alsa-info:

````
[    4.266357] sof-audio-pci-intel-mtl 0000:00:1f.3: No SoundWire machine driver found
[    4.266360] sof-audio-pci-intel-mtl 0000:00:1f.3: hda codecs found, mask 4
[    4.266362] sof-audio-pci-intel-mtl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    4.266364] sof-audio-pci-intel-mtl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[    4.269590] sof-audio-pci-intel-mtl 0000:00:1f.3: Firmware paths/files for ipc type 1:
[    4.269593] sof-audio-pci-intel-mtl 0000:00:1f.3:  Firmware file:     intel/sof-ipc4/mtl/sof-mtl.ri
[    4.269595] sof-audio-pci-intel-mtl 0000:00:1f.3:  Firmware lib path: intel/sof-ipc4-lib/mtl
[    4.269596] sof-audio-pci-intel-mtl 0000:00:1f.3:  Topology file:     intel/sof-ace-tplg/sof-hda-generic-idisp-2ch.tplg
[    4.270561] sof-audio-pci-intel-mtl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.9.0.1
[    4.383815] sof-audio-pci-intel-mtl 0000:00:1f.3: Booted firmware version: 2.9.0.1
[    4.405010] sof-audio-pci-intel-mtl 0000:00:1f.3: Topology: ABI 3:29:0 Kernel ABI 3:23:0
[    4.405051] sof-audio-pci-intel-mtl 0000:00:1f.3: error: can't connect DAI alh-copier.Playback-SimpleJack.0 stream Playback-SimpleJack
````

The last error means that the topology file for HDaudio was overidden with a topology file for SoundWire. Obviously this can't work. it's like putting diesel instead of regular gas in your car. Don't do it.

plbossart, 2024-04-29:
So the summary is that this device has TWO PCH-attached DMICs, and RT712 on link0. The topology used for Acer Swift Go 14 cannot be reused.

````
DMICs detected in NHLT tables: 2
Realtek 0x000030025d071201
````

shumingfan, 2024-04-30:
@liquidev Could you try to test the patch (https://github.com/shumingfan/alsa-ucm-conf/commit/395ed2a6cb70d992967cb756ca84e021caef025c)?

ujfalusi, 2024-04-30:
> @bardliao Same thing as in the journal:
>
> ```
> ALSA lib utils.c:364:(uc_mgr_config_load_into) could not open configuration file /usr/share/alsa/ucm2/sof-soundwire/rt712.conf
> ALSA lib [...]

This kernel log does not have the exception printed, how did it happened when it did?

Unfortunately it is missing the start of the boot (SDW boards tend to be pretty chatty on boot). Adding `log_buf_len=40M` to the kernel cmdline parameter list would allow the capture of the whole log.

marc-hb, 2024-04-30:
> Unfortunately it is missing the start of the boot (SDW boards tend to be pretty chatty on boot). Adding log_buf_len=40M to the kernel cmdline parameter list would allow the capture of the whole log.

Or, much simpler and better: use `journalctl -k` or `journalctl -b`. There's very rarely ever a reason to "downgrade" to `dmesg` instead.

liquidev, 2024-04-30:
@plbossart 

> And there's something completely corrupted in the setup reported in alsa-info

That does not occur in the patched kernel, see logs in https://github.com/thesofproject/sof/issues/9085#issuecomment-2081644342.

---

@ujfalusi @marc-hb Here's the journal from a fresh boot (I only opened Firefox to start typing out this comment, then ran `journalctl -k --this-boot` to fetch the logs): [journalctl-k-this-boot.txt](https://github.com/thesofproject/sof/files/15169766/journalctl-k-this-boot.txt)

---

@shumingfan I assume the patch you sent is supposed to get rid of the errors thrown by ALSA. Should I expect any other changes in behavior?

So far there seems to be no observable side effects. I haven't looked at the logs yet, but the journal logs I attached above are from a boot with your patch applied.

plbossart, 2024-04-30:
@liquidev at the dmesg/journalctl level there's nothing suspicious.

the alsa-ucm patch is to expose endpoints to PipeWire, this doesn't change anything at the journaltctl level. You would need to open the Gnome audio settings app to see if the endpoints are exposed.

You can alternatively test using aplay/arecord with the -Dhw: devices.

liquidev, 2024-04-30:
I don't see much difference in GNOME's audio settings, but here's the output of `aplay -l` in case you'd like to verify that the device list is what you'd expect:

```
**** List of PLAYBACK Hardware Devices ****
card 0: sofsoundwire [sof-soundwire], device 0: Jack Out (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsoundwire [sof-soundwire], device 2: Speaker (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsoundwire [sof-soundwire], device 5: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsoundwire [sof-soundwire], device 6: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsoundwire [sof-soundwire], device 7: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsoundwire [sof-soundwire], device 31: Deepbuffer Jack Out (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

To the uninitiated eye it would seem that three outputs should be present (jack, speakers, HDMI.) What are `HDMI2`, `HDMI3`, and `Deepbuffer Jack Out` for?

plbossart, 2024-04-30:
we can have 3 concurrent HDMI outputs, so that looks ok. you can safely ignore the deepbuffer stuff.

ujfalusi, 2024-05-02:
@liquidev, I'm a bit lost. Can you summarize what works (Speaker/Headset/HDMI) and what not? Also note the kernel, topology, ucm that is used.

Thanks

liquidev, 2024-05-03:
@ujfalusi What works:
- [ ] Speakers
- [x] Jack
- [ ] HDMI

```sh
$ uname -a
Linux construct 6.9.0-rc3-sof-2-sof #2 SMP PREEMPT_DYNAMIC Sat, 27 Apr 2024 20:41:01 +0000 x86_64 GNU/Linux
```
Compiled from thesofproject/linux `a2a39cb962fa4485287790fbfdc3d1f87cde9776`.

Topology file used can be found in the logs:
```
Apr 30 21:09:02 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Firmware paths/files for ipc type 1:
Apr 30 21:09:02 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3:  Firmware file:     intel/sof-ipc4/mtl/sof-mtl.ri
Apr 30 21:09:02 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3:  Firmware lib path: intel/sof-ipc4-lib/mtl
Apr 30 21:09:02 construct kernel: sof-audio-pci-intel-mtl 0000:00:1f.3:  Topology file:     intel/sof-ace-tplg/sof-mtl-rt712-l0-2ch.tplg
```

I'm not sure how to check the UCM config used, but previously the warnings mentioned `/usr/share/alsa/ucm2/sof-soundwire/rt712.conf` being missing. I applied @shumingfan's patches suggested in https://github.com/thesofproject/sof/issues/9085#issuecomment-2084591189 and the warnings are now gone, so I assume that's the config file used.

Let me know if you need anything else.

ujfalusi, 2024-05-03:
@liquidev, strange, HDMI should just work, the Speaker might needs some tweaking to let audio out.
--- For the `HDMI`, connect to an audio capable output, then
```
amixer -c0 contents | grep -A3 ELD 
```
Check which section have non 0 or empty `values` line and mark down the device=X value, in this example, it is 5:
```
numid=16,iface=PCM,name='ELD',device=5
```
now, use `aplay -Dhw:0,X -dat /dev/urandom` to play white(ish) noise.

--- For the `Speaker`
```
aplay -Dhw:0,2 -dat /dev/urandom
```
and in another terminal start `alsamixer -c0`
I usually do a monkey heuristic on the controls and unmute and push the volume up whatever I can have access to and see (hear) if it does anything.

You preferable want PW stopped, but if it is idle, that is fine also.

It might be that the UCM profile is not correct, it might be that the rt712 driver has some bugs, something else, but if playback 'works' but silent then it is most likely wither of the two.

liquidev, 2024-05-03:
@ujfalusi The trick with `aplay`/`alsamixer` managed to get audio coming out of the speakers :smile: 

I did
```
aplay -Dhw:0,2 -f s32 -c 2 /dev/urandom
```
then in another terminal `alsamixer -c0`, and had to unmute `rt712 FU06`:
[image]
and white noise ended up getting played through the speakers.

After unmuting that output in the mixer though, PipeWire still doesn't seem to be able to output to the speakers. Perhaps something in the PW config has to be tweaked?

---

As for HDMI, I'll try what you suggested later as I currently don't have time.