thesofproject/sof #9092: [BUG] sof-ctl blob format is not the same as with ALSA UCM, should align the sof-ctl format

singalsu, 2024-05-02:
**Describe the bug**

Sof-ctl supports ASCII text format uint32_t numbers data  (txt) and binary (blob) format. The binary blobs supported by UCM's cset-tlv need to have an additional 8 bytes header to work. The header contains for ALSA control API an index SOF_CTRL_CMD_BINARY (value 3) and size of following blob data without this header.

As result we need two kinds of binary blobs.

**To Reproduce**
Trying to cset-tlv an existing blob is not working. There is a fix in #9070 to produce new format binary blobs for UCM. But those blobs do not work with sof-ctl.

**Reproduction Rate**
Always

**Expected behavior**
Tools support for blobs without confusion to developers. Likely it would be best to change in sof-ctl both text and binary format data files to use the version with header from UCM.

The topology embedded format is same as sof-ctl's current format, no 8 bytes header. The topology blob format (.m4 or .conf ASCII hex bytes) should stay the same to avoid a compatibility break.

**Impact**
Annoyance.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

lgirdwood, 2024-05-02:
@singalsu is the intention to align sof-ctl with how cset-tlv works or the other way around ?

singalsu, 2024-05-03:
> @singalsu is the intention to align sof-ctl with how cset-tlv works or the other way around ?

There's already a lot of products mass with UCM's cset-tlv so we should follow it with sof-ctl (used only by developers).

kv2019i, 2024-09-05:
@singalsu Is this still planned for v2.11 (rc1 tag next Friday)?

singalsu, 2024-12-20:
I today started I to work with the feature to target 2.13, updating Octave scripts for blobs exports and update sof-ctl. The patch depends on #9749.

singalsu, 2025-01-28:
The fix is in #9760, now ready for review.

singalsu, 2025-02-13:
The sof-ctl patch is now merged, this can be closed!

closed 2025-02-13 by singalsu