thesofproject/sof #9095: [BUG]  Steam Deck OLED/Galileo sof-vangogh-nau8821-max DSP Panic with audio crash until full device reboot

matte-schwartz, 2024-05-03:
**Describe the bug**
On every recent kernel that I've tried (back to 6.5.0 from 6.8.7) with my Steam Deck OLED (Galileo), I have been able to replicate a DSP panic and snd_sof_amd_vangogh driver crash with the upstreamed portions for the OLED Deck's audio configuration. If you suspend the device while on desktop, try to switch the audio profile to something like Pro Audio, switch it back to SOF Vangogh, and then attempt to play a test sound, the audio driver will completely crash until you reboot the device. Relevant dmesg logs are in the console output section.

**To Reproduce**
from desktop: sleep -> resume -> change audio profile from SOF Vangogh to Pro Audio or Off -> change audio back to SOF Vangogh and attempt to play a test noise. should crash the entire audio driver until you reboot.

**Reproduction Rate**
With repro steps outlined above, 100%

**Expected behavior**
audio driver should not crash at any point, whether resuming from suspend or changing sessions.

**Impact**
only reliable fix is a device reboot, so I'd classify this as a showstopper when it appears. up until the audio crash, there are no other audio issues.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 6.8.7-201.fsync.fc39 which utilizes this additional kernel patch https://gist.github.com/matte-schwartz/fafad0a2a15c02e78fab21c348ff790f 
2) Name of the topology file
    * Topology: sof-vangogh-nau8821-max.tplg
    * Source: https://gitlab.com/evlaV/valve-hardware-audio-processing from git
3) Name of the platform(s) on which the bug is observed.
    * Platform: Steam Deck OLED/Galileo, confirmed bug on both Fedora and Arch

**Screenshots or console output**
```
Apr 26 16:44:30 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: ipc tx error for 0x60010000 (msg/reply size: 108/20): -22
Apr 26 16:44:30 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: HW params ipc failed for stream 1
Apr 26 16:44:30 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: ASoC: error at snd_soc_pcm_component_hw_params on 0000:04:00.5: -22
Apr 26 16:44:30 steamdeck-oled pipewire[7442]: spa.alsa: set_hw_params: Invalid argument
Apr 26 16:44:30 steamdeck-oled pipewire[7442]: pw.node: (alsa_output.pci-0000_04_00.5-platform-nau8821-max.HiFi__hw_sofnau8821max_1__sink-79) suspended -> error (Start error: Invalid argument)
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: ipc tx error for 0x60010000 (msg/reply size: 108/20): -22
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: HW params ipc failed for stream 1
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: ASoC: error at snd_soc_pcm_component_hw_params on 0000:04:00.5: -22
Apr 26 16:44:31 steamdeck-oled pipewire[7442]: spa.alsa: set_hw_params: Invalid argument
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: ------------[ DSP dump start ]------------
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: DSP panic!
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: fw_state: SOF_FW_BOOT_COMPLETE (7)
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: reason: runtime exception (0x6)
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: trace point: 0x0dead006
Apr 26 16:44:31 steamdeck-oled kernel: [56B blob data]
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: error: DSP Firmware Oops
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: EXCCAUSE 0x0000000d EXCVADDR 0xe010082c PS       0x00060f25 SAR     0x1bd7dffa
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: EPC1     0x20000fa7 EPC2     0x00000000 EPC3     0x20000a6e EPC4    0x00000000
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: EPC5     0x20000a6e EPC6     0x65ef7eff EPC7     0x00000000 DEPC    0x00000000
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: EPS2     0x00000000 EPS3     0x00060d20 EPS4     0x00000000 EPS5    0x00060d20
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: EPS6     0x028c2241 EPS7     0x00000000 INTENABL 0x000500a8 INTERRU 0x00000045
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: stack dump from 0xe00e0980
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0xe00e0980: 57487000 65efa008 6cb03800 c7a9f4f6
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0xe00e0990: 3fe8f53e 3ff97be8 000003ff e00e09a0
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0xe00e09a0: e00e0980 3ff97be8 9ecd9607 3fe8f53e
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0xe00e09b0: 3feaa370 df800000 3fe149f4 9f400000
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0xe00e09c0: 3ff93331 9364290b 00000000 41a00000
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0xe00e09d0: 20000fa7 40000e11 d4000000 40008d4f
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0xe00e09e0: df000000 4527d27d 3ff2e64b 204f65ec
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0xe00e09f0: 3feaa370 df800000 3fe149f4 9f400000
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: AR registers:
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0x0: ffffffff a46175e9 ffffffff a235dcf4
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0x10: ffffffff a2432fba ffffffff 00000000
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0x20: 00000000 57487000 65efa008 6cb03800
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0x30: c7a9f4f6 3fe8f53e 3ff97be8 000003ff
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0x40: e00e09a0 e00e0980 3ff97be8 9ecd9607
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0x50: 3fe8f53e 3feaa370 df800000 3fe149f4
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0x60: 9f400000 3ff93331 9364290b 00000000
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0x70: 41a00000 20000fa7 40000e11 d4000000
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0x80: 40008d4f df000000 4527d27d 3ff2e64b
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0x90: 204f65ec 3feaa370 df800000 3fe149f4
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0xa0: 9f400000 00000000 00000000 75f9b800
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0xb0: ffff8881 a5aad825 ffffffff f7e2fc00
[... 1 lines cut ...]
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0xd0: ffffffff a235dcf9 ffffffff f7e2fc00
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0xe0: e4e37cb7 6c245028 ffff8881 c4613140
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: 0xf0: ffffffff 00000003 00000000 00000000
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: ------------[ DSP dump end ]------------
Apr 26 16:44:31 steamdeck-oled pipewire[7442]: spa.alsa: set_hw_params: Connection timed out
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: ipc tx timed out for 0x30010000 (msg/reply size: 140/0)
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: ------------[ IPC dump start ]------------
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: dsp_msg = 0x0 dsp_ack = 0x0 host_msg = 0x0 host_ack = 0x0 irq_stat = 0x1
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: ------------[ IPC dump end ]------------
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: IPC timeout
Apr 26 16:44:31 steamdeck-oled kernel: snd_sof_amd_vangogh 0000:04:00.5: Failed to setup widget SMART_AMP2.0
```

Sorry, I tried to figure out sof-logger as well but I was unable to figure out how to extract info from it following the wiki instructions. I'm not too experienced with kernel/firmware debugging.

[... cut ...]
-  same exact topology file as SteamOS since we use valve-hardware-audio-processing
-  config files match up
-  no missing kernel patches
-  confirmed this issue is NOT present on 6.1.52-valve16 but IS present on Valve's latest linux-integration branch, 6.5.0-valve, which makes me think it's either a difference in audio configuration that I am missing or a regression of some sort changed the way the audio configuration behaves on later kernels.

any guidance you have would be much appreciated, and sorry for the long-winded explanation. Thanks!

lgirdwood, 2024-05-03:
@thesofproject/amd  looks like its a panic/assert with hw_params.
@matte-schwartz anyway you can dump the HW params being used by ALSA, it looks like the configuration is fatal on thid device (it should not assert, but return an error up the stack).

Venkata-Prasad-Potturu, 2024-05-03:
Hi matte-schwartz,
We are working on this and will come back soon.

matte-schwartz, 2024-05-03:
> @thesofproject/amd looks like its a panic/assert with hw_params. @matte-schwartz anyway you can dump the HW params being used by ALSA, it looks like the configuration is fatal on this device (it should not assert, but return an error up the stack).

```
matt@steamdeck-oled:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Generic [HD-Audio Generic], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: Generic [HD-Audio Generic], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: Generic [HD-Audio Generic], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: Generic [HD-Audio Generic], device 9: HDMI 3 [HDMI 3]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sofnau8821max [sof-nau8821-max], device 0: I2SSP (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sofnau8821max [sof-nau8821-max], device 1: I2SHS (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sofnau8821max [sof-nau8821-max], device 2: I2SBT (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

```
matt@steamdeck-oled:~$ aplay -D hw:1,0 --dump-hw-params /dev/zero
Playing raw data '/dev/zero' : Unsigned 8 bit, Rate 8000 Hz, Mono
HW Params of device "hw:1,0":
--------------------
ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD MSBITS_MAX
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: [1000 85334)
PERIOD_SIZE: [48 4096]
PERIOD_BYTES: [192 16384]
PERIODS: [2 16]
BUFFER_TIME: [2000 341334)
BUFFER_SIZE: [96 16384]
BUFFER_BYTES: [384 65536]
TICK_TIME: ALL
--------------------
aplay: set_params:1387: Sample format non available
Available formats:
- S16_LE
```
let me know if these are not the hw-params you meant.

Venkata-Prasad-Potturu, 2024-05-20:
Hi matte-schwartz,
I'm unable to reproduce the issue with below kernel version in Steam Deck device. 
Working with Valve on this issue, we will get back to you soon.

(root@steamdeck deck)# uname -a
Linux steamdeck **_6.5.0-valve5-1-neptune-65-g6efe817cc486_** #1 SMP PREEMPT_DYNAMIC Mon, 20 May 2024 05:53:14 +0000 x86_64 GNU/Linux
(B+)(root@steamdeck deck)# 

Thanks.

matte-schwartz, 2024-05-20:
```
(deck@steamdeck ~)$ uname -a
Linux steamdeck 6.5.0-valve5-1-neptune-65-g6efe817cc486 #1 SMP PREEMPT_DYNAMIC Thu, 09 May 2024 21:09:52 +0000 x86_64 GNU/Linux
```
definitely still happens for me here following the repro steps outlined above. attaching a video from SteamOS Main along with [dmesg](https://gist.github.com/matte-schwartz/fc71d9115b8b928f4cecd7af13e68fe9) taken after the audio crash

video: 

https://github.com/thesofproject/sof/assets/136293710/51262ead-732f-4090-9107-57eb55a540dc

matte-schwartz, 2024-05-23:
I noticed [this](https://gitlab.com/evlaV/linux-integration/-/commit/2f42ba23d32f37ac2f283672593607b8b089bee5) commit in Valve's linux-integration which seems to print helpful information about this specific sof error. I built a 6.9.1 kernel including this patch and installed it on my OLED Deck. I am including a dmesg dump for sof logging I took when I triggered a DSP Panic, this time with the extra debugging info from Valve's patch

[sof_panic.log](https://github.com/thesofproject/sof/files/15410288/sof_panic.log)

hopefully this is helpful to you as well

lgirdwood, 2024-06-18:
@Venkata-Prasad-Potturu is this resolved ? can it be closed ?

matte-schwartz, 2024-06-18:
the issue has been resolved in the firmware that Valve provides via `linux-firmware-neptune` on https://steamdeck-packages.steamos.cloud/archlinux-mirror/jupiter-main/os/x86_64/?C=M&O=D 

unless the upstream firmware will be different, this can probably be closed.

bhiregoudar, 2024-06-20:
> @Venkata-Prasad-Potturu is this resolved ? can it be closed ?

Yes, please close the issue

closed 2024-06-20 by matte-schwartz