thesofproject/sof #9101: [BUG] IPC3/4 fuzzer build broken with Zephyr PR 71378

kv2019i, 2024-05-07:
**Describe the bug**
As discussed in https://github.com/zephyrproject-rtos/zephyr/pull/71378#pullrequestreview-2011374611 , recent changes in Zephyr mainline break the SOF fuzzer builds and the github CI checks using these builds.

**To Reproduce**
https://github.com/thesofproject/sof/actions/runs/8972185189/job/24639608648?pr=9098

**Reproduction Rate**
100%

**Expected behavior**
Fuzzer build works.

**Impact**
Fuzzer not usable with SOF.

**Environment**
https://sof-ci.01.org/sofpr/PR9098/build4470/build/index.html

kv2019i, 2024-05-07:
Marking as P1 for now as we have the fuzzer tests in our github PR test flow.

andyross, 2024-05-07:
Hm.  That PR's original version was breaking, but the refactored one was safe I thought.  Looks like it's just a missing kconfig we can replace upstream, give me a moment to try it out.

andyross, 2024-05-07:
Confirmed that reverting the PR works to recover a fuzz build: `git revert --no-edit 2bff629d296f...82fa83048e66`

A few things got moved to the sample that were part of the core fuzz rig and I missed them in review.  Trying to decide whether it's best to move those back piecewise upstream, or just add them to SOF as that's the new architecture anyway.  (The end result here is that the "outer loop" in the fuzzer becomes part of the app and not the architecture layer, in preparation for deprecating "native_posix" in favor of the equivalent "native_sim" arch which is based on an out of tree project).

andyross, 2024-05-07:
First cut at a fix up, @marc-hb I'm not 100% sure how to exercise all the CI cases (though... maybe CI will just run them on the PR, heh) so wouldn't hurt for you to give it a spin.

PR #9107 (2024-05-09): platform/posix: Unbreak fuzzer support
Upstream Zephyr moved the LLVM fuzzer entry point out of the arch layer and made it an app responsibility, so we broke.  Add back the support here that got removed.

Fixes #9101
commit e7f76eec (2024-05-09): platform/posix: Unbreak fuzzer support

Upstream Zephyr moved the LLVM fuzzer entry point out of the arch
layer and made it an app responsibility, so we broke.  Add back the
support here that got removed.

Fixes #9101
closed 2024-05-09 by kv2019i