thesofproject/sof #9123: LNL NOCODEC alsa-bat failure

plbossart, 2024-05-14:
Failure reported in 
https://sof-ci.01.org/linuxpr/PR4899/build2589/devicetest/index.html?model=LNLM_RVP_NOCODEC&testcase=check-alsabat-headset-playback-821

This could be the consequence of a xrun.

plbossart, 2024-05-20:
Now we have a broken test with NOCODEC alsa-bat capture

https://sof-ci.01.org/linuxpr/PR5007/build3042/devicetest/index.html?model=LNLM_RVP_NOCODEC&testcase=check-alsabat-headset-capture-821

I must admit I don't know if what the difference is between alsa-bat-headset-playback and alsa-bat-headset-capture, we need to play and capture on the same devices and test the same loopback

playback:
https://sof-ci.01.org/linuxpr/PR5007/build3042/devicetest/index.html?model=LNLM_RVP_NOCODEC&testcase=check-alsabat-headset-playback-821
2024-05-20 19:59:03 UTC [REMOTE_COMMAND] alsabat -Phw:sofnocodec,0 --standalone -n 240000 -r 48000 -c 2 -f S16_LE -F 821 -k 2.1
2024-05-20 19:59:04 UTC [REMOTE_COMMAND] alsabat -Chw:sofnocodec,0 -c 2 -r 48000 -f S16_LE -F 821 -k 2.1

capture:
https://sof-ci.01.org/linuxpr/PR5007/build3042/devicetest/index.html?model=LNLM_RVP_NOCODEC&testcase=check-alsabat-headset-capture-821
2024-05-20 20:00:49 UTC [REMOTE_COMMAND] alsabat -Phw:sofnocodec,0 --standalone -n 240000 -r 48000 -c 2 -f S16_LE -F 821 -k 2.1
2024-05-20 20:00:51 UTC [REMOTE_COMMAND] alsabat -Chw:sofnocodec,0 -c 2 -r 48000 -f S16_LE -F 821 -k 2.1

Looks the same to me? What am I missing @fredoh9 @marc-hb @kv2019i ?

kv2019i, 2024-05-21:
@plbossart At least in the capture case, there is again glitches in capture audio. In specific case of 
https://sof-ci.01.org/linuxpr/PR5007/build3042/devicetest/index.html?model=LNLM_RVP_NOCODEC&testcase=check-alsabat-headset-playback-821
... it's smaller than before and seems to be exactly one 1ms tick of audio missing:

[image]

marc-hb, 2024-05-22:
> Looks the same to me? What am I missing @fredoh9 @marc-hb @kv2019i ?

@fredoh9 just explained this to me: NOCODEC is special.

Outside NOCODEC:
- alsabat-playback: SOF playback, USB capture
- alsabat-capture:  USB playback, SOF capture

NOCODEC is special, it has no USB. So:
- SOF playback, SOF capture.

So both tests do more or less the same thing on NOCODEC.

plbossart, 2024-05-22:
Right, so can we skip the capture part to avoid unnecessary confusions?

lgirdwood, 2024-05-22:
> ... it's smaller than before and seems to be exactly one 1ms tick of audio missing:

and its also suspiciously close to 0.5 seconds of audio. 0.5 second USB audio buffer size ???

plbossart, 2024-05-22:
> > ... it's smaller than before and seems to be exactly one 1ms tick of audio missing:
> 
> and its also suspiciously close to 0.5 seconds of audio. 0.5 second USB audio buffer size ???

There's no USB involved in Nocodec mode, it's an SSP loopback.

marc-hb, 2024-05-22:
> Right, so can we skip the capture part to avoid unnecessary confusions?

"send patches"

marc-hb, 2024-05-22:
Just spotted a capture failure on HDA

https://sof-ci.01.org/sofpr/PR9151/build4789/devicetest/index.html?model=LNLM_RVP_HDA&testcase=check-alsabat-headset-capture-997

```
2024-05-22 10:38:13 UTC [REMOTE_COMMAND] alsabat -Phw:CODEC,0 --standalone -n 240000 -r 48000 -c 2 -f S16_LE -F 997 -k 2.1
2024-05-22 10:38:14 UTC [REMOTE_COMMAND] alsabat -Chw:sofhdadsp,0 -c 2 -r 48000 -f S16_LE -F 997 -k 2.1
 FAIL: Peak freq too low 973.39 Hz
 FAIL: Peak freq too low 974.85 Hz
 FAIL: Peak freq too low 977.05 Hz
 FAIL: Peak freq too low 978.52 Hz
 FAIL: Peak freq too low 979.98 Hz
 FAIL: Peak freq too low 982.18 Hz
 FAIL: Peak freq too low 983.64 Hz
 FAIL: Peak freq too low 987.30 Hz
 FAIL: Peak freq too low 988.77 Hz
alsa-utils version 1.2.6

Entering capture thread (ALSA).
Get period size: 3000  buffer size: 24000
Recording ...
Capture completed.

BAT analysis: signal has 65536 frames at 48000 Hz, 2 channels, 2 bytes per sample.

Channel 1 - Checking for target frequency 997.00 Hz
Amplitude: 9878.5; Percentage: [30]
Detected peak at 973.39 Hz of 17.57 dB
 Total 17.6 dB from 973.39 to 973.39 Hz
Detected peak at 974.85 Hz of 17.64 dB
 Total 20.6 dB from 974.85 to 974.85 Hz
Detected peak at 977.05 Hz of 17.74 dB
 Total 22.4 dB from 977.05 to 977.05 Hz
Detected peak at 978.52 Hz of 18.62 dB
 Total 23.9 dB from 978.52 to 978.52 Hz
Detected peak at 979.98 Hz of 18.92 dB
 Total 25.1 dB from 979.98 to 979.98 Hz
Detected peak at 982.18 Hz of 18.84 dB
 Total 26.7 dB from 981.45 to 982.18 Hz
Detected peak at 983.64 Hz of 19.88 dB
 Total 27.5 dB from 983.64 to 983.64 Hz
Detected peak at 987.30 Hz of 20.82 dB
 Total 29.8 dB from 985.11 to 987.30 Hz
Detected peak at 988.77 Hz of 22.17 dB
 Total 30.5 dB from 988.77 to 988.77 Hz
Detected peak at 996.83 Hz of 34.94 dB
 Total 41.0 dB from 990.23 to 1004.15 Hz
 PASS: Peak detected at target frequency
Detected at least 10 signal(s) in total

Return value is -1003
```

plbossart, 2024-05-22:
> Just spotted a capture failure on HDA

different interface, different issue please

wszypelt, 2024-05-24:
@plbossart can I ask for the logs with ipc payload?

plbossart, 2024-05-24:
In theory the dmesg log should already have the entire payload provided, an option must be missing in the CI setup.
@fredoh9 @marc-hb @ssavati do we turn the BIT(11) on for sof_debug?

````
#define SOF_DBG_DUMP_IPC_MESSAGE_PAYLOAD	BIT(11) /* On top of the IPC message header
							 * dump the message payload also
````

fredoh9, 2024-05-28:
All LNL RVPs in SOF CI have the IPC dump enabled.

marc-hb, 2024-05-28:
> do we turn the BIT(11) on for sof_debug?

It's hit and miss because:
- we haven't productized Ansible yet, so it breaks often when someone reserves a device.
- we can't set bits one by one and we don't have "official" values documented anywhere yet.

marc-hb, 2024-05-29:
According to @kv2019i 👍 

> "1" for others, and 3073/0x0c01 for nocodec .

fredoh9, 2024-05-29:
need to keep bit(11) enabled.
- nocodec mode 0xC01
- Others 0x801

ssavati, 2024-05-29:
> All LNL RVPs in SOF CI have the IPC dump enabled.
i have checked all BA devices also for IPC dump. its enabled with above values

plbossart, 2024-05-29:
IIRC we can increase the MAX_STRLEN. I worked on this months ago and of course don't remember any details.
There was also a buffer dump option but that has some dependencies.

plbossart, 2024-05-29:
@mathias-nyman has a suggestion to get a large string, see
https://github.com/bpftrace/bpftrace/issues/305#issuecomment-1737129371

marc-hb, 2024-05-29:
> I have this semi working implementation:

This seems very interesting - and 100% off-topic. Please file a new issue.

ujfalusi, 2024-05-30:
@plbossart, OK, I have this now wich can print up to 500 lines of 16 bytes lines out:
```
#!/usr/bin/bpftrace
   
struct sof_ipc4_msg {
	union {
		unsigned long header_u64;
		struct {
			unsigned int primary;
			unsigned int extension;
		};
	};

	unsigned long data_size;
	void *data_ptr;
};
 
BEGIN
{
	printf("SOF IPC4 message logging. Ctrl-C to end.\n");
}

kprobe:sof_ipc4_log_header {
	$msg = (struct sof_ipc4_msg *)arg2;

	if (arg3 == 1 && $msg->data_size != 0) {
		printf("%s : 0x%x|0x%x [data size:: %llu]\n", str(arg1),
		       $msg->primary, $msg->extension, $msg->data_size);
		if (!strcontains(str(arg1), "done")) {
			$count = (int64) $msg->data_size;
			$ptr = (uint8*) $msg->data_ptr;
			$line = 0;

			printf("Message payload:\n");
			while ($line < 500) {
				if ($count <= 16) {
					printf("%rh\n", buf($ptr, $count));
					break;
				}
				printf("%rh\n", buf($ptr, 16));
				$count -= 16;
				if ($count == 0) {
					break;
				}
				$ptr += 16;
				$line++;
			}

		} else {
			printf("\n");
		}
	} else {
		printf("%s : 0x%x|0x%x\n", str(arg1), $msg->primary,
		       $msg->extension);
		if (strcontains(str(arg1), "done")) {
			printf("\n");
		}
	}	
}
```
I'm not sure how to format it, should we drop the `* done` and `ipc tx reply :` lines (the reply print indicates a fail for example)? Is it OK to print the payload as I do, as bytes?

Example output
```
sof-up-xtreme:[root]:~# ./ipc4_msg_trace.bt
Attaching 2 probes...
SOF IPC4 message logging. Ctrl-C to end.
ipc rx       : 0x1b080000|0x0
ipc rx done  : 0x1b080000|0x0

[... 324 lines cut ...]
ipc rx       : 0x1b060000|0x0
ipc rx done  : 0x1b060000|0x0

ipc rx       : 0x1b060000|0x0
ipc rx done  : 0x1b060000|0x0

ipc tx reply : 0x32000000|0x0
ipc tx done  : 0x12010000|0x0

ipc tx       : 0x44000000|0x3060004c [data size:: 76]
Message payload:
00 01 00 00 00 10 00 00 00 00 00 00 ef 01 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00
ipc tx reply : 0x64000000|0x3060004c
ipc tx done  : 0x44000000|0x3060004c [data size:: 76]

ipc tx       : 0x47000000|0x0 [data size:: 8]
Message payload:
01 00 00 00 00 00 00 00
ipc tx reply : 0x67000000|0x0
ipc tx done  : 0x47000000|0x0 [data size:: 8]
```

plbossart, 2024-05-30:
Nice!

For the size, don't we have a limit of 4k per IPC?

Also we should ask our firmware friends if they have a preference for the format, if their tools can eat our traces directly that would be great.

kv2019i, 2024-05-30:
Just a note: given reproduction rate is low (so getting the logs without CI is not so easy) and this is a P1 item, can we enable the old-style IPC dump on the DUTs where the error is seen, so we can move this bug forward ASAP?

kv2019i, 2024-06-03:
We have a winner, finally hit the bug on a machine with IPC payload dump enabled:
https://sof-ci.01.org/linuxpr/PR5034/build3319/devicetest/index.html?model=LNLM_RVP_NOCODEC&testcase=check-alsabat-nocodec-32bits-599

FYI @wszypelt @abonislawski

softwarecki, 2024-06-07:
@kv2019i Payload includes the creation of 4 different pipelines: playback on SSP, capture on SSP and twice playback on ssp with loopback from SRC. On which topology is the issue was reproduced?

marc-hb, 2024-06-07:
> On which topology is the issue was reproduced?

Each page of test results shows the topology filename in the second line of the logs.

danobi, 2024-06-10:
FWIW, I have a draft PR up for big string support in bpftrace: https://github.com/bpftrace/bpftrace/pull/3228

Still some stuff to do, but I expect when it lands it'll unblock the majority of use cases.

lgirdwood, 2024-06-12:
@softwarecki did you manage to unblock and reproduce ?

kv2019i, 2024-06-14:
While testing other issues on LNL and alsabat, I tried to locally to reproduce this case on a NOCODEC machine, but could not hit the error in 3 hour repeated run of the failing alsabat case (750/750 successful iteartons). SOF commit 1470e6c58529 .  I also checked Intel daily CI results for last 7 days and did not see any occurences of this bug anymore.

softwarecki, 2024-06-14:
@lgirdwood We are still working on reproduce

kv2019i, 2024-06-20:
Issue seen on a kernel PR test run with a fresh SOF FW version (https://github.com/thesofproject/sof/commit/3da8e6474531411bef64819113a29a7edbd51fcf ):

https://sof-ci.01.org/linuxpr/PR5075/build3762/devicetest/index.html
https://sof-ci.01.org/linuxpr/PR5075/build3762/devicetest/LNLM_RVP_NOCODEC/check-alsabat-headset-capture-599/bat.wav.r9zwcb

FYI @softwarecki

kv2019i, 2024-06-25:
@softwarecki can you post a quick update. I believe you've identified the cause of failure and are working on a patch.

softwarecki, 2024-06-26:
I reproduced the issue and prepared python tests. The issue is related to the fact that when the capture pipeline is running, the playback pipeline is deleted and then created again. The precise trouble occurs in the copier component in capture pipeline, which sends the capture data to the smart amp reference pin. If it is unbinded and next binded again, glitches appears in the component's output. I'm working on finding the issue in the sof code and fixing it.

kv2019i, 2024-06-28:
Reproduction rate low in past week, continue with P2 and assign to v2.11.

marc-hb, 2024-07-02:
> Reproduction rate low in past week, continue with P2 and assign to v2.11.

This just happened again in CI:

https://sof-ci.01.org/softestpr/PR1213/build576/devicetest/index.html?model=LNLM_RVP_NOCODEC&testcase=check-alsabat-headset-capture-821

Have you made sure the kernel, firmware and system configuration is 100% identical? Even better: have you just downloaded and installed packages from CI directly? On the same system(s).

marc-hb, 2024-07-15:
To be closed after one round of daily tests.

marc-hb, 2024-07-18:
This just failed again:
https://sof-ci.01.org/sofpr/PR9313/build6571/devicetest/index.html?model=LNLM_RVP_NOCODEC&testcase=check-alsabat-nocodec-32bits-997

softwarecki, 2024-07-19:
@marc-hb I don't see any glitches in the attached wave file
[image]

only the peak at 1000 Hz is visible in fft
[image]

I think this is a different issue than the one I fixed.

wszypelt, 2024-08-23:
@marc-hb Could you please take a look at the @softwarecki respone?

lgirdwood, 2024-08-23:
@ssavati @fredoh9 do we have the non USB loopback connectors enabled yet ?

marc-hb, 2024-08-23:
> I don't see any glitches in the attached wave file

I don't know anything about the details sorry but I remember that in the past every time a "human eye" and alsabat disagreed, alsabat won eventually. I have no idea what caused this particular failure: it could be a LNL-specific USB bug, or a LNL-specific audio bug, or even something else but all the evidence so far has shown alsabat itself to be the least suspect.

fredoh9, 2024-08-24:
> @ssavati @fredoh9 do we have the non USB loopback connectors enabled yet ?

I haven't purchased the loopback cables yet.

kv2019i, 2024-09-09:
Reviewing a week worth of daily test plans with this configurationg (starting  2024-09-02), this test is passing all the time. It is likely https://github.com/zephyrproject-rtos/zephyr/pull/77805 fix was responsible for the occasional 1ms glitches seen in this test and those are now gone. There is one failure left in the alsa-bat-capture test, I'll file a separate bug to track that.

PR #9276 (2024-07-15): copier: Add bind function to configure source/sink buffers params
Setting the source/sink buffers parameters in the copier_update_params function is not sufficient. If a sink buffer is attached during copier operation, the module will not set buffers parameters. Move the buffer parameter configurations to the new copier_bind bind function. Delete the sink buffers configurations from the copier_module_copy function. There is no need to configure sink buffers parameters on each copy. We are assured that they were configured at the time of bind.

Fixes: #9123
commit 5c725697 (2024-07-15): copier: Add bind function to configure sink buffers params

Setting the sink buffers parameters in the copier_update_params function is
not sufficient. If a sink buffer is attached during copier operation, the
module will not set buffers parameters. Add bind function to configure sink
buffers parameters.

There is no need to configure sink buffers parameters on each copy. We are
assured that they were configured at the time of bind. Remove
ipc4_update_buffer_format from the copier_module_copy function.

Fixes: #9123
closed 2024-09-09 by kv2019i