thesofproject/sof #9128: [BUG] Fail to compile topology

dbaluta, 2024-05-16:
**Describe the bug**

Compiling topology on latest main fails (`fdb67ee78 (HEAD -> main, origin/main, origin/HEAD) west.yml: update Zephyr to e97d33d0c896)`

**To Reproduce**

# Run
```
$ ./scripts/build-tools.sh -T
```
[2/257] Generating sof-hda-generic-cavs25-2ch-48k-96k.tplg
ALSA lib confeval.c:272:(snd_config_evaluate_string) wrong expression '$[($in_channels * ($[($in_rate + 999)] / 1000)) * ($in_bit_depth / 8)]'
ALSA lib pre-process-object.c:1866:(tplg_build_object) Failed to evaluate attributes ibs in input_audio_format, from '$[($in_channels * ($[($in_rate + 999)] / 1000)) * ($in_bit_depth / 8)]'
ALSA lib pre-process-object.c:2066:(tplg_pre_process_objects) Error building object Base.input_audio_format.0
ALSA lib pre-process-object.c:1928:(tplg_build_object) error processing child objects in object data
ALSA lib pre-process-object.c:2066:(tplg_pre_process_objects) Error building object Widget.dai-copier.1
ALSA lib pre-process-object.c:1928:(tplg_build_object) error processing child objects in object `���U
ALSA lib pre-process-object.c:2066:(tplg_pre_process_objects) Error building object Pipeline.mixout-gain-efx-dai-copier-playback.0
```

**Reproduction Rate**
Always

**Expected behavior**

Topology should compile fine

dbaluta, 2024-05-16:
Cc: @ranj063  @marc-hb

marc-hb, 2024-05-16:
> error processing child objects in object `���U

This corruption looks like a bug in `alsatplg`. Maybe it was fixed in 1.2.11?

plbossart, 2024-05-16:
yeah, I don't think using the native alsa-lib is likely to work, try with docker.

dbaluta, 2024-05-17:
Works fine with

$ alsatplg --version
alsatplg version 1.2.11
libasound version 1.2.11
libatopology version 1.2.11

Sorry for the noise

closed 2024-05-17 by dbaluta