thesofproject/sof #9164: LNL HDA alsabat capture failed, "Peak freq too low"

marc-hb, 2024-05-23:
Just spotted a capture failure on HDA

https://sof-ci.01.org/sofpr/PR9151/build4789/devicetest/index.html?model=LNLM_RVP_HDA&testcase=check-alsabat-headset-capture-997

```
2024-05-22 10:38:13 UTC [REMOTE_COMMAND] alsabat -Phw:CODEC,0 --standalone -n 240000 -r 48000 -c 2 -f S16_LE -F 997 -k 2.1
2024-05-22 10:38:14 UTC [REMOTE_COMMAND] alsabat -Chw:sofhdadsp,0 -c 2 -r 48000 -f S16_LE -F 997 -k 2.1
 FAIL: Peak freq too low 973.39 Hz
 FAIL: Peak freq too low 974.85 Hz
 FAIL: Peak freq too low 977.05 Hz
 FAIL: Peak freq too low 978.52 Hz
 FAIL: Peak freq too low 979.98 Hz
 FAIL: Peak freq too low 982.18 Hz
 FAIL: Peak freq too low 983.64 Hz
 FAIL: Peak freq too low 987.30 Hz
 FAIL: Peak freq too low 988.77 Hz
alsa-utils version 1.2.6

Entering capture thread (ALSA).
Get period size: 3000  buffer size: 24000
Recording ...
Capture completed.

BAT analysis: signal has 65536 frames at 48000 Hz, 2 channels, 2 bytes per sample.

Channel 1 - Checking for target frequency 997.00 Hz
Amplitude: 9878.5; Percentage: [30]
Detected peak at 973.39 Hz of 17.57 dB
 Total 17.6 dB from 973.39 to 973.39 Hz
Detected peak at 974.85 Hz of 17.64 dB
 Total 20.6 dB from 974.85 to 974.85 Hz
Detected peak at 977.05 Hz of 17.74 dB
 Total 22.4 dB from 977.05 to 977.05 Hz
Detected peak at 978.52 Hz of 18.62 dB
 Total 23.9 dB from 978.52 to 978.52 Hz
Detected peak at 979.98 Hz of 18.92 dB
 Total 25.1 dB from 979.98 to 979.98 Hz
Detected peak at 982.18 Hz of 18.84 dB
 Total 26.7 dB from 981.45 to 982.18 Hz
Detected peak at 983.64 Hz of 19.88 dB
 Total 27.5 dB from 983.64 to 983.64 Hz
Detected peak at 987.30 Hz of 20.82 dB
 Total 29.8 dB from 985.11 to 987.30 Hz
Detected peak at 988.77 Hz of 22.17 dB
 Total 30.5 dB from 988.77 to 988.77 Hz
Detected peak at 996.83 Hz of 34.94 dB
 Total 41.0 dB from 990.23 to 1004.15 Hz
 PASS: Peak detected at target frequency
Detected at least 10 signal(s) in total

Return value is -1003
```

_Originally posted by @marc-hb in https://github.com/thesofproject/sof/issues/9123#issuecomment-2125449102_

marc-hb, 2024-05-23:
Again https://sof-ci.01.org/softestpr/PR1200/build467/devicetest/index.html?model=LNLM_RVP_HDA&testcase=check-alsabat-headset-capture-821

plbossart, 2024-05-23:
@fredoh9 how many jack connectors do we have on the RVP? is the same 3.5mm jack connector for HDaudio and SoundWire reworks?

marc-hb, 2024-05-23:
> @fredoh9 how many jack connectors do we have on the RVP?

Only one jack. I don't remember _any_ RVP with more than one jack.

> is the same 3.5mm jack connector for HDaudio and SoundWire reworks?

I don't know. I looked at `jf-lnlm-rvp-hda-1` and the USB loopback goes through the AIOC board, the RVP jack is _not_ connected. On the other hand, this device is not enabled in CI right now :-( afraid we have to wait for @fredoh9 for this one.

marc-hb, 2024-05-23:
I just noticed someone disabled ba-lnlm-rvp-hda-02, on which the two failures were found...

fredoh9, 2024-05-28:
@plbossart We have only one jack in the RVP, i don't remember any RVP with multiple JACK.

@marc-hb For LNLM_RVP_HDA, the RVP doesn't have HDA codec, so we attached external AIOC. Hence the jack is in external AIOC is being used.

marc-hb, 2024-05-29:
> I just noticed someone disabled ba-lnlm-rvp-hda-02

From @ssavati : silicon being upgraded.

ssavati, 2024-05-29:
@marc-hb now ba-lnlm-rvp-hda-02 board is up with silicon upgrade. Its uses community firmware

marc-hb, 2024-06-13:
Very recent one today:
https://sof-ci.01.org/softestpr/PR1180/build512/devicetest/index.html?model=LNLM_RVP_HDA&testcase=check-alsabat-headset-playback-821

```
Underrun: Broken pipe(-32)
 FAIL: Peak freq too low 799.07 Hz
 FAIL: Peak freq too low 802.73 Hz
 FAIL: Peak freq too low 806.40 Hz
 FAIL: Peak freq too low 807.86 Hz
 FAIL: Peak freq too high 834.96 Hz
 FAIL: Peak freq too high 838.62 Hz
 FAIL: Peak freq too high 842.29 Hz
```

marc-hb, 2024-06-13:
I would guess the reproduction rate is somewhere between 0.1% and 5% so, yes: 1% falls in that range :-)

jsarha, 2024-06-13:
Did 249 runs more on ba-lnlm-rvp-hda-02, no failures. I'll pick a machine tomorrow again, run couple of hours more. It bothers me a bit that there is an obvious glitch in the audio in the both referred occurrences, but they are not at all similar. Its very hard to imagine a common cause for both of them.

jsarha, 2024-06-14:
[occurrence_2024-06-14-07.zip](https://github.com/user-attachments/files/15836377/occurrence_2024-06-14-07.zip)

Here is one more occurrence, very much like the one in https://github.com/thesofproject/sof/issues/9164#issuecomment-2164285904 . Gap af around 0.2s seconds (one bit longer another shorter) in the middle of the capture, then a little glitch ~ 0.015s before the signal starts to come back in a ramp.

jsarha, 2024-06-14:
That was bit under 300 test runs this morning, and one occurrence.

jsarha, 2024-06-14:
[occurrence_2024-06-14-10.zip](https://github.com/user-attachments/files/15837746/occurrence_2024-06-14-10.zip)
[occurrence_2024-06-14-11-25.zip](https://github.com/user-attachments/files/15837747/occurrence_2024-06-14-11-25.zip)
[occurrence_2024-06-14-11-46.zip](https://github.com/user-attachments/files/15837748/occurrence_2024-06-14-11-46.zip)

Here is three more occurrences, but I do not think its the same issue. They look more like test setup failures to me. The bat.wav files look perfectly Ok, but for some reason the validation fails.

marc-hb, 2024-06-14:
> The bat.wav files look perfectly Ok, but for some reason the validation fails.

I'm not familiar with alsabat but I've been told that it is much more sensitive than the human eye or even ear.

jsarha, 2024-06-17:
> > The bat.wav files look perfectly Ok, but for some reason the validation fails.
> 
> I'm not familiar with alsabat but I've been told that it is much more sensitive than the human eye or even ear.

I did not try to analyze it with my ears, but did frequency analysis with Audacity, and the frequency peak was there exactly in the right place, with no other local peaks.

kv2019i, 2024-06-17:
There's a small glitch in occurrence_2024-06-14-11-46.zip around 0.6740, which explains the alsabat fail.

jsarha, 2024-06-17:
[occurrence-2024-06-17-11-04.zip](https://github.com/user-attachments/files/15872100/occurrence-2024-06-17-11-04.zip)
[occurrence-gap-2024-06-17-12-02.zip](https://github.com/user-attachments/files/15872101/occurrence-gap-2024-06-17-12-02.zip)

Two more cases. One without any immediately obvious fault, but probably some subtle discontinuation in the sine-wave somewhere. The other has the obvious gap-pattern. The gap is a bit wider this time, about 700ms.

marc-hb, 2024-06-18:
> There's a small glitch in occurrence_2024-06-14-11-46.zip around 0.6740, which explains the alsabat fail.

Can you please summarize how you found (with Audacity?) what @jsarha didn't?

kv2019i, 2024-06-18:
> Can you please summarize how you found (with Audacity?) what @jsarha didn't?

I think the same tools can be used. I missed this as well at first as expectation was a big visible gap or a repeating glitch pattern. I noticed this when listening to the file and noted the glitch. Then freq analysis in Audacity in small segments to limit the search space further and final bits by manual analysis of sample values (zooming into waveform display and/or exporting sample data values to text file) to find the exact point.

jsarha, 2024-06-20:
Not sure how much light this sheds the issue, but I first run 673 successful round of alsabat test [1] using sof-hda-benchmark-gain32.tplg, and then quit the test script without a single failure. Then I restored the original sof-hda-generic-ace1-4ch.tplg and was able to run 263 rounds when I hit the error. BUT, the error I hit is of a completely new class. The signal is simply cut off in the middle of sample, and it does not resume. The test logs and the failed test-case is in the attached zip.

[testlogs-and-failed-testcase.zip](https://github.com/user-attachments/files/15917076/testlogs-and-failed-testcase.zip)

[1] TPLG=/lib/firmware/intel/sof-ipc4-tplg/sof-hda-generic-ace1-4ch.tplg MODEL=LNLM_RVP_HDA SOF_TEST_INTERVAL=5 ~/sof-test/test-case/check-alsabat.sh -p hw:sofhdadsp,0 -c hw:CODEC,0 -C 2 -F 821

marc-hb, 2024-06-20:
https://sof-ci.01.org/linuxpr/PR5075/build3763/devicetest/index.html

jsarha, 2024-06-24:
> 
> June 20th: https://sof-ci.01.org/softestpr/PR1180/build552/devicetest/index.html?model=LNLM_RVP_HDA&testcase=check-alsabat-headset-capture-997

Oh, this is yet a new type of failure. There is a gap of only ~6ms and - bit suspiciously - the signal continues from exactly the same phase after the gap. Buffer under-run somewhere?

There is couple of FW log messages like these in the middle of the test log (so not at the setup or tear-down time):
```
[  771.127966] <inf> host_comp: host_get_copy_bytes_normal: comp:1 0x10004 no bytes to copy, available samples: 0, free_samples: 384
[  771.127991] <wrn> dai_comp: dai_common_copy: comp:0 0x4 dai_zephyr_copy(): nothing to copy
[  771.128913] <inf> host_comp: host_get_copy_bytes_normal: comp:1 0x10004 no bytes to copy, available samples: 0, free_samples: 384
```
[

jsarha, 2024-06-24:
I run today another 680 round of alsabat test with sof-hda-benchmark-gain32.tplg to be sure that I just did not get lucky last friday. E.g. run again this test:

https://github.com/thesofproject/sof/issues/9164#issuecomment-2181086368

jsarha, 2024-06-24:
I hacked alsa-utils alsabat to allow 100M frames in the test instead of just 10M and increased MAX_PEAK also 10 fold to 100. Then I needed also fix sof-test check-alsabat.sh to pass -n <frames> parameter also to capture part of the test. This allowed me to run bit over 36min alsabat test, which was success full. I do not have full understanding of alsabat code, so the result should be taken with a grain of salt, but in any case the test was successful:

[alsabat-test-36mis.txt](https://github.com/user-attachments/files/15960987/alsabat-test-36mis.txt)

lgirdwood, 2024-06-25:
> I hacked alsa-utils alsabat to allow 100M frames in the test instead of just 10M and increased MAX_PEAK also 10 fold to 100. Then I needed also fix sof-test check-alsabat.sh to pass -n parameter also to capture part of the test. This allowed me to run bit over 36min alsabat test, which was success full. I do not have full understanding of alsabat code or how my hacks may affect it, so the result should be taken with a grain of salt, but in any case the test was successful:
> 
> [alsabat-test-36mis.txt](https://github.com/user-attachments/files/15960987/alsabat-test-36mis.txt)

Thanks @jsarha - results look good to me. We should upstream the alsa-bat update.

kv2019i, 2024-06-25:
One case today in PR testing, see Intel test run 43099. This has a 3000 sample (one host period?) gap like https://github.com/thesofproject/sof/issues/9164#issuecomment-2167684205 but in this case, audio is corrupt after the gap (dominant 599Hz tone present but multiple other tones). The left-right signals show different values.

jsarha, 2024-06-25:
It took 724 rounds but then I got a glitch with sof-hda-benchmark-drc32.tplg. Not sure if this is directly pointing at drc widget, but anyway here is the eveidence:

[benchmark-drc32-2024-06-25-11.zip](https://github.com/user-attachments/files/15973387/benchmark-drc32-2024-06-25-11.zip)

kv2019i, 2024-06-28:
Reproduction rate low in past week, continue with P2 and assign to v2.11.

jsarha, 2024-07-01:
I became suspicious about the last weeks findings and decided to try them again with enough cycles to know with reasonable certainty in what configurations the issue happens and in what it does not. So I run the test again using following configurationns:
- benchmark-gain32 topology, issue occurred after 583 cycles [1]
- DSPless mode, issue occurred after 852 cycles [2]
- nocodec, playback and capture from hw:0,0, no issue found after 1214 cycles [3]

This starts to point now either to the HDA interface or to the USB audio device that is used for capture in the device. It would be nice to be able to test this with a loop-back cable from RVP line-out to line-in.

[1] [alsabat-benchmark-gain32.zip](https://github.com/user-attachments/files/16056182/alsabat-benchmark-gain32.zip)
[2] [alsabat-DSPless.zip](https://github.com/user-attachments/files/16056183/alsabat-DSPless.zip)
[3] [alsabat-nocodec-log.txt](https://github.com/user-attachments/files/16056184/alsabat-nocodec-log.txt)

plbossart, 2024-07-02:
@jsarha when I look at the captured waveform for DSPless I don't see anything suspicious, there's one nice sinewave with a 0.08 FS value?

marc-hb, 2024-07-02:
> issue occurred after 583 cycles

This just happened again in CI:

https://sof-ci.01.org/softestpr/PR1213/build576/devicetest/index.html?model=LNLM_RVP_NOCODEC&testcase=check-alsabat-headset-capture-821

The reproduction higher feels much higher than 1 out of 583 attempts.... Have you made sure the kernel, firmware and system configuration is 100% identical? Even better: have you just downloaded and installed packages from CI directly? On the same system(s).

jsarha, 2024-07-02:
> @jsarha when I look at the captured waveform for DSPless I don't see anything suspicious, there's one nice sinewave with a 0.08 FS value?

There is an obvious glitch at 1.313s:
[image]

plbossart, 2024-07-03:
@jsarha indeed this looks like a 0.5ms loss of data

[image]

jsarha, 2024-07-04:
To pull together all findings the phase shift glitch is seen on all these configurations, but one. The reproduction rate varies between 1/200 and 1/1000. The nocodec test was running more than 1200 cycles (about 8 hours) without a glitch.

Topology						Playback		Capture	Glitch seen		
sof-hda-generic-ace1-4ch.tplg	sofhdadsp,0	USB		Yes
sof-hda-benchmark-gain32.tplg	sofhdadsp,0	USB		Yes
sof-hda-benchmark-drc32.tplg	sofhdadsp,0	USB		Yes
DSPLess mode				sofhdadsp,0	USB		Yes
sof-lnl-nocodec.tplg				0,0			0,0		No

One common thing with all failed setups is the USB audio device doing the capture. We should still do one more test with analog loopback cable connected back to SOF HDA audio device to rule out a USB audio failure.

The other failure type is the long gap (hundreds of milliseconds) and a ramp when the test signal resumes. My theory there is that the test script, that starts the playback and capture parts assychronously, somehow fails with the timings, and the capture catches the gap between the tests.

marc-hb, 2024-07-09:
Recent reproduction: https://sof-ci.01.org/softestpr/PR1217/build594/devicetest/index.html?model=LNLM_RVP_HDA&testcase=check-alsabat-headset-capture-599

kv2019i, 2024-07-10:
Another case with PR test https://sof-ci.01.org/sofpr/PR9116/build6414/devicetest/index.html?model=LNLM_RVP_HDA&testcase=check-alsabat-headset-capture-997

marc-hb, 2024-07-17:
This looks very similar on... MTL:
https://sof-ci.01.org/linuxpr/PR5110/build4073/devicetest/index.html?model=MTLP_RVP_HDA&testcase=check-alsabat-headset-capture-997

wszypelt, 2024-07-19:
@marc-hb Is it possible to get logs from payloads?

marc-hb, 2024-07-20:
IPC logs should be enabled on LNL. Where are they missing?

marc-hb, 2024-07-23:
Are IPC logs missing in today's failure https://sof-ci.01.org/softestpr/PR1224/build686/devicetest/index.html?model=LNLM_RVP_HDA&testcase=check-alsabat-headset-playback-997?

wszypelt, 2024-07-26:
@marc-hb these logs will be good, thanks!

marc-hb, 2024-08-14:
Still failing regularly: https://sof-ci.01.org/sofpr/PR9351/build6997/devicetest/index.html?model=LNLM_RVP_HDA&testcase=check-alsabat-headset-playback-821

lgirdwood, 2024-08-16:
> Still failing regularly: https://sof-ci.01.org/sofpr/PR9351/build6997/devicetest/index.html?model=LNLM_RVP_HDA&testcase=check-alsabat-headset-playback-821

We can stress this after upgrading to new loopback dongles.

wszypelt, 2024-08-30:
@lgirdwood do you know when we can expect the upgrade?

lgirdwood, 2024-09-02:
> @lgirdwood do you know when we can expect the upgrade?

Probably Q4.

kv2019i, 2024-09-09:
Keeping this open although likely same rootcause as https://github.com/thesofproject/sof/issues/9449

Let's see another week of daily plan results and conclude then.

closed 2024-12-13 by kv2019i