thesofproject/sof #9165: LNL SDW "signal too weak" alsabat capture failure

marc-hb, 2024-05-23:
_Originally posted by @marc-hb in https://github.com/thesofproject/linux/issues/4681#issuecomment-2125436314_
            

"signal too weak" failure in daily test run 41404?model=LNLM_SDW_AIOC&testcase=check-alsabat-headset-capture-599, see logs below. 

EDIT: also in https://sof-ci.01.org/sofpr/PR9159/build4831/devicetest/index.html?model=LNLM_SDW_AIOC&testcase=check-alsabat-headset-capture-821

cc:
- thesofproject/linux#4681

```
2024-05-22 14:27:58 UTC [REMOTE_COMMAND] alsabat -Phw:CODEC,0 --standalone -n 240000 -r 48000 -c 2 -f S16_LE -F 599 -k 2.1
2024-05-22 14:27:59 UTC [REMOTE_COMMAND] alsabat -Chw:sofsoundwire,1 -c 2 -r 48000 -f S16_LE -F 599 -k 2.1
WARNING: Signal too weak!
 FAIL: Peak freq too low 7.32 Hz
 FAIL: Peak freq too low 13.18 Hz
 FAIL: Peak freq too low 16.11 Hz
 FAIL: Peak freq too low 17.58 Hz
 FAIL: Peak freq too low 24.90 Hz
 FAIL: Peak freq too low 42.48 Hz
 FAIL: Peak freq too low 56.40 Hz
 FAIL: Peak freq too low 65.19 Hz
 FAIL: Peak freq too low 68.12 Hz
 FAIL: Peak freq too low 71.04 Hz
alsa-utils version 1.2.6

Entering capture thread (ALSA).
Get period size: 3000  buffer size: 24000
Recording ...
Capture completed.

BAT analysis: signal has 65536 frames at 48000 Hz, 2 channels, 2 bytes per sample.

Channel 1 - Checking for target frequency 599.00 Hz
Amplitude: 1.1; Percentage: [0]
Detected peak at 7.32 Hz of 3.97 dB
 Total 4.0 dB from 7.32 to 7.32 Hz
Detected peak at 13.18 Hz of 4.88 dB
 Total 9.1 dB from 13.18 to 13.92 Hz
Detected peak at 16.11 Hz of 4.13 dB
 Total 10.3 dB from 16.11 to 16.11 Hz
Detected peak at 17.58 Hz of 8.03 dB
 Total 12.3 dB from 17.58 to 17.58 Hz
Detected peak at 24.90 Hz of 9.65 dB
 Total 19.1 dB from 19.78 to 27.10 Hz
Detected peak at 42.48 Hz of 11.11 dB
 Total 25.1 dB from 29.30 to 53.47 Hz
Detected peak at 56.40 Hz of 10.20 dB
 Total 25.8 dB from 55.66 to 60.79 Hz
Detected peak at 65.19 Hz of 9.45 dB
 Total 26.3 dB from 62.26 to 66.65 Hz
Detected peak at 68.12 Hz of 10.30 dB
 Total 26.5 dB from 68.12 to 69.58 Hz
Detected peak at 71.04 Hz of 7.98 dB
 Total 26.6 dB from 71.04 to 71.78 Hz
Detected at least 10 signal(s) in total

Return value is -1003
```

marc-hb, 2024-05-23:
Again in
https://sof-ci.01.org/sofpr/PR9159/build4831/devicetest/index.html?model=LNLM_SDW_AIOC&testcase=check-alsabat-headset-capture-821

marc-hb, 2024-06-04:
Still happening with B0 and April BIOS:
https://sof-ci.01.org/softestpr/PR1203/build477/devicetest/index.html?model=LNLM_SDW_AIOC&testcase=check-alsabat-headset-capture-599

marc-hb, 2024-06-06:
In this run, alsabat playback failed first: https://sof-ci.01.org/softestpr/PR1204/build488/devicetest/index.html?model=LNLM_SDW_AIOC&testcase=check-alsabat-headset-playback-997

Then all capture failed too but not as "usual". The playback failure seems to have left a bad state behind.

lgirdwood, 2024-06-07:
@singalsu @plbossart could this be DC ? 
50/60Hz peaks could also be mains power ?

plbossart, 2024-06-07:
I'd like to park this issue until we've solved the more important Controller Timeout and initialization failed. 
Walk then run, etc.

marc-hb, 2024-06-17:
> I'd like to park this issue until we've solved the more important Controller Timeout and initialization failed.

"Signal too weak" just spotted after the SDW clock fix was merged:

https://sof-ci.01.org/softestpr/PR1205/build514/devicetest/index.html

kv2019i, 2024-06-18:
This starts to look like a codec/setup issue with rt711. I logged on to one CI machine that triggers the issue and run the tests manually. I can obverse:
 - I can reproduce the error, very weak signal in captured by "sofsoundwire,1" (rt711 headset codec)
 - if I run the test in the loop, sometimes test passes, sometimes not (DUT not rebooted, drivers not reloaded, only thing is a runtime suspend to both codec and DSP in between succesful and failing cases)
 - if I boost rt711 ADC gains when the error happens, I can see increased gain in captured signal by the DSP, so seems the DSP is working properly... codec is just not providing correct data
 - I tried playing with "GE49 selected mode" control when the error happens, but this has not impact (at least when modifying the value during streaming)

marc-hb, 2024-06-18:
https://sof-ci.01.org/softestpr/PR931/build518/devicetest/index.html?model=LNLM_SDW_AIOC&testcase=check-alsabat-headset-capture-599

kv2019i, 2024-06-20:
The kernel patch seems to help:

https://github.com/thesofproject/linux/pull/5075
https://sof-ci.01.org/linuxpr/PR5075/build3762/devicetest/index.html

This confirms the hypothesis, so moving the bug to kernel.