thesofproject/sof #9205: [BUG] Noisy DMIC recording with concurrent audio playback over speakers

syedk008, 2024-06-07:
**Describe the bug**
We observed Noisy DMIC recordings with concurrent playback over speakers on the ARL-S platform.
The issue happens only with 4-channel Dmic recording but not with 2-channels.
**To Reproduce**
start and stop DMIC recording and playback over speakers using arecord and aplay.
1. start playback over speakers: aplay -Dhw:0,0 -c 2 -r 48000 <file>
2. start DMIC recording: arecord -Dhw:0,6 -f s32_LE -c 4 -r 48000 <file>
3. wait 10 sec and stop both playback and records
4. playback recorded file to see if any noise.
repeat the steps several times. 
you can use the attached script to reproduce the issue.

**Reproduction Rate**
~25%
**Expected behavior**
No noise in the recording. 

**Impact**
critical

**Environment**
1) kernel: branch: sof-dev and commit:4b0046880585
   sof: v2.9 mtl
2) Name of the topology file
    * Topology: sof-hda-generic-ace1-4ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: ARL-S

[concurrent_rec_play_.txt](https://github.com/user-attachments/files/15710055/concurrent_rec_play_.txt)

lgirdwood, 2024-06-07:
@syedk008 is there any capture WAV file that you can share here ?  Does it also matter if capture is started 1st then playback 2nd ?

syedk008, 2024-06-07:
@lgirdwood attached is the WAV file, rename it to .wav. 
[rec.txt](https://github.com/user-attachments/files/15743088/rec.txt)
I could not reproduce the issue when capture starts 1st and playback 2nd. Also the issue always in last two channels 3-4.

singalsu, 2024-06-10:
Can you please try this pass-through topology version:

[0001-Tools-Topology2-Add-sof-hda-generic-ace1-4ch-48k-pas.patch.gz](https://github.com/user-attachments/files/15775256/0001-Tools-Topology2-Add-sof-hda-generic-ace1-4ch-48k-pas.patch.gz)

syedk008, 2024-06-10:
> Can you please try this pass-through topology version:
> 
> [0001-Tools-Topology2-Add-sof-hda-generic-ace1-4ch-48k-pas.patch.gz](https://github.com/user-attachments/files/15775256/0001-Tools-Topology2-Add-sof-hda-generic-ace1-4ch-48k-pas.patch.gz)

Issue is reproducible with this topology, but the frequency seems less.

syedk008, 2024-06-10:
> Please try also this, 2ch with pdm1 (the controller that handles in 4ch channels 3-4):
> 
> [0001-Tools-Topology2-Add-topology-sof-hda-generic-ace1-2c.patch.gz](https://github.com/user-attachments/files/15775394/0001-Tools-Topology2-Add-topology-sof-hda-generic-ace1-2c.patch.gz)

The issue is not reproducible with this topology.

abonislawski, 2024-06-11:
@syedk008 could you please attach dmesg log with IPC payload data? (for full iteration - aplay - arecord - end)
Our validation will try to reproduce exactly the same scenario on ARL-S without linux driver.

I can confirm it works fine in basic dmic 4ch tests.

syedk008, 2024-06-11:
> @syedk008 could you please attach dmesg log with IPC payload data? (for full iteration - aplay - arecord - end) Our validation will try to reproduce exactly the same scenario on ARL-S without linux driver.
> 
> I can confirm it works fine in basic dmic 4ch tests.

Attached are the dmesg logs. 
[dmesg_clean.txt](https://github.com/user-attachments/files/15794912/dmesg_clean.txt)
[dmesg_noisy.txt](https://github.com/user-attachments/files/15794916/dmesg_noisy.txt)

singalsu, 2024-06-12:
I did yesterday testing with the ARL-S device. Some observations:

- Issues happen with both gcc and xcc builds
- The DSP load seems to impact occurrence, more frequent corruption with higher load topologies. With DMIC capture in passthough but normal HDA playback pipelines
- Switched DCCOMP off in DMIC blob. Also set capture IIR to passthrough with sof-ctl. When corruption happens, the DC level of corrupted signal shifts the noise to fully to positive PCM codes side.
- Set in DMIC blob FIR coefficients to zero to force decimator output to zero. Then applied with DC channel registers a fixed offset to channels. With these forced DC values I was never able to see corruption, after some 20 runs of Karim's script. Just clean stream of constant PCM code values, except the start gain ramps.

singalsu, 2024-06-18:
Fix proposal: https://github.com/zephyrproject-rtos/zephyr/pull/74414

lyakh, 2024-06-25:
With https://github.com/zephyrproject-rtos/zephyr/pull/74414 this problem couldn't be reproduced any more so far. Since the noise when it appears with unpatched firmware begins directly at the start of a capture, it makes sense that modifying DMIC initialisation flow fixes the problem.

kv2019i, 2024-06-28:
Fixed part of release v2.10 (a Zephyr fix so backported directly), removing the tag. Let's close once the Zephyr fix lands in SOF main via west.yml update  (i.e. #9258 or similar).

kv2019i, 2024-07-04:
Fix merged via #9278

closed 2024-07-04 by kv2019i