thesofproject/sof #9209: [BUG] MFCC HiFi4 optimized version is broken

singalsu, 2024-06-07:
**Describe the bug**

With (ipc3) testbench run the HiFi3 build and generic C version produce similar output. The HiFi4 build creates totally different output that looks wrong.

**To Reproduce**

Apply first patch https://github.com/thesofproject/sof/pull/9207, otherwise load to testbench is not working.
Environment variable SOF_WORKSPACE should be set to top level thesofproject directory

Generic C build
```
cd $SOF_WORKSPACE/sof
scripts/rebuild-testbench.sh
scripts/build-tools.sh -t
```

MTL testbench build
```
export XTENSA_TOOLS_ROOT=~/xtensa/XtDevTools
export ZEPHYR_TOOLCHAIN_VARIANT=xt-clang
scripts/rebuild-testbench.sh -p mtl
source $SOF_WORKSPACE/sof/tools/testbench/build_xt_testbench/xtrun_env.sh
```

Run MFCC, generic C
```
cd tools/tune/mfcc
./run_mfcc.sh /usr/share/sounds/alsa/Front_Center.wav

```

Run MTL build of xt-testbench
```
$XTENSA_PATH/xt-run ../../testbench/build_xt_testbench/testbench -q -r 16000 -R 16000 -c 1 -n 1 -b S16_LE \
-t ../../build_tools/test/topology/test-playback-ssp5-mclk-0-I2S-mfcc-s16le-s16le-48k-24576k-codec.tplg \
-i in.raw -o mfcc-hifi4.raw 2> /dev/null

Start octave
```
octave --gui &
[ceps1, t, n] = decode_ceps('mfcc.raw',  13);
[ceps2, t, n] = decode_ceps('mfcc-hifi4.raw',  13);
```
![Screenshot from 2024-06-07 19-31-02](https://github.com/thesofproject/sof/assets/28754793/430f1fda-e424-477f-b53f-bd88d300b68d)

The leftmost plot is cepstral coefficients for utterance "front center". The rightmost is failed hifi4 code run. HiFi3 code version does not have this issue.

```
**Reproduction Rate**
100%

**Expected behavior**
Similar output as with generic C build and HiFi3.

**Impact**
Showstopper for MFCC in MTL.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: 0bda13d0bdaa953f6d98d759f4e805adfb9fb331
2) Name of the topology file
    * Topology: test-playback-ssp5-mclk-0-I2S-mfcc-s16le-s16le-48k-24576k-codec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTL on xt-run simulation

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

kv2019i, 2024-06-19:
@singalsu @lgirdwood I suggest pushing to to v2.11. We are not using MFCC in any upstream topology yet, so this has no impact to 2.10 release.

singalsu, 2024-08-28:
The fix is in #9415.

closed 2024-09-10 by kv2019i