thesofproject/sof #9223: [FEATURE] GUI to demonstrate and control SOF components on target hardware

alexb3103, 2024-06-12:
## Problem Statement:

There is currently no easy way to demonstrate SOF on target hardware, especially to non-developers and people who are not familiar with DSP.

## Proposed Solution:

Develop a generic GUI and TUI that interacts with SOF components on target HW, utilizing sof-ctl for real time control.

This UI will be developed in python, with the GUI utilizing the python GTK library.

The initial features for the first version of the UI (Est. completion Mid July) will be:

- Real time control of audio playback and record through ALSA
- Real time control of Volume, EQ, and DRC SOF components
- Tested functionality on NXP target HW

The features for the final version (completed Late August) will be:

- All previous features
- Clean/professional graphics and intuitive control
- Functionality on all supported HW (some may only work through the TUI)
- Clear interfaces and procedures for adding new SOF components to the UI

## Alternatives

alsamixer and sof-ctl provide some of these functionalities, however utilizing them on their own has a high lack of customization and is also not very presentable to an interested party.

## Additional Context

Hi SOF community, I'm a new GSoC contributor for SOF through the Linux Foundation this summer and will be developing this feature as my project for that. I'm very excited to get started.

lgirdwood, 2024-06-12:
@alexb3103 btw @singalsu has octave scripts that generate the coefficients for all modules (to change processing settings at runtime). I've also got some python that parses topology and creates a Gtk4 listview of all objects (I will clean up and post it soon).

lgirdwood, 2024-06-24:
@alexb3103 I was going to paste here, but GH wont allow me to post python. I will create a staging repo for python tooling that can share common libraries. In the mean time I've pasted the topology2 parser here.
```python
#!/usr/bin/python3
# SPDX-License-Identifier: BSD-3-Clause

import os
import re

# import the Gtk module
import gi
gi.require_version('Gtk', '4.0')
gi.require_version('Adw', '1')
from gi.repository import Gtk, Adw, Gio, Gdk, GLib, GObject

glb_search_dirs = [""]
glb_root_objects = []
glb_root_tokens = []

# keywords to search for in the topology file
tplg_keywords = ["DefineAttribute",
            "DefineArgument",
            "Class",
            "Control",
            "attributes",
            "constraints",
            "[", "]",
            "{", "}",
            ".",
            "!",
            "Object",
            "constructor",
            "immutable",
            "unique",
            "deprecated",
            "mandatory",
            "Define",
            "IncludeByKey"]

def obj_name(line):
    """Extracts the object name from a line."""
    start = line.find('"')
    end = line[start+1:].find('"')
    name = line[start + 1:start + 1 + end]
    return name

def file_name(line):
    """Extracts the file name from a line."""
    start = line.find('<')
    end = line[start+1:].find('>')
    name = line[start + 1:start + 1 + end]
    return name

def search_dir(line):
    """Extracts the search directory name from a line."""
    start = line.find('<searchdir:') # 11chars long
    end = line[start + 11:].find('>')
    name = line[start + 11:start + 11 + end]
    return name

# get the start and end index of a section
def section_get_limits(tokens, idx):
    """Get the start and end index of a section."""
    count = 0
    start = 0
    end = 0
    for i in range(idx, len(tokens)):
        token = tokens[i].get("token")
        if token == "{":
            count += 1
            if count == 1:
[... 666 lines cut ...]
    glb_search_dirs.clear()
    glb_search_dirs.append("")

def import_tplg_as_objects(filename, verbose):

    tplg_reset()

    # Load the topology file into a buffer
    buffer = load_file_into_buffer(filename)
    tokens = tokenize_with_metadata(filename, buffer)

    # get the next object in the tokens
    while True:
        object = tokenizer_next(tokens, 0)
        if object == None:
            break
        glb_root_objects.append(object)

    # print the objects
    if verbose:
        for object in glb_root_objects:
            object.print_tplg()

    return glb_root_objects
```

alexb3103, 2024-06-27:
Wow, this is incredibly useful, I appreciate it a lot. Definitely something for further along in the project but it'll be invaluable for sure.

Let me know when you create the repo, very open contributing to that as well if I make anything useful for it.

singalsu, 2024-07-24:
> @alexb3103 btw @singalsu has octave scripts that generate the coefficients for all modules (to change processing settings at runtime). I've also got some python that parses topology and creates a Gtk4 listview of all objects (I will clean up and post it soon).

Yep. It should be quite straightforward to convert the Matlab/Octave scripts to Python SciPy and NumPy code. In first step retrieving current settings blobs and displaying them graphically as frequency responses etc. would also be quite simple. I have for some component blobs a decoder script.

kv2019i, 2024-12-13:
Cutoff date for v2.12 features today. @alexb3103 @lgirdwood can we close this? It seems the #9406 PR was merged, right. If there's more development, can we perhaps file new tickets to follow-up that work (this issue starts to have a lot of history).

kv2019i, 2025-01-03:
Closing as per above comment.

closed 2025-01-03 by kv2019i