thesofproject/sof #9254: Alderlake/Amstonlake da7219 support in I2S

rhenry-hai, 2024-06-24:
For the purposes of my project, we're working on an amstonlake Atom "x7433RE" CPU with a da7219 codec.

So far, I've been able to create a consistent topology like our hardware implementation and rework the codec configuration with ACPI detection.

I've also customized the sof_da7219 file to bypass the hdmi and dmic outputs and remove the ssp1 link for the purposes of our project.
So far, all sof and bios parameters are correctly set to use SSP0 in I2S.
But I still can't get an audio signal from the codec.

I've measured the various signals coming from the SSP0 link and they don't seem to be correctly configured.
-SCLK Pulse of 170ns every 24µs
-MCLK not present
-BCLK ok
-and any signal on I2S0_TX or I2S0_RX

I've also checked the log of dmesg, but there's no trace of errors or warnings.
I don't really know what's wrong with my installation or configuration.

setup:
Intel CRB with dialog da7219 performance board connected trough I2S.

**Reproduction Rate**
Every times.

**Expected behavior**
for the moment I looking to validate the audio pileline from kernel side to user.

**Impact**
This feature is critical for our project because the purpose of the project is closely related to audio transmission.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 6.1.66
    * SOF: {SHA}
2) Name of the topology file
    * Topology:
```
#
# Topology for Amstonlake with Dialog7219.
#

# Include topology builder
include(`utils.m4')
include(`dai.m4')
include(`pipeline.m4')
include(`ssp.m4')
include(`hda.m4')

# Include TLV library
include(`common/tlv.m4')

# Include Token library
include(`sof/tokens.m4')

# Include Alderlake DSP configuration
include(`platform/intel/adl.m4')

DEBUG_START

#
# Define the pipelines
#
# PCM1 <---> volume (pipe 1,2) <----> SSP0 (headset - da7219, BE link 1)

dnl PIPELINE_PCM_ADD(pipeline,
dnl     pipe id, pcm, max channels, format,
dnl     period, priority, core,
dnl     pcm_min_rate, pcm_max_rate, pipeline_rate,
dnl     time_domain, sched_comp)

# Low Latency playback pipeline 1 on PCM 1 using max 2 channels of s32le.
# Schedule 48 frames per 1000us deadline with priority 0 on core 0
PIPELINE_PCM_ADD(sof/pipe-volume-playback.m4,
	1, 1, 2, s32le,
	1000, 0, 0,
	48000, 48000, 48000)

# Low Latency capture pipeline 2 on PCM 1 using max 2 channels of s32le.
# Schedule 48 frames per 1000us deadline with priority 0 on core 0
PIPELINE_PCM_ADD(sof/pipe-volume-capture.m4,
	2, 1, 2, s32le,
	1000, 0, 0,
	48000, 48000, 48000)

#
# DAIs configuration
#

dnl DAI_ADD(pipeline,
dnl     pipe id, dai type, dai_index, dai_be,
dnl     buffer, periods, format,
dnl     deadline, priority, core, time_domain)

# playback DAI is SSP0 using 2 periods
# Buffers use s16le format, with 48 frame per 1000us on core 0 with priority 0
DAI_ADD(sof/pipe-dai-playback.m4,
	1, SSP, 0, SSP0-Codec,
	PIPELINE_SOURCE_1, 2, s16le,
	1000, 0, 0, SCHEDULE_TIME_DOMAIN_TIMER)

# capture DAI is SSP0 using 2 periods
# Buffers use s16le format, with 48 frame per 1000us on core 0 with priority 0
DAI_ADD(sof/pipe-dai-capture.m4,
	2, SSP, 0, SSP0-Codec,
	PIPELINE_SINK_2, 2, s16le,
	1000, 0, 0, SCHEDULE_TIME_DOMAIN_TIMER)

#
# Bind PCM with the pipeline
#

dnl PCM_PLAYBACK_ADD(name, pcm_id, playback)
PCM_DUPLEX_ADD(Headset, 1, PIPELINE_PCM_1, PIPELINE_PCM_2)

#
# BE configurations - overrides config in ACPI if present
#

dnl DAI_CONFIG(type, dai_index, link_id, name, ssp_config/dmic_config)
dnl SSP_CONFIG(format, mclk, bclk, fsync, tdm, ssp_config_data)
dnl SSP_CLOCK(clock, freq, codec_master, polarity)
dnl SSP_CONFIG_DATA(type, idx, valid bits, mclk_id)
dnl mclk_id is optional

# SSP 0 (ID: 1)
DAI_CONFIG(SSP, 0, 1, SSP0-Codec,
	SSP_CONFIG(I2S, SSP_CLOCK(mclk, 24000000, codec_mclk_in),
		SSP_CLOCK(bclk, 2400000, codec_slave),
		SSP_CLOCK(fsync, 48000, codec_slave),
		SSP_TDM(2, 25, 3, 3),
		SSP_CONFIG_DATA(SSP, 0, 16)))

DEBUG_END
```
    
3) Name of the platform(s) on which the bug is observed.
    * Platform: Alderlake/Amstonlake firmware used (adl-n.ri)

Dmesg outputs : 

```
[   11.169100] da7219_aad: loading out-of-tree module taints kernel.
[   11.208066] da7219_i2c_probe: da7219_i2c_probe 
[   11.213161] da7219 i2c-DLGS7219:00: Using default DAI clk names: da7219-dai-wclk, da7219-dai-bclk
[   11.223134] da7219_aad_probe: da7219_aad_probe 
[   11.609225] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[   11.620253] sof-audio-pci-intel-tgl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[   11.631327] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   11.644098] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[   11.908970] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[   11.917938] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[   11.927204] sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
[   12.030099] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[   12.038789] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[   12.062952] sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:22:0 Kernel ABI 3:23:0
[   12.072069] sof-audio-pci-intel-tgl 0000:00:1f.3: tplg: connected widget SSP0.IN -> DAI link SSP0-Codec
[   12.089939] da7219_probe: da7219_probe 
[   12.129425] da7219_aad_init: da7219_aad_init 
[   12.143055] hai_sof sof_hai_da7219: ASoC: driver name too long 'sof-da7219_custom' -> 'sof-da7219_cust'
[   12.160701] input: sof-da7219_custom Headset Jack as /devices/pci0000:00/0000:00:1f.3/sof_hai_da7219/sound/card0/input2
```

Thank you for your help, Sincerely Romain

plbossart, 2024-06-25:
I would start with the "nocodec" configuration, as is, and try to see if there are any signals on the SSP0 pins. If not, check the BIOS settings and possibly the pin-mux configuration. SSPs are typically muxed with other interfaces and they are not the default interface.

IIRC you have to use the sof_debug option with the setting 0xC01 to force the nocodec option.

#define SOF_DBG_FORCE_NOCODEC      BIT(10) /* ignore all codec-related  configurations */ 

options snd-sof sof_debug=0xC01

rhenry-hai, 2024-06-25:
Hi plbossart,

Thanks for your feedback, I've been able to experiment with the sof-debug option of nocodec activated, I can identify the codec "nocodec" and the selected topology "sof-adl-nocodec" but I still can't get the correct outputs on the I2S0 link.

I also re-checked the bios configuration to activate the different channels ssp0,ssp1,ssp2 for the "nocodec" mode.

Regards Romain.

Regards Romain.

plbossart, 2024-06-25:
It's not easy to enable these interfaces, there are multiple things that can go wrong.
It doesn't help that this version of hardware is three generations old, i.e. not under active development.
I would update the kernel, 6.1.66 is about 8 versions old, that's way too old for this sort of integration endeavors.

lgirdwood, 2024-06-25:
@rhenry-hai the nocodec configurations upstream are all verified by CI so they should work as they do in CI. However, as @plbossart has suggested, there may be BIOS or pinmux configuration that has the SSP pins being configured for another usage and therefore you wont see any I2S signal until this is resolved..

closed 2025-07-24 by github-actions[bot]