thesofproject/sof #9262: [BUG]  PTL doesn't have an open source toolchain, "xtensa-build-zephyr.py ptl" fails by default

andyross, 2024-06-26:
The PTL platform merged without a toolchain added to the Zephyr SDK for the ace30_ptl SOC, or any other way to create one.  This means that builds from a bare tree checkout are always going to fail without access to the Cadence SDK for the device.  Also new work that might impact the platform (c.f https://github.com/thesofproject/sof/pull/9261) can't be validated by the submitter.

At the very least, if there's resistance to adding another toolchain to the already huge Zephyr SDK[1] an overlay file should be added to the SOF xtensa-overlays project  ( https://github.com/thesofproject/xtensa-overlay ) so that users can build their own.

[1] There are, to be sure, **a lot** of Xtensa variants in there now.

andyross, 2024-06-26:
FWIW: another side band option would be to get partners copies of the relevant Cadence SDK integration files so they could do the work themselves.

abonislawski, 2024-07-02:
> At the very least, if there's resistance to adding another toolchain to the already huge Zephyr SDK[1]

Do you have this in mind?
https://github.com/zephyrproject-rtos/sdk-ng/pull/767

@lgirdwood not sure about the current policy, we should have it also in xtensa-overlay repo or sdk-ng is enough?

marc-hb, 2024-08-14:
Current GitHub workflows in `sof/.github/` use a Zephyr docker image exclusively - because it provides includes the Zephyr SDK.
https://github.com/zephyrproject-rtos/docker-image

- https://github.com/zephyrproject-rtos/zephyr/issues/63738
- https://github.com/zephyrproject-rtos/zephyr/pull/63759

kv2019i, 2025-01-03:
With #9732 we can now close this.

closed 2025-01-03 by kv2019i